Arduino-IRremote/Arduino-IRremote #199: Disscussion: Open Source Alternative to Analysir

z3t0, 2015-08-16:
@zenwheel @DaAwesomeP @sEpt0r @joshuajnoble @sstefanov @levsa @pcoughlin @csBlueChip @crash7 @mariocarta @cyborg5

## Open Source Alternative to Analysir

NOTE: This initial post will continue to be updated as I further iron out my ideas and gather other ideas from everyone else.

First of all let me just make it clear that I do not mean to offend @Analysir as they work on an amazing project which, I am CERTAIN, is useful to many people. However, I have come to realize that there are likely to be many people out there who would be interested in an open source alternative which had proper support. 

I was thinking it would be an interesting project to work on and would allow the community to add their own contributions. Also this project would mainly be community driven and of course it would be free, although donations would probably be welcome (I am not sure of this yet).

If you are interested or have any ideas I would love to hear them.

### Features
- Open Source Code and Hardware Designs(Either MIT or GPLv3)
- Official Support from Arduino IRremote (this name may change as I am considering merging efforts with @cyborg5)
- Cross Platform Support: I know that one of the many shortcomings of Analysir is the fact that it only works on windows. I have reached out to them offering to help support other operating systems, but they have responded claiming that the cost would be too much. This is completely understandable. However, this open source project would have Cross-Platform support at the forefront.

### Cons
- Community Driven: This is not exactly a con but rather here to point out that the project will only stay alive so long as there are others out there supporting it, whereas Analysir is maintained by a group of developers who are dedicated to delivering the best service/product for their customers.

Thank you for reading!

z3t0, 2015-08-16:
### Update 

I have gone ahead and created a repository at https://github.com/z3t0/OpenIR

z3t0, 2015-08-16:
### Thoughts:
- Qt seems like a good idea for an open source project, especially when the visual aspects do not need to be "hand-coded", this also means going to C++ which would fit in with the fact that C++ is used in this library

z3t0, 2015-08-16:
If anyone has purchased Analysir (I personally do not have a copy) then I would be very interested if they were to share the features of the software so that I can work on creating similar support on OpenIR.

z3t0, 2015-08-16:
###Update:
I have updated the [README](https://github.com/z3t0/OpenIR) with some more information

z3t0, 2015-08-16:
### Update:
I have realized that this project is a great opportunity for an excuse for me to spend some time delving into the source code of this library and making some changes as to how things work and such, check #200 and the link to the new repository can be found here: https://github.com/z3t0/SmartIRLib.git

jassing, 2015-08-16:
AnalysIR isn't that expensive, especially when compared to the time it
saves you or when compared to other kit tools.
I believe you can get a demo version...
I use it to decode IR signals so I can duplicate them, including frequency
of the signal.
It has two channels, so you can decode from two different remotes at the
same time & easily compare.
It's nice as you can send signals just as easily to test things before you
commit it to actual code.
The support for multiple input/output modules is nice.
You can zoom in/out on the raw signal (and now it can zoom in-sync with
both graphs).
You can adjust tolerances for decoding.
There's a bunch more; but I think those are the big ones.
They really did a good job decoding a bunch of IR protocols, and they
included an easy way to add your own.
Don't forget the longer A/C (Air Conditioning, not the current) codes.

I think an open source tool would be great for the community, but I doubt
it would be easy to achieve the same level of AnalysIR.  It's really a well
done, top notch program.  Forget it's the only game in town, even if it
wasn't - it would be hard to surpass.   Their support is also fantastic.

But for the very low requirement users, I can see the cost of AnalysIR *might
*be a factor, I would have no problem buying another license, and I'm on a
fixed income & do IR stuff as a hobby only.  When you consider their least
expensive version is only $10 and most expensive is $20 -- it's really hard
to argue a "free version" is needed.  That's my $.02.

DaAwesomeP, 2015-08-16:
So what exactly is the purpose of the software (I'm just wondering)? From the name it seems that it would give you "an oscilloscope for IR commands" as I would word it. 

Is it a method to control things on your computer with IR and then sending those commands regularly (with a service) to control other things? If so, then there's LIRC with that. It's compatible already with everything that it should be and is rock solid at what it does.

Also, only use Qt if you're going to use C/C++ or Python. From my experience, Qt in other languages simply isn't practical. My app experience is mostly with NodeJS, so as far as *not* using Qt with C++, I can't really tell you what else.

Also, stick with Serial/USB over Serial. It will make everything easier. Don't even mess with masking the Arduino as another type of device.

As for the features of AnalysIR, I found this: http://www.analysir.com/blog/wp-content/uploads/2014/10/Product-Sheet-AnalysIR.pdf?41a1b1

z3t0, 2015-08-16:
Firstly I would likely to thank you immensely for commenting.

@jessing 
> AnalysIR isn't that expensive, especially when compared to the time it
saves you or when compared to other kit tools.

That is a completely valid point, but I feel that there are other reasons for an open source alternative including cross platform support and of course being open source :smile:.

>  think an open source tool would be great for the community, but I doubt
it would be easy to achieve the same level of AnalysIR.  It's really a well
done, top notch program.  Forget it's the only game in town, even if it
wasn't - it would be hard to surpass.   Their support is also fantastic.

Another extremely valid point, but I personally believe that with enough determination and effort it will be possible to create something of equal value, especially if many people show their interest in the project. However, we must definitely commend the Analysir team on their excellent job.

>  it's really hard
to argue a "free version" is needed.

Well said, however its more the need for an open source version rather than a cheaper or free version. I personally do not think the cost is an obstacle at all (for most people) and am even considering purchasing a license as it is a great and useful product. One of my main thoughts is that the whole Arduino platform is open source and so I would think that most Arduino users would prefer if they could use open source hardware and software, at least that is how I feel.

> That's my $.02

Maybe so, but its worth a lot more in my eyes :smile: 

Thank you very much for commenting and informing me of the features supported by Analysir, please comment some more as I always enjoy having a good conversation. 

What would you expect from an Open Source alternative? I would love to know.

z3t0, 2015-08-16:
> So what exactly is the purpose of the software (I'm just wondering)? From the name it seems that it would give you "an oscilloscope for IR commands" as I would word it.

I guess it really does come down to that. Ideally it is a Decoder and Analyzer for infrared protocols. The receiver being an arduino or similar. This becomes extremely useful when using this library and is usually more informative than an oscilliscope in general.

> Is it a method to control things on your computer with IR and then sending those commands regularly (with a service) to control other things?

Not really, as you said above, it is more like an oscilloscope. So it would have features such as decoding incoming signals from the receiver, and also adding new protocls, analyzing decoded IR signals etc. The features are quite nicely mentioned by @jassing above. I will be posting a similar list as a goal for the project sometime soon.

> Also, only use Qt if you're going to use C/C++ or Python. From my experience, Qt in other languages simply isn't practical. My app experience is mostly with NodeJS, so as far as not using Qt with C++, I can't really tell you what else.

Thank you very much for the advice, so far my plan is to go with Qt and C/C++ the main reasons being cross platform support and because C/C++ is what is being used to write this library, and I plan for both projects to work nicely together.

> Also, stick with Serial/USB over Serial. It will make everything easier. Don't even mess with masking the Arduino as another type of device.

Excellent advice, I was thinking of doing something like this.

Also thanks for the list of features.

z3t0, 2015-08-16:
Just thought of something.

In being open source, it can easily be magnitudes of times worse than Analysir... but has the potential to be so much better.

DaAwesomeP, 2015-08-16:
> In being open source, it can easily be magnitudes of times worse than Analysir.

@z3t0 if every open source project started with that in mind, then there would be no open source projects. The gain is that if it fails, then no money is lost, and you wouldn't still have to pay the people that helped you with it if you give out only one copy. Plus, you get free support and help on your project.
> ...but has the potential to be so much better.

Most things turn out this way.

bengtmartensson, 2015-08-16:
Hi Guys,

please have a look at my work, [IrScrutinizer](http://www.harctoolbox.org/IrScrutinizer.html), sources at https://github.com/bengtmartensson/harctoolboxbundle. It addresses most of the issues in the "thread". I contrast to AnalysisIR, it

* is opensource (GPL3)
* truly cross platform (Java)
* decodes/encodes hundreds of protocol
* Generated protocols are extensible, using [IRP notation](http://www.hifi-remote.com/wiki/index.php?title=IRP_Notation)
* Many import and export formats. "Simple" export formats can be added by the user.
etc etc

Please have a look, It is a fairly mature work, containing some 60000 lines of (non-comment) Java code. Cooperation is welcome. 

I also have "my" Arduino IR-Library: https://github.com/bengtmartensson/AGirs

Random comments to the above:

    there's LIRC with that. It's compatible already with everything that it should be and is rock solid at what it does.

No, not at all. Lirc sucks. Really. It is a 20 year old program, that does not at all stand up to todays standard.

Please do not start a *new* project using Python 2. Show that much respect to the Python community.

z3t0, 2015-08-16:
@DaAwesomeP 

> > ...but has the potential to be so much better.

> Most things turn out this way.

That's the plan

z3t0, 2015-08-16:
@bengtmartensson 

> please have a look at my work, IrScrutinizer, sources at https://github.com/bengtmartensson/harctoolboxbundle. It addresses most of the issues in the "thread". I contrast to AnalysisIR, it

Thanks for the reply, I will look into the project, as you said it seems to address most of my reasons for creating an alternative.

> Please have a look, It is a fairly mature work, containing some 60000 lines of (non-comment) Java code. Cooperation is welcome.

Sounds, good

> I also have "my" Arduino IR-Library: https://github.com/bengtmartensson/AGirs

Will take a look. Also, me and a few others have been thinking about merging efforts with @cyborg5, https://github.com/cyborg5/IRLib/issues/13

> Please do not start a new project using Python 2. Show that much respect to the Python community.

Sure.

@bengtmarten sson Would you be interested in creating an organization (I have already made one and can invite you if you are interested) and then housing all of the current, supported IR projects there? I have already invited @pcoughlin and will invite you if you are interested.

cyborg5, 2015-08-16:
You do not need to send copies of all of this correspondence. I'm aware of your project and I wish you luck. You are free to use my library in whatever capacity is useful to you as long as you follow the GPL license. Eventually I get back into IR work and I will take a look at how your project is coming then.

 

Will take a look. Also, me and a few others have been thinking about merging efforts with @cyborg5 <https://github.com/cyborg5> , cyborg5/IRLib#13 <https://github.com/cyborg5/IRLib/issues/13>

z3t0, 2015-08-16:
> You do not need to send copies of all of this correspondence.

Not too clear what you mean, if you are referring to being sent emails of updates on this issue it is most likely automatic because I had mentioned your name in the initial post.

> I'm aware of your project and I wish you luck. You are free to use my library in whatever capacity is useful to you as long as you follow the GPL license. Eventually I get back into IR work and I will take a look at how your project is coming then.

Thank you very much, I appreciate it.

cyborg5, 2015-08-17:
Your emails are sent to "z310/Arduino-IRremote@noreply.github.com" with a cc:"cy_borg5@cyborg5.com" as well. Just remove my name from the cc: field.

z3t0, 2015-08-17:
Oh.. I certainly do not have you being cc’d in my email client… Maybe remove yourself from the Issue’s participants list?

bengtmartensson, 2015-08-17:
> Would you be interested in creating an organization (I have already made one and can invite you
>     if you are interested) and then housing all of the current, supported IR projects there?...

Sure, I would be interested. Although (of course) the statement is not an unconditional commitment...

pcoughlin, 2015-08-17:
Unconditional Love is the only place I know that word works for me ...
And, Ben, I don't know you that well. LOL!

Basically just asking if you would want to participate/help.  No commitment needed.  We are identifying people who can lend incite (you're obviously able).  With any project, there are insiders and others that contribute from time to time.

z3t0, 2015-08-17:
@bengtmartensson Agreed, no commitment needed. Just looking for people interested to lend a hand or some advice

z3t0, 2015-08-19:
@bengtmartensson I took a look at the projects you mentioned and they seem to have met many of the goals for this project. I will take a close look at the source code once I play around with it some more.

bengtmartensson, 2015-08-20:
I am open to all constructive suggestions. Please also have a look at the issues, where I, in some cases in some detail, describe what I intend to do in the near future.

z3t0, 2015-08-21:
@bengtmartensson For now I am going to go ahead and work on OpenIR, mainly to learn about how everything works in greater detail. After that I will probably just contribute to what you already have working.

bengtmartensson, 2016-01-23:
Five months have passed since last comment. There does not seem to be any activities on OpenIR, at least not visible on GitHub.

I have a lot of [open issues for IrScrutinizer](https://github.com/bengtmartensson/harctoolboxbundle/issues), many of these explicitly marked as "help wanted". I have also written a [ROADMAP file](https://github.com/bengtmartensson/harctoolboxbundle/blob/master/doc/ROADMAP.md) stating some more long-time goals. 

Love to have cooperation, and/or qualified feedback.

z3t0, 2016-01-25:
@bengtmartensson Yeah sorry for that... I have been working on a few other projects but I did get a chance to look at your IrScrutinizer project and from what I can see it seems to meet the requirements perfectly. As soon as I get a chance I will try and contribute to that. 
Thanks!

closed 2020-10-19 by ArminJo