Arduino-IRremote/Arduino-IRremote #204: decoding issue

maxwlchow, 2015-08-23:
I just download and try the IRrecvDump2 example. I use uno with 16MHz crystal.
I found only few remote was recognized correctly. I test many remote and seems only Sony remote is correctly recognized (12bit). Sharp, pioneer, panasonic and LG remote all can't be recognized. sometimes pioneer and LG will recognized as NEC. More interesting is on Sharp remote, two different code may come out with the same button (32bit). Also I try a universal remote, it recognized as type RC5, for the same button, sometimes it recognize is D05, sometime it is 505. another button sometime is D02, sometime is 202.
I also use my android photo with IR transmitter to test, the photo work correctly with all my TV with various brand such that I can try many many brand on the Arduino. Just like the REAL remote, only SONY was correctly recognized and each button gives same code everytime. All other brand seems having problems.
I am using a 38KHz IR receiver, run at 5V same as the MCU. I try two different receiver with the same result.

AnalysIR, 2015-08-23:
The RC5 issues are most likely the 'Toggle' bit, which would be normal. Assum eyou can get 2 different buttons or when checking - mask out the toggle bit.

For the other issues - what model of IR receiver are you using? Data sheet link?

maxwlchow, 2015-08-23:
You are right, the RC5 sending out D05, 505 alternatively with the same button.

I use this:

http://wenku.baidu.com/view/9687788d84868762cbaed501.html

everlight IRM-3638T-R

AnalysIR, 2015-08-23:
On paper your IR  Receiver looks acceptable, but adding in the additional 2xresistors & 1xcapacitor may imporove performance (ie reduce interference). 

FYI: not every IR protocol is recognised by IRreomte, but you should be able to use the output of IRrecvDump2 with sendRAW to get your signals working.

chaeplin, 2015-08-23:
decodeLG is for A/C.
https://github.com/z3t0/Arduino-IRremote/pull/69/files

LG A/C remote : two type of hdr mark/space
1) floor/tower
2) wall

closed 2020-10-12 by ArminJo