Arduino-IRremote/Arduino-IRremote #208: IR led not emitting pulses, using IrRecord for transmission

curiousmech, 2015-08-25:
Hi,
I am using IrRecord to record Sony code and replay the code using arduino mega 2560.
I have tested the ir led individually and its in working condition,but as soon as i press button to send the code it displays that the code is send but i can't see the ir led working (using camera to observe emission). The button is continuously pressed to transmit code.
Also i have checked the pwm pin 3 by using other code by toggling led at 38khz and it works but its not working using IrRecord.
What could be the problem?

pcoughlin, 2015-08-25:
Please include your sketch so we can take a look at what your program does.
If you have your breadboard setup, then a picture of that as well.

What is the Make/Model of your camera?

Happy to help, but we need more detailed information.

curiousmech, 2015-08-26:
@pcoughlin since i have problem in transmitting code i thought to test only the irSendDemo .I included the libraries -> IRremote.h and IRremoteInt.h but it is also not working .
it gives following error:

C:\Program Files (x86)\Arduino\libraries\RobotIRremote\src/IRremote.cpp:76: multiple definition of `IRrecv::IRrecv(int)'

C:\Program Files (x86)\Arduino\libraries\RobotIRremote\src/IRremote.cpp:69: multiple definition of `irparams'

:\Program Files (x86)\Arduino\libraries\RobotIRremote\src/IRremote.cpp:69: multiple definition of `IRrecv::decode(decode_results*)'
IRremote\IRremote.cpp.o:C:\Program Files (x86)\Arduino\libraries\IRremote/IRremote.cpp:169: first defined here

Irled connected directly to pwm pin3 with resistor of 100ohm in series. also checked it via connecting using transistor as a switch.
Kindly guide me.
Note: I am using lg g2 cam to verify the working of irled.

AnalysIR, 2015-08-26:
The Tx pin on the Mega 2560 is different....try pin 9 instead of pin 3

pcoughlin, 2015-08-26:
If you would like us to comment on your Sketch, then you will need to provide a copy of it.
Please put your code into a Gist through GitHub and put the link here.

https://gist.github.com/    <-- Create an account on GitHub is you do not have one (it is free and very useful).

We need the code to help on code issues.

@AnalysIR -- Good catch on the pin usage. +1

z3t0, 2015-08-26:
@curiousmech This folder needs to be deleted, it conflicts with the library, C:\Program Files (x86)\Arduino\libraries\RobotIRremote\

curiousmech, 2015-08-27:
@z3t0 thanks sir, the program compiled without any error , but the ir led not sending any data. i have checked on both pin9 and pin 3. @AnalysIR 
though using IRLib by Chris Young works (http://cyborg5.com), transmitiing data using pin 9

AnalysIR, 2015-08-27:
Please read the above replies from @pcoughlin again.

curiousmech, 2015-08-28:
@AnalysIR @pcoughlin 
Here you go:
https://gist.github.com/curiousmech/75a7147b8ab8caa22dde

AnalysIR, 2015-08-28:
remove the #include <IRremoteInt.h> from your sketch.

also add in a 5 second delay at the end of the for loop.

curiousmech, 2015-08-29:
@AnalysIR 
the header file <IRremoteInt.h> is added automatically as soon as I include the library.

z3t0, 2015-08-30:
@curiousmech Please try the latest update. The send demo has been changed.

AnalysIR, 2015-08-30:
@curiousmech You are correct, by including the IRrremote library via the IDE menu, it results in a compile error.

In the meantime, just follow my last suggestion and it should be fine. Also, using the latest send demo (via the examples menu of the IDE) as suggested, is a good idea.

@z3t0 looks like including the library via the Aduino IDE results in a compiler error, because ```#include <IRremoteInt.h>``` gets included automatically. Sounds like a :bug: bug?

On my system it conflicts with the robotIRremote, but when IRremoteint.h is commented out it compiles fine.

curiousmech, 2015-08-31:
@AnalysIR @z3t0  i have tested it with the updated code but it is still not working, checked both pin3 and pin 9. While, using Chris Young IRLIB its working and transmitting data using PIN9.

TejaAllani, 2017-02-15:
@z3t0 I have the same problem, I am using arduino Pro Micro and connected my IR LED to pin 9 through a 330 ohm resistor, It worked fine yesterday I was able to successfully turn my TV on and off but its not working today, I don't know what might have gone wrong. I tested my LED and I also tested the default IRsendDemo using arduino nano by connecting IR LED to pin 3. It is also not working. When I use IRlib2 library the IR LED seems to be pulsing just fine.
`#include <IRremote.h>

IRsend irsend;

void setup()
{
}

void loop() {
	for (int i = 0; i < 3; i++) {
		irsend.sendSony(0xa90, 12);
		delay(40);
	}
	delay(5000); //5 second delay between each signal burst
}`

TejaAllani, 2017-02-15:
@curiousmech I fixed my problem by changing timer pins in boarddefs.h library. 
```
#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
    //#define IR_USE_TIMER1   // tx = pin 11
    #define IR_USE_TIMER2     // tx = pin 9
    //#define IR_USE_TIMER3   // tx = pin 5
    //#define IR_USE_TIMER4   // tx = pin 6
    //#define IR_USE_TIMER5   // tx = pin 46
```
check your boards timer pins and select unoccupied timer pins itseems your timer2 is occupied by some other program use timer 1 or 3

closed 2020-10-12 by ArminJo