Arduino-IRremote/Arduino-IRremote #209: Adding support for ESP8266

renno-bih, 2015-08-25:
These wireless modules are pretty popular these days. They are more powerful than Arduino, smaller, they have WiFi connectivity and Arduino IDE can be used for programming.

It would be amazing to support ESP8266 in this library.
Any thoughts ?

pcoughlin, 2015-08-25:
I have one of these modules, but have not hooked it up.

If the module can be programmed with the Arduino IDE, then a majority of the compatibility issues are already solved.

Have you tried it?  You should build a sketch and include the library in the Arduino IDE and let us know what happens.

Also, if you include the Hardware setup, then we can duplicate it and make the effort to get it all working.

z3t0, 2015-08-26:
Please correct me if I am wrong, but these modules are add-ons for the Arduino. And they only interface with Wi-Fi, so how do we add support for them if this is an infra red library? Thanks

pcoughlin, 2015-08-26:
The ESP8266 is actually a complete MCU.  They originally created the little card to be an add on for Arduino.  The software in the MCU does just that.  It is also possible to re-flash the chip and make use of the other I/O pins and use the radio as well.

Many of those little boards have MCUs and are independently programmable.
https://www.adafruit.com/images/product-files/2471/0A-ESP8266__Datasheet__EN_v4.3.pdf
See section 2.5 and 2.6.

AnalysIR, 2015-08-26:
There are several ports of this already on github - via search. At least one uses a version of the Arduino IDE. Another uses bit banging for PWM.

It is a very different platform from AVRs.

z3t0, 2015-08-26:
@pcoughlin Thanks, that makes more sense. Do you know what MCU's they run on?

baget, 2015-08-27:
@z3t0 here is the ESP8266 spec
https://www.adafruit.com/datasheets/ESP8266_Specifications_English.pdf
and wikipage
https://github.com/esp8266/esp8266-wiki/wiki

e-dredon, 2015-08-31:
Mark Szabo has ported this lib for the ESP8266 with the arduino framework : https://github.com/markszabo/IRremoteESP8266/
His library supports Receive & send.

You can find another port for RTOS that requieres the unofficial SDK + eclipse IDE 
(the lib is already included in the unofficial SDK

sebastienwarin, 2015-09-23:
Hi everyone,

The Mark Szab's lib only enables you to send infra-red signals (not receive) !

So I have fork his repo and add the receiving part by including the Espressif ESP8266 SDK to access native functionality (Note that this makes the code dependent on the ESP8266).

https://github.com/sebastienwarin/IRremoteESP8266

Have fun,

Nca78, 2016-07-07:
Hello, IRemoteESP8266 has been implementing both sending and receiving for some time now.
But the code is based on an old version of IRemote (all protocols in one file etc etc) so I imported the code in Arduino-IRremote, I am testing now and when it's ok (and I figure out how it's done, I'm very fresh to GitHub :)) I will create a pull request.
I'm using preprocessor directives so it stays compatible with the platforms currently managed by IRremote.

@z3t0 sorry to bother you but I have a question, why is the _ISR (TIMER_INTR_NAME)_ routine defined in IRremote.cpp and not in irRecv.cpp as it seems to be the most logical place for me ?
I have some specific methods for ESP8266 that I put in irRecv.cpp as they are called from enableIRIn, but now I have a _#ifdef ESP8266_ for those in irRecv.cpp and a _#ifndef ESP8266_ in IRremote for the ISR routine and it doesn't sound "clean" to me ;)

AnalysIR, 2016-07-07:
FYI: we have just published an article on our blog, with examples, showing how to accurately generate hardware based PWM for IR carrier signals on the ESP8266 NodeMCU, using our uPWM method. (hardware based PWM on the ESP8266 is supported from 100Hz to 1kHz (AFAIK). The method we published supports all common IR frequencies & a good selection of duty cycles. Previously all implemetations on this platform relied on software loops to generate the PWM. For anyone with the time available, there is no reason it couldn't be incorporated into the main IRremote repository. Let us know if you find it useful.
[ESP8266 NodeMCU Backdoor uPWM hack for IR signals wit UART](http://dangerousprototypes.com/blog/2016/07/06/esp8266-nodemcu-backdoor-upwm-hack-for-ir-signals/)

Nca78, 2016-07-08:
Wow, looks interesting indeed !
I'll have a more detailed look later, but the complexified output circuit is putting me off a bit ;)

AnalysIR, 2016-07-08:
yes, it could be simplified using one PNP....but using the 2 NPNs serves to highlight that the output is inverted vs normal.

The circuit displayed can be easily modified for low power/battery operation if required, by driving the 3V3 to R2 from a GPIO pin combined with disabling the UART when not in use.

closed 2020-08-17 by ArminJo