Arduino-IRremote/Arduino-IRremote #210: NEC sending signal twice

trunner10, 2015-08-26:
When I use the sendNEC command it sends the ir command twice.  So when i try to change the channel it changes it twice.  Is there a way to not send the command twice or am i doing something wrong.  Thanks

AnalysIR, 2015-08-26:
Based on the information you provided, you are probably doing something wrong...

...more information, like your code would help.

z3t0, 2015-08-26:
@trunner10 Please edit your comment with a link to a gist containing your sketch.

AnalysIR, 2015-08-28:
you are probably not parsing the incoming http header & get request properly. the text in the get request is likely causing your case statement to mis-fire....I could be wrong as i didn't spend much time on it and your are right....you may be better posting on the Arduino forum.

z3t0, 2015-08-28:
@trunner10 Would love to help but I have never played around with the web and arduino. I will take a look at your code and see if I notice anything.

closed 2020-08-17 by ArminJo