Arduino-IRremote/Arduino-IRremote #214: Issues when compiling IRremote with Arduino-Makefile

audetto, 2015-08-31:
Hi,

I've been using for a while the Makefile for Arduino 

https://github.com/sudar/Arduino-Makefile

but recently had an issue where IRremote would be miscompiled.
Compiles, links, uploads, but hangs in

irrecv.enableIRIn();

everything started here

https://github.com/z3t0/Arduino-IRremote/commit/88e243fe068e06d2ed602fefe6bd5effd07006e1

when the library was split in many files.

(issue for Arduino-Makefile: https://github.com/sudar/Arduino-Makefile/issues/376)

Do you know guys of anything that would cause issues in the new layout of IRremote code?
Comparing the Makefile with Arduino IDE 1.6.5, I can see that the output of the Makefile is a bit smaller (0.5KB), while before the refactoring they were the same.

Probably some obscure compiler or linker flag. Just checking if it rings a bell.

audetto, 2015-09-01:
I've found what the issue is (long description in the linked issue).
Basically irISR.cpp exports no symbols so accordingly to the tool chain setup it might get included or not.

In the Makefile, a static lib of the libraries used is created, so at the end, the linker does not have any reason to include it in the final output.
The IDE does link in a different way and it works.

Probably a weak feature to rely on, and forcing the linker to include the ISR is not difficult.

I would propose to merge irISR.cpp back into irRecv.cpp so whenever the receive code is linked in, the ISR is set no matter what the linker chooses.

(I seem to understand the ISR is only needed to decode, otherwise the best location would be some file that gets included all the times)

I will prepare a pull request: what do you think?

z3t0, 2015-09-01:
@audetto Sure, please prepare a PR and I will merge it. I was not aware of this problem because I use the Arduino IDE to compile and upload, thanks for pointing it out.

audetto, 2015-09-09:
I've prepare the PR.
Is it good? Do you like it?

audetto, 2015-09-12:
Perfect. All done, working now.

closed 2015-09-12 by audetto

z3t0, 2015-09-12:
No problem, thanks for the contribution