Arduino-IRremote/Arduino-IRremote #216: resources problem

maxwlchow, 2015-09-05:
I want to use IR remote to select different LED light show and Luminance. But since the LED show always need to run into loop, this remote library can't process the IR signal after the first show was started.

z3t0, 2015-09-05:
It is not very clear what you want to do or what the problem is. Please
share your sketch with us and give us more details about your project.

pcoughlin, 2015-09-06:
@maxwlchow Do you have a specific device that you are trying to get your Arduino to control?
If YES, then put the URL for the specifications to that device here.

Please give more details about your project as well.

kenan-aisy, 2015-09-07:
Hi,
I had the same problem, you need to check the ir-input every loopcycle or every x cycles.
This ist my LEDdriver code:https://github.com/kenan-aisy/LEDDriver.git i am useing this IR library and have started some days ago.
Maybe this can give you some inspiration. I am still working on it and keep updateing

closed 2020-08-17 by ArminJo