Arduino-IRremote/Arduino-IRremote #217: more of RAWBUF Raw len

lettierimarco, 2015-09-07:
i've just sampled a signal of more of RAWBUF len (>200) byte on a haier AC control.
What's the way to solve this issue?

pcoughlin, 2015-09-09:
In the file IRremoteInt.h on or about line 40 you should see this:
```
//------------------------------------------------------------------------------
// Information for the Interrupt Service Routine
//
#define RAWBUF  101  // Maximum length of raw duration buffer
```
Change the 101 to some value that will be big enough to read your signal.

closed 2015-09-09 by lettierimarco

AnalysIR, 2015-09-10:
Try the sketch in ourblog post linked below, which will provide more accurate timings. & works for longer AC signals.

http://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/