Arduino-IRremote/Arduino-IRremote #218: change irsend pin

knh6733, 2015-09-08:
I want to change irsend pin3 to another pin
how could i do it
help me TT

pcoughlin, 2015-09-09:
What board are you using?
What pin would you like it to be?
Why do you need to change the pin?
What are you trying to achieve?

I will be happy to help, just need to know more to be able to give you advice.

knh6733, 2015-09-09:
I use arduino uno
Irsend 3pin to another any pin
I am now using smarthings arduino shield
I am trying to use both irsend code and smartthings shield code.
The problem is both use pin3 
I wanna make Tv remote control by smartthings shield
Please help me TT

pcoughlin, 2015-09-09:
It looks like the [SmartThings shield](https://shop.smartthings.com/#!/products/smartthings-shield-arduino) can connect on pins 0,1 OR 2,3.
>The Shield connects to the Arduino on pins 0,1 or 2,3 and reserves pin 6. Its features include a breadboard style prototyping area and a controllable RGB LED.

In the [Shield_on_off_hello.ino](https://gist.github.com/aurman/6546221):
``` 
#define PIN_THING_RX    3
#define PIN_THING_TX    2
```

kenan-kajkus, 2015-09-13:
e.g.

byte SEND_PIN = 11;
IRsend irsend(RECV_PIN);

Adminius, 2017-02-10:
@kenan-kajkus i have the simmilar idea. See also #125 
but it not so easy :(
as-is: boarddefs.h has hardcoded tx pins and timers for that pins.
to be: 
1. if sendpin is defined -> find timer for it -> do things as today
2. if sendpin not defined (legacy support) ->do thing as today

something like that:
```
// Arduino Mega
#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
	switch (sendpin):
		case 11:
			define IR_USE_TIMER1
			break;
		case 9:
			define IR_USE_TIMER2
			break;
		case 5:
			define IR_USE_TIMER3
			break;
		case 6:
			define IR_USE_TIMER4
			break;
		case 46:
			define IR_USE_TIMER5
			break;
		default:
#	error "Internal code configuration error, no known IR_USE_TIMER# defined\n"
```
instead of:
```
// Arduino Mega
#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
	//#define IR_USE_TIMER1   // tx = pin 11
	#define IR_USE_TIMER2     // tx = pin 9
	//#define IR_USE_TIMER3   // tx = pin 5
	//#define IR_USE_TIMER4   // tx = pin 6
	//#define IR_USE_TIMER5   // tx = pin 46
```

kenan-kajkus, 2017-02-10:
@Adminius i was referring to the receiv functinality. There it is possible to write this : 
`byte RECV_PIN = 11;
IRrecv irrecv(RECV_PIN);`

Adminius, 2017-02-10:
for receive yes, but not for send...

kenan-kajkus, 2017-02-10:
@Adminius I know. I meant to implement it this way so it would be coherent.

Adminius, 2017-02-10:
theoreticaly we can do this:
```
IRsend::IRsend ()
{
#define STANDARD_TX_PIN
}

IRsend::IRsend (int sendpin)
{
#define TX_PIN sendpin
}
```

and 
```
#elif defined(__AVR_ATmega32U4__)
	#if defined(STANDARD_TX_PIN)
		//#define IR_USE_TIMER1   // tx = pin 14
		//#define IR_USE_TIMER3   // tx = pin 9
		#define IR_USE_TIMER4_HS  // tx = pin 10
	#else
		#if (TX_PIN == 14)
			#define IR_USE_TIMER1
		#elif (TX_PIN == 9)
			#define IR_USE_TIMER3
		#elif (TX_PIN == 10)
			#define IR_USE_TIMER4_HS
		#else
			#error "Internal code configuration error, no known pin defined\n"
		#endif
	#endif
```
but it doesn't work...
i guess header files should be reworked for this changes...

z3t0, 2017-02-10:
This is being closed as it's an advanced use case that the library does not provide functionality for. However looking at the source code should allow an advanced programmer to switch send pins by replacing registers

closed 2017-02-10 by z3t0