Arduino-IRremote/Arduino-IRremote #220: Attiny85 support

lechacal, 2015-09-10:
Hi all,

Thank you for the great library. Has anyone been running the library on Attiny85 before?

I have had my first try but realised it might not have full support yet.

I am also asking some help on the Arduino forum here. http://forum.arduino.cc/index.php?topic=346847

lechacal, 2015-09-11:
For now just a yes/no answer to say whether the library currently supports attiny85 or not would be great.

z3t0, 2015-09-11:
please check #97

z3t0, 2015-09-11:
and #39

madmalkav, 2015-10-02:
I have it running on Attiny85 but the library blocks SoftSerial. I.E. if I have a loop writing a message a serial like "Waiting for IR signal", as soon as an IR signal arrives, the serial outputs stops. IR receiving is working ok as the feedback LED is working.

AnalysIR, 2015-10-02:
Maybe time to upgrade the MCU being used to one with more resources???

madmalkav, 2015-10-02:
It works ok with another IR library so I don't think it is a lack of
resources problem

AnalysIR, 2015-10-02:
The other library(?) may not be using the same peripherals(timers).

madmalkav, 2015-10-02:
Yeah, now that you say, it is probably that.

joshuajnoble, 2015-10-06:
I looked into this and IIRC to get this library working with the timers available on the ATTiny85 would require a pretty substantial rewrite of the library that I haven't had time for, sadly.

cklim01, 2016-09-02:
Can IRremote and Softwareserial use in same sketch on Attiny85?

z3t0, 2016-10-06:
@cklim01 Do you know which timer is used by Software Serial?

cklim01, 2016-10-06:
Sorry @z3t0 , i have no idea about it. i have very limited knowledge about mcu and programming. just started to learn.

PaulStoffregen, 2016-10-06:
SoftwareSerial doesn't use any timers on AVR chips.

But it does block interrupts, so it'll play havoc with IRremote's receive code which depends on a timer interrupt at regular intervals.  It'll also mess up IRremote transmitting if a character arrives and SoftwareSerial takes too much CPU time with its interrupt while IRremote is delaying to generate IR bursts at the correct speeds.

PaulStoffregen, 2016-10-06:
If you're using slow baud rates, my AltSoftSerial library might be an alternative.  But it requires a 16 bit timer, and my understanding is the ATTINY chips usually have only 8 bit timers, so you might be out of luck there too unless you're using an AVR with a 16 bit timer.

cklim01, 2016-10-07:
That too bad. need to use other AVR. Anyway, thanks for your info.

z3t0, 2016-10-07:
> SoftwareSerial doesn't use any timers on AVR chips.
@PaulStoffregen 

Thanks for pointing that out.. considering it has "soft" as a part of the name I should have figured, no idea what I was thinking last night!

@cltnschlosser 
Yes, it seems that chip only has 8-bit timers.

ArminJo, 2020-10-12:
IRremote and [IRMP ](https://github.com/ukw100/IRMP) now supports ATtinies and Digispark.

closed 2020-10-12 by ArminJo