Arduino-IRremote/Arduino-IRremote #221: Cross project portability

jacklol13, 2015-09-13:
Some defines and per-project configurations are defined internally in the library code and therefore when changed, affect all projects that use the library.
Specifically, disabling unnecessary protocols requires digging through the library's internal files.
Naturally, every project requires a different set of protocols which requires editing internal files for every project (Or saving that project and the library somewhere else).

Probably the best way to solve this problem is to further separate protocols to a point where every protocol can be included using a different `include` directive (Thus preventing other protocols code from being included).

z3t0, 2015-09-13:
thats definitely a good idea, at the moment I am only submitting patches and bug fixes to this project as I am very busy with school. However, if any one is interested in submitting a pull request I will look over it and merge it.

AnalysIR, 2015-09-14:
FYI: for people starting out, they probably don't know what protocols they are trying to decode. Therefore any new solution/appraoch, should meet their requirements.

pcoughlin, 2015-09-14:
Yes, to AnalysIR's point, the default should be all things are "enabled".  The tricky part is to have the library and reduce the Memory AND Code requirements.  As was said above, someone can come up with a clever way to do it and it can be implemented.

jacklol13, 2015-09-14:
Every protocol should be pretty much on its own and that way one could include any protocol desired.
An "all" file can be created to simply include all protocols and that file can be included by people starting out that wish to include all protocols.

pcoughlin, 2015-09-15:
Could you create a Pull Request that would embody your proposed changes for review.

That is the next step in this process.

risitt, 2015-10-14:
I think being able to #include needed protocols and not others is key. I have my own IR library that only supports NEC and Sony SIRC via an external interrupt pin. It's tiny (about 30 lines of code) but can't handle Manchester encoding and won't be able to without a rewrite. So I came here to see if I really have to re-invent the wheel...

My IR library is part of a much larger project. Current sketch occupies 56% of available space on an UNO, before swapping my IR library for yours. My project is a how-to so other people will be building it, and for that reason I thought I'd give them the option of a wider array of remote protocols, so that they don't have to go out and buy new remotes. Your library is a promising way to achieve that, but V1 was already huge, and the re-release of V2 is larger still. It probably won't even fit at all.

A compiler won't bother compiling a method/function unless it's used somewhere. But placement in a switch/if that's part of another method that gets used, even if that specific case never evaluates to true, will end up counting. To avoid that you might think about providing methods like irRecv.decodeNEC(), irRecv.decodeSIRC() instead of just irRecv.decode().

An alternate (maybe even better) way is to accept a parameter in your constructor that specifies the protocol to use. Use an enum, include an "ALL" option, and make the "ALL" option the default if a protocol is not explicitly specified. Use an initializer list to save the choice to a private const member variable. When testing constants in an if statement, a compiler should be able to determine which cases will never evaluate to true and will exclude the containing code, including references to other methods. That could be a way to acchieve this result without relying on #include stuff.

So, I'm off to reinvent the wheel but I'll be watching this project - good luck with it! I'm sure you'll find a good way to deal with this.

pcoughlin, 2015-10-14:
Library modules are linked into the final object file at Link time and are pre-compiled.

I have thought about this several times ... if you have a concrete suggestion, then please let us know.

The only way I have done it is to manipulate the code to not include items in the library module.

What ideas do you have?
It would be preferable to put something into the main source that would make the library re-compile.

risitt, 2015-10-14:
I don't quite know your project well enough to say, but based on the current structure of your library, I would guess that going with #includes is the easiest way. It's also easy for newbies to understand, and it's something they can comment out / uncomment easily when they change remotes.

One other possibility is to make a base class that has most/all of the logic except for the protocols, and have a number of derived classes: IRrecvNEC, IRrecvSONY, etc that inherit the base class and then define the protocol. If a class is not instantiated, it won't end up in the final compiled code that gets uploaded to the MCU.

closed 2017-08-22 by z3t0