Arduino-IRremote/Arduino-IRremote #222: 20000 us mark and space

shanehou, 2015-09-24:
Hi, I'm using a Genuino (just like Arduino UNO) to make an air conditioner remote.
This air conditioner has a special protocol, which is very similar to NEC but not exactly the same. The starting parts are both `9000 us` mark and `4500 us` space, but it sends out multiple data sections.
The raw codes I dumped from the original remote is like:

`-31902 8900 -4350 700 -500 650 -550 650 -550 650 -1600 700 -500 650 -550 650 -550 650 -550 650 -1600 650 -550 650 -550 650 -1600 700 -500 650 -550 650 -550 650 -550 650 -500 700 -500 700 -500 650 -550 650 -550 650 -1600 700 -500 650 -550 650 -550 650 -550 650 -500 700 -500 650 -1650 650 -550 650 -1600 650 -550 650 -550 650 -1600 700 -500 700 -18800 (how can I send a space this large?) 650 -550 650 -550 650 -550 650 -500 700 -500 650 -550 650 -550 650 -550 650 -500 700 -500 650 -550 650 -550 650 -550 650 -500 700 -500 700 -500 650 -550 650 -550 650 -500 700 -500 700 -500 650 -550 650 -550 650 -500 700 -500 650 -550 650 -550 650 -550 650 -1600 700 -1600 650 -550 650 -1600 (here is the end of the first part) 700 25636 (and the beginning of the second part) 8900 -4400 650 -500 700 -500 650 -550 650 -1650 650 -500 700 -500 700 -500 650 -550 650 -1650 650 -500 700 -500 650 -1650 650 -550 650 -500 700 -500 700 -500 650 -550 650 -550 650 -500 700 -500 650 -550 650 -1650 650 -500 700 -500 700 -500 650 -550 650 -550 650 -500 700 -1600 650 -1650 650 -1600 700 -500 700 -500 650 -1650 650 -550 650 -18800 (how can I send a space this large?) 700 -500 650 -550 650 -550 650 -550 650 -500 700 -500 650 -550 650 -550 650 -550 650 -500 700 -500 650 -550 650 -550 650 -550 650 -500 700 -500 700 -500 650 -550 650 -550 650 -500 700 -500 700 -500 650 -550 650 -550 650 -500 700 -500 650 -550 650 -550 650 -1600 700 -1600 650 -550 650 -1650 650`

I know it's too long, but basically there are two parts. The first part would send out 36 binary bits (but the last bit has a `-20000` instead of `-550` or `-1650` as the space), and then 32 binary bits. Then the second part would come in, with `700 20000` between the two parts (yes, `20000`, not `-20000`). The second part is almost the same as the first part (only 1 binary bit is different).

The problem is, the board seemed not to send space larger the `3600 us`, even if I set a raw code array variable with `20000` inside.

And, how can I send `700 20000` combination? The `IRSend` module would always take turns to send out mark and space.

AnalysIR, 2015-09-24:
try using the latest version from the 'master'. Many of these issues have been fixed.

Use sendRAW for this signal.

shanehou, 2015-09-24:
Okay I'm give it try (I'm using experimental branch).

And another thing: because I need to send large data sets, I changed some definitions in `IRremoteInt.h` and `IRremote.h`, mainly `#define _GAP 5000` to `#define _GAP 50000`, `uint8_t rawlen;` to `unsigned int rawlen;`, and `#define RAWBUF 100` to `#define RAWBUF 511`. I noticed the `_GAP` macro would affect `GAP_TICKS`, and would affect the results of some comparisons with `irparams.timer`, so I don't know if it's okay.

shanehou, 2015-09-24:
I've tried to switch the branch to `master` and change some definitions as I said above, but things didn't change.

I noticed executing a `delay()` function will generate a space (negative number on dumped data), so I changed the way I sent data, making the `20000 us` space in each part be generated by `delay()` function. But still, I'm unable to send a `700 20000` combination. And I'm curious how that combination come, because if 700 and 20000 are all high level, how can the program identify them?

AnalysIR, 2015-09-24:
I suspect you are somehow not using the latest master...?...as it can now handle sending 20000uSecs spaces.!

pcoughlin, 2015-11-12:
Please look in your Library file named irSend.cpp:
  Please paste line 74 of that file here.

closed 2020-10-12 by ArminJo