Arduino-IRremote/Arduino-IRremote #224: IRSend pin on Arduino Yun

karkowg, 2015-09-30:
Hello,

I'm having trouble to send IR signals on the Arduino Yun. I learned that Yun has the same processor from Leonardo, and that Leonardo uses pin 13 for send by default. But the Yun has a built-in led on pin 13 and I don't know for sure what to do to use the IR transmitter in another pin or to disable the L13.

If anyone could help me I would appreciate it.

AnalysIR, 2015-09-30:
try pins 10, 9, or 14 for the pins....you can also change the code blow as required

```   
 #elif defined(__AVR_ATmega32U4__)

	//#define IR_USE_TIMER1   // tx = pin 14

	//#define IR_USE_TIMER3   // tx = pin 9

	#define IR_USE_TIMER4_HS  // tx = pin 10 ```

karkowg, 2015-09-30:
If I set IR_USE_TIMER4_HS, I should wire the IR LED in digital pin 10 or in the digital pin equivalent to the value of TIMER_PWM_PIN defined in #elif defined(IR_USE_TIMER4_HS), which in this case is 13?

This is my first time with the Arduino and I don't know much.

AnalysIR, 2015-09-30:
Arduino digital pin 10 etc

karkowg, 2015-09-30:
Thanks, I will test these pins.
Btw, has anyone succeeded sending IR signals with a Yun?

AnalysIR, 2015-09-30:
As you mentioned it is essentiallly the same as a Leonardo - so yes.

karkowg, 2015-10-01:
I tried pin 10 with two different IR LEDs but nothing happened. Pin 9 uses timer3, that doesn't have a **if defined(\__AVR_ATmega32U4\__)**, so I didn't know what to set as TIMER_PWM_PIN.
I even used the IRLib bit-bang with other pins, but the max I could get was a visible blink in one of the IR LEDs, which I think is not supposed to happen.

q2dg, 2015-10-01:
Could you add a simple table on README.md listing the suitable pins of different boards to send IR signals? I mean...

328-based: Pin ?
2560-based: Pin ?
32U4-based: Pin ?
Due: Pin ?
Zero: Pin ?

Thanks!!

AnalysIR, 2015-10-02:
@karkowg 
As a workaround, just set the pinmode to output and write it as LOW on the pin you are using after all the setup and before the initial send. Essentially achieves the same as....

````
pinMode(TIMER_PWM_PIN, OUTPUT);
	digitalWrite(TIMER_PWM_PIN, LOW); // When not sending PWM, we want it low
````

The source doesn't seem 100% to me, it looks as if it is set up correctly for Teensy but no obvious mention of leonardo/yun. I wonder was it inadvertently remove recently? @z3t0

z3t0, 2015-10-11:
Hi, I don't remember making any changes in regards to the Yun/Teensy... will take a look through my commits. Sorry for not replying for such a long time, if possible please send me an email if my attention is needed as I am going through a very busy time in life and won't be checking the issues page for the meantime - i will definitely check my email though.

z3t0, 2016-02-21:
@q2dg That is a great idea!

z3t0, 2016-02-21:
Closing for inactivity, will follow up with table in README for clarification

closed 2016-02-21 by z3t0

q2dg, 2016-02-21:
Oh, thanks a lot!!

q2dg, 2016-02-21:
@z3t0 Sorry, but...table in README doesn't show used pins in 32U4-based boards (Leonardo, Yún) and ARM-based boards (Due, Zero, etc). Thanks, anyway

z3t0, 2016-02-21:
@q2dg At the moment I don't think we have support for ARM based boards, I will check for 32U4 boards

AnalysIR, 2016-02-21:
@z3t0 32u4 is Lenoardo/Yún 8-bit AVR and they should be functionall equivalent to the Teensy2 ???

z3t0, 2016-02-21:
@AnalysIR Yes, although I am not sure which pin that maps out to on the Lenoardo?

AnalysIR, 2016-02-21:
The default setting, appears to be 
Timer4/OC4A => Arduino pin 13 for Leonardo.

If timer 1 is configured
Timer1/OC1A => Arduino pin 9 for Leonardo.

If timer 3 is configured
Timer3/OC3A => Arduino pin 5 for Leonardo.

The way it is set up for Teensy2/Leonardo may not be 100% clean and may need some rework.

z3t0, 2016-07-11:
#334 has the changes, sorry for the delay :smile: