Arduino-IRremote/Arduino-IRremote #229: Please add your library to Arduino Library Manager

ffissore, 2015-10-29:
Please consult https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ and submit request for inclusion into Arduino Library Manager

pcoughlin, 2015-11-07:
Good idea.  There seem to be several files that are required for inclusion, with some specific formatted information ("library.properties" for one).

If anyone would like to write those additional files, then it will speed up this process.

per1234, 2015-11-07:
It looks like it's all ready to go as is. The library.properties file already exists and is correctly formatted. The FAQ linked by ffissore above makes it sound like the source files might need to be moved to a src subfolder but from reading this: https://github.com/FastLED/FastLED/issues/159#issuecomment-98837773 it turns out to be optional. So someone just needs to submit an issue to https://github.com/arduino/Arduino/issues requesting the library to be added to Library Manager. I'd be happy to do so, just give me the go ahead.

pcoughlin, 2015-11-07:
That i OK with me.  Please post the info you receive back and if there is some way to check the status (a link, I suppose), then please post that as well.

If you get into it and there are some sorts of encryption keys or something, then please let me know and maybe I might need to do it.

Other than that ... "Go Ahead"

per1234, 2015-11-07:
Ok, issue has been submitted: https://github.com/arduino/Arduino/issues/4108 I'll let you know when it is added or if there are any issues.

pcoughlin, 2015-11-08:
Well done.

per1234, 2015-11-11:
IRremote has been added to the Arduino IDE Library Manager and I have successfully installed it.

pcoughlin, 2015-11-12:
Very good @per1234 
This is a wonderful step.

closed 2015-11-12 by pcoughlin