Arduino-IRremote/Arduino-IRremote #230: Project Transfer

z3t0, 2015-11-01:
Hi,

As you may have realized I have become very busy lately and so will be unable to maintain this project in the foreseeable future, would anyone be interested in maintaining this project? Thanks and I apologize for the inconvenience

shirriff, 2015-11-03:
Rafi, thanks for your hard work on the project! You accomplished so much on it.

z3t0, 2015-11-04:
@shirriff No problem at all! I really enjoyed working on this and am quite sad to have to take some time off as I know that I wont be able to give this project the time it deserves for the foreseeable future.

Thanks to all the contributors who worked so hard!!!

z3t0, 2015-11-07:
@pcoughlin has volunteered to look over PR's and manage the project so I have added him as a collaborator. I know he will do an excellent job! Thanks Paul!

closed 2015-11-07 by z3t0

shirriff, 2015-11-12:
Welcome, Paul! Glad to have you taking over the project.

pcoughlin, 2015-11-12:
Thanks, Glad to help.  To paraphrase Tom Bodett ... 
I'll leave the light on for you. The IR light in this case!