Arduino-IRremote/Arduino-IRremote #231: Unused variable

ram-0000, 2015-11-03:
in file ir_Aiwa.cpp, line 31, in function IRsend::sendAiwaRCT501(), the local variable mask is not used.

...\Arduino\libraries\IRremote\ir_Aiwa.cpp: In member function 'void IRsend::sendAiwaRCT501(int)':
C:\Users\ng50ce6\Documents\Arduino\libraries\IRremote\ir_Aiwa.cpp:31:17: warning: unused variable 'mask' [-Wunused-variable]
  int            mask;
                 ^

pcoughlin, 2015-11-07:
Would you like to create a code update and create a Pull Request?

ram-0000, 2015-11-08:
Hi Paul

I'm totally newbie with GitHub (but not with C++ code)

Would you like to help me and give me indications to propose a code update ?

Kind regards
Raymond

pcoughlin, 2015-11-08:
Certainly, that is the way I started with this.  It is actually quite straight forward.

1. You should fork the code to your account.
2. Make the changes you want put into this Repo (and yours).
3. On your Repo page, click on the Pull Request (top right-side menu item)
4 Complete the Pull Request process and your changes will be proposed for inclusion in the z3t0/Arduino-IRremote repo.
5. I will review your changes, make comments, etc.
6. Then it can be merged into this Repo.

IF you need any more info, then just send me an email and I will help you via email.

Good way to get started!!!

ram-0000, 2015-11-08:
should be done in the way you described

Thanks for help

z3t0, 2015-11-10:
Merged #233 fixed this?

closed 2015-11-10 by z3t0

pcoughlin, 2015-11-12:
The point of changing the typo was NOT to get it done ... it was to help
ram-0000 learn how to make changes and the process.

I could have made the typo change and merged it myself after I saw it.

I have a purpose to the madness.

z3t0, 2015-11-12:
@pcoughlin Oh, in that case sorry about that, just saw an opportunity to help and jumped in :D