Arduino-IRremote/Arduino-IRremote #232: signed or unsigned value

ram-0000, 2015-11-03:
In file IRrecv.cpp, my compiler is warning

...\Arduino\libraries\IRremote\irRecv.cpp:197:2: warning: this decimal constant is unsigned only in ISO C90 [enabled by default]
  long  hash = FNV_BASIS_32;
  ^

Forcing FNV_BASIS_32 to unsigned seems to fix the problem

line 193 : #define FNV_BASIS_32 2166136261u

z3t0, 2015-11-10:
Which version of the IDE?

ram-0000, 2015-11-11:
Arduino 1.6.5 (the last one I suppose)

ram-0000, 2015-11-11:
In Menu File / Settings, the compiler warnings level is set to "All" (which is not the default setting)

z3t0, 2016-02-17:
I dont think it needs to be changed, its simply warning that it would not be unsigned if we were not using ISO C90:

https://github.com/sudar/Arduino-Makefile/issues/376
second post

ram-0000, 2016-02-17:
As you wih

But I hate warning that may hide potential bugs or other important error message (difficult to see an error line among thousand lines or warning)

z3t0, 2016-02-19:
Hmm I see, I'll reopen this as a reminder to look for something to make the warning disappear (if that is possble)

z3t0, 2016-02-19:
Oh, it seems the solution is provided in your OP

ram-0000, 2016-02-19:
Yes, a solution is provided free of charge !!!

ArminJo, 2020-07-03:
Can not reproduce the warning anymore.

closed 2020-07-03 by ArminJo