Arduino-IRremote/Arduino-IRremote #235: disable some receiver

fireport, 2015-11-13:
For timing reasons i need to disable same decoder, but when i set the corispondent define to 0 i receive a compiler error.
For example, if i set in  IRemote.h
#define DECODE_RC5           0
the compile fails with followinge error

irRecv.cpp:40:23: error: 'decodeRC5' was not declared in this scope
  if (decodeRC5(results))  return true ;

MatrixMike, 2015-11-13:
Hope this is not just a typo - UPPER case - lower case type of thing...

z3t0, 2015-11-28:
Please post gist with modification

AnalysIR, 2015-11-28:
I just checked with the latest Master by changing the setting of
 #define DECODE_RC5 0

..and found no plroblem with compiling.

fireport, 2015-11-29:
Sorry, I tried with latest version and again I have the same error
//------------------------------------------------------------------------------
// Supported IR protocols
// Each protocol you include costs memory and, during decode, costs time
// Disable (set to 0) all the protocols you do not need/want!
//
#define DECODE_RC5           0
#define SEND_RC5             0

C:\Users\b487000\Dropbox\arduino-1.6.4\libraries\IRremote\irRecv.cpp: In member function 'int IRrecv::decode(decode_results*)':
C:\Users\b487000\Dropbox\arduino-1.6.4\libraries\IRremote\irRecv.cpp:40:23: error: 'decodeRC5' was not declared in this scope
  if (decodeRC5(results))  return true ;
                       ^
Errore durante la compilazione

AnalysIR, 2015-11-29:
try out this sketch we just wrote and post the output
http://www.analysir.com/blog/2015/11/28/helper-utility-for-troubleshooting-irremote/

fireport, 2015-12-01:
Here the sketch output
IRremoteInfo - by AnalysIR (http://www.AnalysIR.com/)
             - A helper sketch to assist in troubleshooting issues with the
library by reviewing the settings within the IRremote library
             - Prints out the important settings within the library, which
can be configured to suit the many supported platforms
             - When seeking on-line support, please post or upload the
output of this sketch, where appropriate

IRremote Library Settings
=========================
RAWBUF: 101
Timer defined for use: Timer2
IR Tx Pin: 3
MCU Clock: 16000000
MCU Platform: ATmega328(P) / (Duemilanove, Diecimila, LilyPad, Mini, Micro,
Fio, Nano, etc)
Mark Excess: 100 uSecs
Microseconds per tick: 50 uSecs
Measurement tolerance: 25%
Minimum Gap between IR Signals: 5000 uSecs
Arduino IDE version: 1.6.4
Debug Mode: OFF (Normal)

IR PROTOCOLS  SEND     DECODE
============= ======== ========
RC5:          Disabled Enabled
RC6:          Disabled Enabled
NEC:          Enabled  Enabled
SONY:         Disabled Enabled
PANASONIC:    Enabled  Enabled
JVC:          Enabled  Enabled
SAMSUNG:      Enabled  Enabled
WHYNTER:      Enabled  Enabled
AIWA_RC_T501: Enabled  Enabled
LG:           Disabled Enabled
SANYO:        Disabled Enabled
MITSUBISHI:   Disabled Enabled
DISH:         Enabled  Disabled
SHARP:        Enabled  Disabled
DENON:        Enabled  Enabled
PRONTO:       Enabled  (Not Applicable)

Notes:
     - Most of the seetings above can be configured in the following files
included as part of the library
     - IRremteInt.h
     - IRremote.h
     - You can save SRAM by disabling the Decode or Send features for any
protocol (Near the top of IRremoteInt.h)
     - Some Timer conflicts, with other libraries, can be easily resolved
by configuring a differnt Timer for your platform

AnalysIR, 2015-12-01:
OK thanks for that - the output looks standard & the same as mine except I am running on IDE 1.6.5

Given that is the only obvious difference, please try updating your IDE to 1.6.5 (I haven't tried any later version yet)

AnalysIR, 2015-12-01:
You could also enable verbose mode in the IDE preferences (File menu) and see if it gives any more clues.

AnalysIR, 2015-12-01:
Sorry...correction of post 2 above...

RC5: Disabled Enabled 
RC6: Disabled Enabled

of course they are both showinf as disabled....can you post the contents of IRremoteInt.h
the following lines
````
#define DECODE_RC5           1
#define SEND_RC5             1

#define DECODE_RC6           1
#define SEND_RC6             1

AnalysIR, 2015-12-01:
looking back at your original post and an early comment.
````
irRecv.cpp:40:23: error: 'decodeRC5' was not declared in this scope
if (decodeRC5(results)) return true ;
````
it looks indeed very like a lower/upper case issue & you may have an older version of IRremote from when a lot of changes were being made.

So my suggestion is to download the library again - current Master version and make sure to restart the IDE after installing IRremote again from github.

z3t0, 2016-02-17:
Closed for inactivity, assuming the problem was solved but there is no confirmation provided?

closed 2016-02-17 by z3t0