Arduino-IRremote/Arduino-IRremote #236: Class Error on Multiple IRremote by Khair-Edinne

jrgess, 2015-11-20:
Hi!

I'm having some problem to upload my sketch (Arduino Mega on IDE 1.6.5) with the following code:

#include <IRremote.h>

IRsend  irsend0;
IRsend1 irsend1;
IRsend2 irsend2;

void setup()
{
  Serial.begin(9600);
}

void loop() {
  if (Serial.read() != -1) {
    for (int i = 0; i < 3; i++) {
      irsend0.sendSony(0xa90, 12); // pin 3
      irsend1.sendSony(0xa90, 12); // pin 10
      irsend2.sendSony(0xa90, 12); // pin 9
    }
  }
}

The error I'm getting is:

'class IRsend' has no member named 'sendSony'

I need help to solve this.
Thank's

pcoughlin, 2015-11-21:
Is the first line "#include" 
or "#include <IRremote.h>"

You can also use the markdown of   \`\`\`C  at top and  \`\`\`  at the bottom of code to make it more readable.

Like this:
```C
#include

IRsend irsend0;
IRsend1 irsend1;
IRsend2 irsend2;

void setup()
{
   Serial.begin(9600);
}

void loop() {
   if (Serial.read() != -1) {
      for (int i = 0; i < 3; i++) {
         irsend0.sendSony(0xa90, 12); // pin 3
         irsend1.sendSony(0xa90, 12); // pin 10
         irsend2.sendSony(0xa90, 12); // pin 9
      }
   }
}
```

Also, the IRremote library does not currently accommodate the three instances like you put it in the code.
Please start with #128 and read through the posts in that chain.

Why do you want to use 3 IR Send pins?

jrgess, 2015-11-22:
Hi!

Thanks for reply and sorry for incomplete code. The original code is bellow:

#include <IRremote.h>

IRsend irsend0;
IRsend1 irsend1;
IRsend2 irsend2;

void setup()
{
    Serial.begin(9600);
}

void loop() {
    if (Serial.read() != -1)  {
        for (int i = 0; i < 3; i++)  {
             irsend0.sendSony(0xa90, 12); // pin 3
             irsend1.sendSony(0xa90, 12); // pin 10
             irsend2.sendSony(0xa90, 12); // pin 9
         }
    }
}

The are no spaces after the #include <IRremote.h>
I only have overwriten the files "IRremote.cpp", "IRremote.h" and "IRremoteInt.h" from original Ken Shirriff library.

When I try to upload my sketch, I have a warning message too:
WARNING: Category 'infrared' in library IRremote is not valid. Setting to 'Uncategorized'

I need multiple IR Emitters cause I'm developing a home control project which will send commands to some devices, like TV screens and other devices individually.

What am I missing?

pcoughlin, 2015-11-22:
Did you read the issue threads starting with #128 and all the attached other threads?

The current library does NOT allow for multiple output pins.  It is a code modification that is intended, but not implemented yet.

If you are able to use the code in those Issue posts, then you should be able to get it to work.

Also, if the three devices are all Sony, then would it be possible to have three IR LEDs and attach them all to one output pin? you might need some circuitry to give them more power than the standard MCU output pin will provide.

@AnalysIR do you have any advice for jrgress?

AnalysIR, 2015-11-22:
@pcoughlin I think you covered it all above, based on the info provided.

Increasing current to the IR LEDs or having multiple IR LEDs in the same circuit or using IR LEDs with different beam angles is a good way to get 'flood fill' of a room/area.
Look at the constant current circuit on our blog:
http://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/

With a slight circuite redesign you could get even more IR LEDs into the same circuit & even if the devices are different it shouldn't matter as they will ignore unknown signals.

jrgess, 2015-11-23:
Hi, guys!

Thanks again.

I have read the post #128 and found a referenced post #184, that have a similar problem.
I'm not so familiar with C++, but from what I understood, I need to change some variables to make it work. 

I know that exists a hardware solution for my problem, but I'm interested in a software solution. In the original Ken Shirriff library, I was emiting and receiving signals really fine. The problem is controlling devices of same model, otherwise, like AnalysIR said, they will simply ignore unknown signals.

I commented the "ifdef" variables on "IRremote.cpp" and "IRremote.h" and now I'm getting the following errors:

undefined reference to `IRsend1::sendSony(unsigned long, int)'
undefined reference to `IRsend2::sendSony(unsigned long, int)'

What am I still missing?

pcoughlin, 2015-12-10:
There is no IRsend1 or IRsend2 in the code.
You want to send multiple emitters, but the code is not set up that way.  The #128 discusses HOW TO DO IT, not that it was put into the permanent library code.

You can adjust your code the same way and get the result you want, but that involves coding the changes.

Hope this helps.

z3t0, 2016-02-17:
Closing for inactivity

closed 2016-02-17 by z3t0