Arduino-IRremote/Arduino-IRremote #237: Improve RC6 reception by changing MARK_EXCESS value

Bekkie, 2015-11-20:
Hi, 

I was experimenting with your library and i found out that the RC6 reception was quite poor with my Arduino Nano (1 out of 10 correct decodes). But by cutting the MARK_EXCESS in half so from 100 to 50 (only for RC5/6 protocols), the reception improved to 19 out of 20 correct decodes.

It even decodes the RC6A protocol now. (36 bits i.s.o. 20)

So i added these lines in  IRremoteInt.h (after line 122)
// RC-5 and RC-6 are more reliable with an MARK_EXCESS value of 50
#define MARK_EXCESS_RCx 50

And change line 23 of file ir_RC5_RC6.cpp to this:
correction = (val == MARK) ? MARK_EXCESS_RCx : - MARK_EXCESS_RCx;

AnalysIR, 2015-11-21:
The problem could also be related to your IR receiver, please post the model number or link.

Bekkie, 2015-11-21:
The receiver model number. is HX1838
http://www.dx.com/nl/p/hx1838-ir-infrared-receiver-module-152705

AnalysIR, 2015-11-22:
Sorry to inform you but it looks like one of these
see http://www.analysir.com/blog/2014/12/08/infrared-receiver-showdown-tsop34438-vs-vs1838b-winner-revealed/
....one of the cheapest & worst IR receivers availabe. The VS1838B IR receiver on its own sells for <$0.2 for a good reason.

Also, it is designed for 38kHz not  36kHz used for RC5/6, which might also explain the issues you saw or make them worse.

I suggest getting any TSOPxx36 receiver instead of changing the IRremote code.

The following may also be of interest:
http://www.analysir.com/blog/2014/03/27/infrared-receivers-signal-lag/

Hoving said all that, the MARK_EXCESS option may work for you, as you described.

closed 2020-10-12 by ArminJo