Arduino-IRremote/Arduino-IRremote #238: Lightweight version

stefa168, 2015-11-21:
Hi, I used this library some time ago and it occupied not too much space on an UNO.
Now it weights about 10K when compiled and it is a lot of space for a single library, compared to the usual flash memory of an Arduino (32K).
By removing all the unnecessary protocols, I can get the size down to about 5K. Still, it is a lot more of the older versions (~3K).
I am working on a project which involves IR communication and this library would be extremely useful. The only problem is the very big size.
Is there any way to bring the occupied space down some more?

pcoughlin, 2015-11-21:
This library is intended to give the widest range of people help without them doing much coding to get them started using IR.  Once the user becomes more familiar with the Library (like you have), then it is a good idea to remove all the parts that do not apply specifically to your project.

Here are some suggestions:
1. Determine the longest IR code that you want to handle.  Then change the buffer size on line 43 of:
github.com/z3t0/Arduino-IRremote/blob/master/IRremoteInt.h

2. Remove Decode if it is not being used. (Send only) There are multiple files to review for parts of the Decode code.  I would comment them out, not delete the lines.
2a.  If not using Decode, then Comment out MATCH(), MATCH_MARK(), and MATCH_SPACE() in IRremote.cpp.
3. Remove Send, if it is not being used. (Decode, Receive IR Only) There are multiple files to review for parts of the Decode code.  I would comment them out, not delete the lines.
4. Confirm that IRremote.h, line 124 is set to 0 (zero) to remove DEBUG code
5. In IRremote.h, review the section that starts on line 27.  Change all the "DECODE_..." and "SEND_..." defines to 0 (zero) to turn off those parts of the code.

Please let us know what you decided to take out and leave in and the resulting memory footprint.  It will be interesting.

Hope this helps get your started.

stefa168, 2015-11-21:
Hi, thanks for the answer.
The first test I made is without changing anything:
`Sketch uses 9,656 bytes (29%) of program storage space. Maximum is 32,256 bytes.

Global variables use 429 bytes (20%) of dynamic memory, leaving 1,619 bytes for local variables. Maximum is 2,048 bytes.`

stefa168, 2015-11-22:
Whoops, I must have closed the issue.

I have to add that in an older version I also created a new protocol for my application and all the "nodes" will have to both send and receive messages.

 * If I go in **IRRemote.h** and I disable all the IR protocols except for LG's, but I keep the **DEBUG** option enabled:
`Sketch uses 5,958 bytes (18%) of program storage space. Maximum is 32,256 bytes.`
`Global variables use 499 bytes (24%) of dynamic memory, leaving 1,549 bytes for local variables. Maximum is 2,048 bytes.`

 * Same options of before, but this time without **DEBUG** enabled:
`Sketch uses 5,316 bytes (16%) of program storage space. Maximum is 32,256 bytes.`
`Global variables use 429 bytes (20%) of dynamic memory, leaving 1,619 bytes for local variables. Maximum is 2,048 bytes.`

All I really need is to work with one specific protocol. The best analyzer that I found was this library, and in fact it works extremely well.

If it was more lightweight, it could fit also in applications whereyou need to save space for program instructions...

pcoughlin, 2015-11-22:
If you see any "wasted code" or variables that are bigger than needed, please let us know. I am sure there are possibilities that have not been improved yet.

Also, what was the RAWBUF setting?  Reducing that number is a 2:1 reduction in the SRAM memory usage. (unsigned int is 2 bytes).

So if the MAX length of your IR Codes is say 37, then you could save 124 SRAM (101-(37+1)=62).

Nice table for your data.  This post should be helpful to others as well.

stefa168, 2015-11-22:
I didn't edit the RAWBUF setting, so it still was at 101.
So, from what I understand, that setting indicates how big the array for the duration of the impulses received is, right?

Also, by googling I found this library (https://github.com/NicoHood/IRLremote) which has a footprint of about 2.5K. It seems to work similarly to this library.

AnalysIR, 2015-11-22:
@stefa168 A simple solution would be to strip the IRremote code down to the bare essentials to meet your requirements.
Note: that a lot of flash/SRAM usage can be attributed to the Arduino overhead.
- If you don't use Serial then you will save some flash & SRAM
- using Serial.write instead of Serial.println can also save resources
- etc etc via google

Considering the features you get, a 5K/429B footprint is pretty good & there are other AVRs that have more resources built in. (you can get mega2560s for less than $10 on Aliexpress, including postage)

 ...probably worth doing a google search on how to use/configure AVRs efficiently (Atmel have a good application note on this)

stefa168, 2015-11-22:
@AnalysIR Yup, that's what I'm doing.
What have been the major changes from the original library? I still have the old library that is signed with version 0.11 from 2009..

z3t0, 2015-11-27:
We could work on a more optimized approach, to allow the user to have more control over memory usage. I am currently coming back to this project as I have a lot more time and will make memory usage my main focus.

stefa168, 2015-11-27:
Surely, one first way to optimize the code would be to have just one/two user defined protocols.
I see that this library is often used as a base for receiving just from one protocol, as it works very well and is reliable.

AnalysIR, 2015-11-27:
As a general comment:
Most users starting out, will not know which protocol they are working with & once they have things up and running may want to reduce the number to one or two protocols to save SRAM/Flash and integrate into a bigger or more complex sketch. So the defualt should be many protocols, with an option to trim it down later (Just my opinion)

I guess the real challenge is to have an efficient way to reduce the number of protocols, combined with overall efficient use of resources when many protocols are defined.

stefa168, 2015-11-27:
Yeah, that for sure is a challenge.

z3t0, 2015-11-27:
Hmm, @AnalysIR definitely something I did not think about

vovokrall, 2015-12-08:
Hi, thanks for the tips to save SRAM. I was able to save more than 200 bytes of SRAM by making BUFFER smaller  - couldn't see no difference on SRAM use by desabling all protocols but NEC. Can I do anything else to make IRRemote library use even less SRAM?

stefa168, 2015-12-09:
@vovokrall
I still haven't read all the code, but I feel that it isn't possible to free more SRAM, since there aren't a lot of things that affect the footprint.
For sure, by removing all the unnecessary protocols, you'll free FLASH memory.

bengtmartensson, 2015-12-09:
@vovokrall:
May I ask; do you _need_ smaller size, or is it "just" an intellectual exercise? (Not that I have anything against intellectual exercises :-).) 

If you really need to save all the space you can, you should consider ditching the library concept all together, just copying the source code into the main file, then removing everything not needed. The compiler can do more optimizations if all code resides in one file, than if the source is spread over multiple files.

If you only consider NEC signals, 68 should be the optimal setting for the buffer size.

vovokrall, 2015-12-09:
My sketch is using 93% of flash and 77% of SRAM. It's so big cause I'm keeping 15 air conditioner commands on FLASH (each of them are an unsigned int with 199 positions) and an unsigned int variable with 199 positions on SRAM to manipulate those commands. I'm using NEC to send commands to a LED lamp. My buffer length is 199 and I disabled all protocols but "send NEC".

I'm using Arduino as an Ethernet server (socket TCP) that sends readings from DHT11 and LDR to a connected client, read and execute requests from client (developed on QT) and print all those variables into Serial. It's doing all that with only 459 bytes available, so I guess that Arduino is on it's limit. I'm thinking of using LM35 to read temperatures, but I'll need to still use DHT11 to read humidity. I don't know if I do that I'll have problems with SRAM, but I'll test it.

PS: I'm using ENC28J60 to connect my Arduino to Ethernet, and it uses a lot of SRAM. If I use the W5100 Ethernet Shield, I would save at least 15% of RAM, but the whole idea is to make a cheap home automation.

PS2: sorry for bad english.

bengtmartensson, 2015-12-11:
Hmm, you/we cannot send a "signal" from PROGMEM, but have to copy it to RAM first? There "should" be a possibility to send directly for the timing data residing in PROGMEM. That would save you 199 bytes of SRAM.

OT: Chinese clones of Mega2560 as well as of W5100 Ethernet shields can be had at Ebay for one-digit €/$ prices. So investing in better hardware may be an alternative.

vovokrall, 2015-12-11:
I don't know. Is it possible? I have no idea how to do it.

And actually this would save me 398 bytes of SRAM. :/

bengtmartensson, 2015-12-11:
What is needed is a function `IRsend::sendRaw_P(PGM_VOID_P buf,  unsigned int len,  unsigned int hz)` that sends an unsigned int[] in PROGMEM. I tried a bit: the following sketch
```
#include <IRremote.h>

IRsend irsend;

static const unsigned char necFrequency = 38U;
static const unsigned int array[] = {
   9024, 4512, 564, 564, 564, 1692, 564, 564, 564, 1692, 564, 1692,
   564, 1692, 564, 1692, 564, 564, 564, 1692, 564, 564, 564, 1692,
   564, 564, 564, 564, 564, 564, 564, 564, 564, 1692, 564, 1692, 564,
   564, 564, 1692, 564, 1692, 564, 1692, 564, 564, 564, 564, 564, 564,
   564, 564, 564, 1692, 564, 564, 564, 564, 564, 564, 564, 1692, 564,
   1692, 564, 1692, 564, 39756
};

void setup() {
}

void loop() {
    irsend.sendRaw(const_cast<unsigned int*>(array), (unsigned char) (sizeof(array) / sizeof(unsigned int)), necFrequency);
    delay(5000);
}
```
sends a (raw) NEC1 122/29 (Yamaha AVR power on). Takes 2116 flash and 356 bytes ram. Instead, using flash,
```
#include <IRremote.h>

IRsend irsend;

static const unsigned char necFrequency = 38U;
static const unsigned int array[] PROGMEM = {
   9024, 4512, 564, 564, 564, 1692, 564, 564, 564, 1692, 564, 1692,
   564, 1692, 564, 1692, 564, 564, 564, 1692, 564, 564, 564, 1692,
   564, 564, 564, 564, 564, 564, 564, 564, 564, 1692, 564, 1692, 564,
   564, 564, 1692, 564, 1692, 564, 1692, 564, 564, 564, 564, 564, 564,
   564, 564, 564, 1692, 564, 564, 564, 564, 564, 564, 564, 1692, 564,
   1692, 564, 1692, 564, 39756
};

void setup() {
}

void loop() {
    irsend.sendRaw_P(array, (unsigned char) (sizeof(array) / sizeof(unsigned int)), necFrequency);
    delay(5000);
}
```
remains declaring `sendRaw_P` in IRremote.h

    void  sendRaw_P    		(PGM_VOID_P buf,  unsigned int len,  unsigned int hz) ;

and defining in irSend.cpp:
```
void  IRsend::sendRaw_P (PGM_VOID_P buf,  unsigned int len,  unsigned int hz)
{
	// Set IR carrier frequency
	enableIROut(hz);

	for (unsigned int i = 0;  i < len;  i++) {
	  uint16_t duration = pgm_read_word_near(buf + sizeof(uint16_t)*i);
	  if (i & 1)  space(duration) ;
	  else        mark (duration) ;
	}

	space(0);  // Always end with the LED off
}
```
Now, this uses 2096 bytes flash and 220 bytes of RAM.

Might be worth considering to include IRsend::sendRaw_P in the code.

vovokrall, 2015-12-14:
I didn't understood most of it, but I'm gonna try it. lol

Did you tested this code?

bengtmartensson, 2015-12-14:
The code is tested and runs.

What happens in the first sketch, and probably in your sketch too, is that, before setup() is called, the array(s) are copied from flash to ram, in order to use them. In the second sketch, they are not copied, but sent directly from flash. Therefore, the first one takes not only more ram, but also more flash. See https://www.arduino.cc/en/Reference/PROGMEM .

vovokrall, 2015-12-14:
I ran a test with my LED lamp and it works! Tomorrow I will test the air conditioner, but I imagine it will work too.

I used your functions on my project and my SRAM use went from 77% to 58%. Now I have 855 bytes free to use on my local variables!!!!

THANK YOU SO MUCH!!!!

requeijaum, 2017-12-27:
I'm trying all those alternatives.

@bengtmartensson 's code is killer! Gonna test it now.
My program went from 80% usage of dynamic mem. to only 41%.

commit b6a9ddad (2020-07-04): Added function sendRaw_P() for sending data from flash, closes #238
commit 8abc69da (2020-07-04): Added function sendRaw_P() for sending data from flash, closes #238
closed 2020-07-04 by ArminJo