Arduino-IRremote/Arduino-IRremote #239: Help RAW with Arduino Mega for air conditioning

carlosD2884, 2015-11-27:
goodnight beforehand I apologize for my English because it is very bad

I am doing my job as part of the degree , I control the air conditioning right through the Arduino using an IR LED

I have an Arduino MEGA
I'm using the pin 12 for pin reception
and modify the library to use the pin 11 for the IR LED
and modify the RAWPUF 250

I was able to receive and send nec encoding TV and air conditioning
I have a problem with an air whose encoding is Samsung with RAWPUF [100 ]

by pressing B modified frequency from 36kHz  to 40 KHz
C key to modify the length of the chain from take two elements to add two elements

with the keys 1,2,3 sending the previously stored frame with the first value to be deleted without the first value and not the start bit of the code SAMSUNG

attached the code I'm using , if they have any recommendations would appreciate it immensely

#include <Wire.h> 
#include <LiquidCrystal_I2C.h>
#include <Keypad.h>
#include <IRremote.h>

LiquidCrystal_I2C lcd(0x27,20,4);  // set the LCD address to 0x27 for a 16 chars and 2 line display
const byte ROWS = 4;
const byte COLS = 4;
char hexaKeys[ROWS][COLS] = {
  {'1','2','3','A'},
  {'4','5','6','B'},
  {'7','8','9','C'},
  {'*','0','#','D'}};  
byte rowPins[ROWS] = {22, 24, 26, 28};
byte colPins[COLS] = {23, 25, 27, 29}; 
Keypad customKeypad = Keypad( makeKeymap(hexaKeys), rowPins, colPins, ROWS, COLS );  

int RECV_PIN = 12;
IRrecv irrecv(RECV_PIN);
IRsend irsend;

decode_results results;
int khz = 38; // 38kHz carrier frequency for the NEC protocol
unsigned int irSignal1[] = {44418,4400,4400,500,1650,500,600,500,1650,500,1700,500,550,500,600,500,1650,500,600,500,600,500,1650,500,600,500,600,500,1650,500,1650,500,600,500,1650,500,1650,500,600,500,1650,500,1700,500,1650,500,1650,500,1700,500,1650,500,600,500,1650,500,600,500,550,500,600,500,600,500,600,500,550,500,600,500,600,500,550,550,550,500,600,500,600,500,550,500,600,500,1650,500,1700,500,1650,500,1650,500,1700,500,1650,500,1650,500,1650,500};
unsigned int irSignal2[] = {4400,4400,500,1650,500,600,500,1650,500,1700,500,550,500,600,500,1650,500,600,500,600,500,1650,500,600,500,600,500,1650,500,1650,500,600,500,1650,500,1650,500,600,500,1650,500,1700,500,1650,500,1650,500,1700,500,1650,500,600,500,1650,500,600,500,550,500,600,500,600,500,600,500,550,500,600,500,600,500,550,550,550,500,600,500,600,500,550,500,600,500,1650,500,1700,500,1650,500,1650,500,1700,500,1650,500,1650,500,1650,500};
unsigned int irSignal3[] = {560,1690,560,560,560,1690,560,1690,560,560,560,560,560,1690,560,560,560,560,560,1690,560,560,560,560,560,1690,560,1690,560,560,560,1690,560,1690,560,560,560,1690,560,1690,560,1690,560,1690,560,1690,560,1650,560,560,560,1690,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,560,1690,560,1690,560,1690,560,1690,560,1690,560,1690,560,1650,560,1690,500};
unsigned int rawCodes[RAWBUF]; // The durations if raw
int codeLen,Len1=2,Len2=2,Len3=2; // The length of the code

/////////////////////////////////////////////////////////
/////////////////////////////////////////////////////////
void setup()
{
  Serial.begin(19200);
  
  lcd.begin(); 
  lcd.backlight();
  lcd.setCursor(0,0);
  lcd.print("lectura de control remoto");
}
////////////////////////////////////////////////////////
////////////////////////////////////////////////////////
void dump(decode_results *results) 
{
  int count = results->rawlen;
  codeLen = results->rawlen - 1;
  switch (results->decode_type) {
    default:
    case UNKNOWN:      Serial.print("UNKNOWN ");       break ;
    case NEC:          Serial.print("NEC ");           break ;
    case SONY:         Serial.print("SONY ");          break ;
[... 201 lines cut ...]
..................................................................
aire del spa
Decoded NEC: C3EAD798 (32 bits)
Raw (100): 18334,8850,-4500,550,-1700,550,-1700,600,-550,550,-550,550,-550,600,-550,550,-1700,550,-1700,600,-1650,600,-1700,550,-1700,600,-500,600,-1700,550,-550,600,-1650,600,-500,600,-1700,550,-1700,600,-500,600,-1650,600,-550,550,-1700,550,-1700,600,-1700,550,-1700,600,-500,600,-550,550,-1700,550,-1700,600,-500,600,-550,600,-500,600,-500,600,-550,600,-500,600,-550,550,-550,550,-1700,600,-500,600,-1700,550,-550,600,-550,550,-550,600,-500,600,-550,550,-550,550,-550,600,-550,550,
..................................................................
aire sala papa
88C0051 (28 bits)
Raw (60): -28940 8450 -4250 500 -1600 550 -550 550 -550 550 -500 550 -1650 500 -550 550 -550 550 -550 500 -1650 550 -1600 550 -550 550 -500 550 -550 550 -550 550 -500 550 -550 550 -500 600 -500 550 -550 550 -550 500 -550 550 -1650 500 -550 550 -1600 550 -550 550 -550 550 -500 550 -1600 550 
...................................................................
tv sala papa
E0E040BF (32 bits)
Raw (68): -18312 4500 -4500 500 -1700 550 -1650 550 -1650 600 -550 550 -550 550 -550 550 -550 550 -600 500 -1700 550 -1650 550 -1650 550 -600 550 -550 550 -550 550 -550 550 -550 550 -550 550 -1700 550 -550 550 -550 550 -550 550 -550 550 -600 550 -550 550 -1650 550 -550 550 -1700 550 -1650 550 -1650 550 -1700 550 -1650 550 -1650 550 
..................................................................
tv mi cuarto
71F9F1D7
Unknown encoding: 71F9F1D7 (32 bits)
Raw (38): 17290,8250,-4200,450,-550,500,-550,500,-1550,500,-550,500,-1550,500,-550,500,-500,500,-550,500,-4100,550,-1550,500,-500,550,-1550,500,-500,550,-1500,550,-500,550,-500,500,-550,500,
..................................................................
rv cuerto de mi mama
20DF10EF
Decoded NEC: 20DF10EF (32 bits)
Raw (68): 24268,9050,-4400,650,-500,600,-500,650,-1600,650,-500,600,-500,650,-500,600,-500,650,-500,600,-1650,600,-1650,600,-500,650,-1600,650,-1600,650,-1600,650,-1600,650,-1600,650,-450,650,-500,600,-500,650,-1600,650,-500,600,-500,650,-500,600,-500,650,-1600,650,-1600,650,-1600,650,-500,600,-1650,600,-1650,600,-1650,600,-1650,600,
.................................................................

[controlremoto5.2.txt](https://github.com/z3t0/Arduino-IRremote/files/45332/controlremoto5.2.txt)

carlosD2884, 2015-11-27:
The .txt file is the .ino

AnalysIR, 2015-11-27:
RAWBUF should be increased from 100 to handle longer signals. The maximum sizze is 255 (so it will not work for many AC signals.

...or use 
http://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/

the following may be useful
http://www.analysir.com/blog/tag/air-conditioner/
http://www.analysir.com/blog/tag/aircon/

carlosD2884, 2015-11-27:
the problem is that the AC in my house has a length of 212 and worked well , but this has AC SAMSUNG code length is 100 and not respond at all

AnalysIR, 2015-11-27:
use the sketch in the first link to capture the signal, it is more accurate than irremote

carlosD2884, 2015-12-01:
good morning , and achieve receive and transmit the signal to go diferenes acondicinados air .

The problem that I had was the samsung for air conditioning signal should be sent twice with a pause between a sign and another 4900us .

finding : My recommendation is to continue to use the library as IRremote.h is complete only must modify RAWPUF 250

AnalysIR, 2015-12-01:
Good news :)

Why not post a few signals that worked for you.

carlosD2884, 2015-12-01:
signal sent by the remote coding samsung

[image] 

signal reproduced by the Arduino
[image] signal reproduced by the Arduino

comparison of the two signals to be followed reporducidas 2 times
[image]

z3t0, 2016-02-21:
Seems to be resolved.

closed 2016-02-21 by z3t0

nblrehman, 2018-05-05:
@carlosD2884    Could you please tell me the source where you came to know about the (4900) misroseconds delay... please.. Thankyou.

AnalysIR, 2018-05-05:
Judging from the traces  posted earlier...it looks like there is a 4979 gap between the signals

There also seems to be a 2500 there as well, but that could be just an artefact of the LA usage.

However as all the signal parts seem to have this 4979 gap **exactly**, it may be questionable???