Arduino-IRremote/Arduino-IRremote #240: Can't get transmission working

djaus2, 2015-11-27:
Hi Ken
I appreciate the work you have done in this sphere. I hope you can shed light on the issue I am having.

I hav been experimenting with your library and reception works OK. I can target it with a remote and codes are decoded OK.
I’ve been using an Arduino Uno and Leonard.
Actually the Freetronics EtherTen and the Robot Expansion Shield for Raspberry PI 2 respectively.

My problem is that I can’t get any transmission when I transmit to my receiver.
I have used my phone camera to observe Infrared as you suggest, a good idea.
No matter how I transmit using the library I don’t “see” any Infrared in the camera.

I even tried continuously transmitting marks without any observation of infrared.

I have though set pin3 to GPIO output and toggled the pin and can “see” the LED flash via my phone's camera.

So I guess something is wrong with the PWM.

Do you have any suggestions. (I don’t have a CRO)

Cheers 

David Jones
Embedded MVP
Sportronics
Australia
http://embedded101.com/Blogs/David-Jones

Windows 10 Developer Resources: 
http://embedded101.com/Blogs/David-Jones/entryid/674/Windows-10-Resources

AnalysIR, 2015-11-27:
Check which timer is defined in your copy of the library for UNO & Leonardo & that should indicate on which pin the tx is.
Leonardo will output on a different pin

The easier way is to check every digital pin   (D3,D9, D10 are good places to start)

djaus2, 2015-11-27:
OK I'll try each of those pins thanks.

djaus2, 2015-11-28:
I found my old CRO and was able to dive deeper into things with that.
In playing around with the PWM pins in IRremoteint,h I tried a few settings for TIMER_PWM_PIN
With the UNO I found if I set it to 3 then the PWM worked. So there must a #define not being picked up:
*After:*:
    #if defined(CORE_OC2B_PIN)
    #	define TIMER_PWM_PIN  CORE_OC2B_PIN  // Teensy
    #elif defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
    #	define TIMER_PWM_PIN  9              // Arduino Mega
    #elif defined(__AVR_ATmega644P__) || defined(__AVR_ATmega644__)
    #	define TIMER_PWM_PIN  14             // Sanguino
    #else
    #	define TIMER_PWM_PIN  3              // Arduino Duemilanove, Diecimila, LilyPad, etc
    #endif

*I just added the following LOC after that:*
    #define TIMER_PWM_PIN  3 

Test IRSend now works :)

AnalysIR, 2015-11-29:
@djaus2 
Check out the sketch we just posted here
http://www.analysir.com/blog/2015/11/28/helper-utility-for-troubleshooting-irremote/
and post the output

z3t0, 2016-02-17:
@djaus2 Have you had a chance to try the sketch suggested by @AnalysIR ?

I'll close this for now as it seems that the problem was fixed? However it is unclear what the problem was.

closed 2016-02-17 by z3t0