Arduino-IRremote/Arduino-IRremote #242: Library Implementation

z3t0, 2015-12-02:
We seem to be hitting a lot of walls in terms of configuration as the libraries defines cannot be manipulated from within the sketch, I was wondering if maybe we should shift towards a model where the library becomes a part of the sketch, instead of being independent. That way it can be much more project specific.

Thanks!

z3t0, 2015-12-03:
Basically it is Project Independent vs Project Dependent

bengtmartensson, 2015-12-03:
"Project dependent library" is strictly speaking a contradiction of terms; it violates the very spirit of a "library". I always strive for replacing conditional compilation by runtime user parameters, see [my pull request for IRLib](https://github.com/cyborg5/IRLib/pull/11).  (Unfortunately, this comes at a price, in memory consumption :-( ). 

You may also have a look at my [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino) which does not use any "user preprocessor symbols" (except for those only dependent on board and CPU).

z3t0, 2015-12-03:
@bengtmartensson Thanks for the feedback and your thoughs.

> (Unfortunately, this comes at a price, in memory consumption :-( ).

This is my main concern... also I realized that many users of this library have a very specific need.
One if the main reasons of course is that memory is quite important in such projects.

z3t0, 2015-12-03:
Could we maybe maintain two editions?

bengtmartensson, 2015-12-04:
I do not believe that it is a good idea to maintain an "extreme light-weight library". The hardcore guys chasing the last byte will not use libraries anyhow, but cut-n-paste the code into their main sketch. It even makes sense in some cases -- the compiler can do some optimizations it cannot do in a multi-file program (like identifying and eliminating dead code).

stefa168, 2015-12-04:
What about a version that just has the necessary functionalities to record, decode, encode and transmit, without many structures and objects?

z3t0, 2015-12-04:
Yes, that was one of my ideas and I thought that maybe the implementations
of protocols would simply be provided as example sketches

pcoughlin, 2015-12-10:
I think the "Library Approach" is good for the newer programmers that want to "just get something working".  If we start keeping multiple versions based on different styles, then it will get complicated and only the well seasoned programmers will understand it.

Starting from a general Library that does a lot is a good place.  As @bengtmartensson said, hardcore programmers are going to cut and paste and use one big project sketch to minimize all aspects of the code and variable use.

The IRremote Library, IMHO, should be easy to use, easy to include, well documented, have tools/utilities ... and create a basis for people to start then become more adept based on their particular needs.

The posts in the Issues/Forum are interesting and range from detailed Hardware implementations to Multi-Receive/Send requests.

What we do not see much of is "How do I get it to work" basic questions ... which means that it does work for most people.  Excellent.  Just the target we should be trying to hit.

I think leaving the main code and purpose as a Library is the goal we should continue to have and let all the rest grow up around that basic model.

IMHO.

bengtmartensson, 2015-12-11:
Let me add that producing reusable software components in the form or libraries is one of the corner stones of modern professional software engineering. This is really no need to invoke the "beginners" arguments.

> As @bengtmartensson said, hardcore programmers are going to cut and paste and use one big project sketch 

I would like to clarify that I was talking about the relatively rare use case where you throw out all structuring in order to gain the last few percents.

z3t0, 2016-02-17:
>I think leaving the main code and purpose as a Library is the goal we should continue to have and let all the rest grow up around that basic model.

Awesome, this does not happen often but my mind has been changed :smile:

closed 2016-02-17 by z3t0