Arduino-IRremote/Arduino-IRremote #243: On Pro Micro IR LED sends signal once

John77177, 2015-12-12:
i am working on Pro Micro Arduino(5V) that could receive an IR signal and transmit other. The code works perfect with Uno, But not with the Pro Micro

Since on the Micro PIN3  is not working as on Uno and there is no PIN13 as on leonardo, i managed to use PIN9 on the Pro Micro as an output for the IR Transmitter by activating Timer1 in IRremoteInt.h file :

#define IR_USE_TIMER1   // tx = pin 14
//#define IR_USE_TIMER3   // tx = pin 9
//#define IR_USE_TIMER4_HS  // tx = pin 10

and making the following changes to Timer1's code

#if defined(CORE_OC1A_PIN)
#define TIMER_PWM_PIN        CORE_OC1A_PIN  /* Teensy */
#elif defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
#define TIMER_PWM_PIN        11  /* Arduino Mega */
//#elif defined(__AVR_ATmega644P__) || defined(__AVR_ATmega644__)
//#define TIMER_PWM_PIN        13 /* Sanguino */
#elif defined(__AVR_ATmega32U4__)
#define TIMER_PWM_PIN        9  /* Leonardo */
#else
#define TIMER_PWM_PIN        9  /* Arduino Duemilanove, Diecimila, LilyPad, etc */
#endif

The problem is Transmitting IR signal happens only once which is after the first value received by IR Receiver
After that IRR stops receiving signal and IRT stops transmitting, but

when i disconnect the IRT output pin from PIN9, the IRR starts again receiving signal while the IRT stays not transmitting

Since the code is working just fine with Uno, i think the problem in the IRremote Lib
so what do you suggest i can do to fix this problem

I really appreciate your help

AnalysIR, 2015-12-12:
you  must re-enable the receiver after every send when using IRremote

````
irrecv.enableIRIn(); // Start the IR receiver
````

John77177, 2015-12-12:
It is re-enabled 

It is like the existence of the  IRT after loading the code doesn't not allow more than one command to be sent and right after that it disables the function of the IRR.

And that shows when i remove the IR Led, the IR receiver starts to work again
but when i place the IR LED back both the IRR and IRT don't work

AnalysIR, 2015-12-12:
OK..thanks for the additional info.

Unfortunately, you have not provided enough info for me to assist any further - sorry.

John77177, 2015-12-12:
Thank you so much i figured it out by myself... i am so happy
The problem, i was connecting the Uno and the Micro to the same circuit

closed 2015-12-12 by John77177