Arduino-IRremote/Arduino-IRremote #244: IR Remote not working

dsurgi, 2015-12-14:
I downloaded the IR remote files last year around this time. (N0v 2014)  I was able to get them running with my sketch.   Now I have updated the Arduino IDE and my sketch failed.  I took and renamed the "RobotIRremote" sketch and programs as that was what it was trying to use.  the sketch compiled and saved to the Arduino Pro trinket and even exhibits running signs (serial output as well as leds on board) However it does not read and decode the IR signals.  I even uploaded the IRdecode sketch which i used to set up the IR remote in the first place.  It still does not work.

Were things changed in the programs in the last year. and how would that effect the lib.

AnalysIR, 2015-12-14:
*Were things changed in the programs in the last year.*
**Yes**

*and how would that effect the lib.*
**In some situations lots and in others not at all**

Sorry, about the style of the vague reply above. You really need to construct your questions in a way that help people to help you.
**i.e.** provide detailed information about the problem, platforms being used, code examples, how to reproduce, example output/input, version #s being used, and especially what steps you have taken to research and resolve the problem before posting.

For example there are already a number of issues related to confilcts with RobotIRremote which have been resolved & can be found under closed issues.(may or may not be related to your problems)

dsurgi, 2015-12-14:
sorry for the  vauge questions.  I was able to make it work by deleting 
the robotirremote libs and closing and opening the ide.

sorry for your trouble.

david

z3t0, 2015-12-15:
Nice, Problem solved.

closed 2015-12-15 by z3t0