Arduino-IRremote/Arduino-IRremote #245: Proposal: Receive only optoin

k1100lt, 2015-12-15:
The library has a Send and Receive option. For sending the Timer function of the ATMega is used. With an option to only receive, the timer is freed. So adding a pre-compiler option in a sketch, like
#define IRremote_RX_ONLY, should lead to not compile the Send function.

z3t0, 2015-12-15:
Any reason this was closed?

k1100lt, 2015-12-16:
I thought I saved it by clicking close. My mistake.

z3t0, 2015-12-16:
Oh ok, I'll incorporate this once I get some time, thanks for the idea

yesyesuk, 2015-12-16:
This would be great. I have a project were I need 5x PWM and IRremote on Arduino Minis. Having a receive-only option would make this possible.

ElectricRCAircraftGuy, 2016-01-24:
I thought receive used the timer too, no? Receiving IR codes requires the Arduino interrupting every 50us to check the pin state. This is done with a timer.

AnalysIR, 2016-01-24:
@ElectricRCAircraftGuy not sure yur post matches the topic?
if you want better timing accuracy, you can use this sketch:
http://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/

ElectricRCAircraftGuy, 2016-01-24:
"@ElectricRCAircraftGuy not sure your post matches the topic?"

Perhaps I'm misunderstanding this statement by @k1100lt: "For sending the Timer function of the ATMega is used."

Isn't the "Timer function" he is referring to, one of the ATmega's hardware timer/counters?

AnalysIR, 2016-01-24:
OK - I see your point now. (The point you are making wasnt obvious to me from your first post)

The Arduino system timer on AVRs is Timer0.

Your point is valid. However, if it was being changed it would be possible to use Timer0 & Interupts instead of one of the other dedicated ones, if the objective was to remove the need for a dedicated timer in rx only mode. 

However, the use of a dedicated Timer for Rx means that any pin can be used for Rx. There are other ways of addressing this, of course, But doing it another way would mean a whole other set of compromises/conflicts and possibly break users code already out in the wild.

The challenge is having a general purpose & flexible library on MCUs with limited resoureces and trying to keep it working with other libraries & users code. In this case the 'simple' approach may be the best.

Experienced users will have no problem adjusting the library to their own needs, which leaves beginners who might prefer simplicity.

ElectricRCAircraftGuy, 2016-01-24:
Ok I see now. I always wondered why @shirriff used a timer interrupting every 50us to read the IR receive pin instead of using Pin Change Interrupts. If receive was done using Pin Change Interrupts instead, time-stamps could be grabbed using micros(), the timer in use for sending/receiving could be freed if no sending was required, and the incoming measurements would be *much more* precise (4us precision for micros() instead of 50us precision currently). Also, *every* digital capable pin (including analog pins except A6 & A7) on the ATmega328 has Pin Change interrupt support. However, implementing it can be tricky, and it's really hard to ensure pin change interrupt compatibility with other code that uses them too.

AnalysIR, 2016-01-24:
True, but it would make it very tricky to use as a library (if the user also needs to use PCINTs), so the cost might be that PCINTs are lost to some users......There are several other ways to solve this, but they all come with their own trade-offs.

Have a look at IRLib, which may have this type of support built in, if I remember correctly.

ElectricRCAircraftGuy, 2016-01-24:
I recommend we close this issue since the library currently requires using a timer to send and receive, and the current method of implementation for sending and receiving is not going to be changed.

AnalysIR, 2016-01-24:
@zt30 above has said he will be looking into it and its always useful to get a range of ideas on proposals like this. There may be some clever way of implementing this proposal, if not immediately obvious.

z3t0, 2016-01-25:
@AnalysIR Sorry I've been off this project for a while as I have been looking into AVRs and timers etc and learning about how all of those things work.

I am slowly working on getting back to speed on all the new issues and PRs and will try and include any new feature requests within a reasonable amount of time.

I feel that this point is an excellent reason for keeping the current model:

> However, the use of a dedicated Timer for Rx means that any pin can be used for Rx. There are other ways of addressing this, of course, But doing it another way would mean a whole other set of compromises/conflicts and possibly break users code already out in the wild.

However if anyone is interested in trying to code the proposal for this issue I would be interested in helping out with that.

ElectricRCAircraftGuy, 2016-01-31:
AnalysIR, thanks for the suggestions. Take a look at my IRLib upgrades here: https://github.com/ElectricRCAircraftGuy/IRLib. Rock-solid timer-free receive now exists. It can handle IR data >255 incoming Marks & Spaces now too, among other things.

z3t0, 2016-02-07:
@ElectricRCAircraftGuy Thanks For the contribution I will try and incorporate this into the project once I get some time

closed 2020-10-12 by ArminJo