Arduino-IRremote/Arduino-IRremote #246: Pin 11 PWM disabled in for loop

Warrior87, 2015-12-19:
When I run a for loop to control PWM values for LED strpis, the PWM on pin 11 is zero until it reaches 255, it then goes to 255, and then back to zero. It is supposed to gradually increase and decrease. Pin 11 is controlling the blue LED, and both green and red work fine. I think that there is something in the library that is causing pin 11 to be written to either ONLY high or low, but I am not certain.

Warrior87, 2015-12-19:
[image]

Warrior87, 2015-12-19:
[IR_Recieving.ino.zip](https://github.com/z3t0/Arduino-IRremote/files/67565/IR_Recieving.ino.zip)

AnalysIR, 2015-12-19:
yes probably the timer configured for IRremote.

It looks like you may have run out of timers???  (or available PWM pins)

Check which timer you have configured for IRremote and make sure it doesn't conflict with the ones you are using for RGB.

If you are only using Rx, you may be able to remove the need for a dedicated timer or use an arduino/AVR with more timers available.

Warrior87, 2015-12-19:
Which file in the library would I look for that in? And Rx as in just reading, correct?

AnalysIR, 2015-12-19:
IRremoteInt.h

Yes rx = reading IR signals.

If you run the new helper sketch in the examples it might help. (recent update)

see also
http://www.analysir.com/blog/2015/11/28/helper-utility-for-troubleshooting-irremote/

z3t0, 2016-02-21:
@Warrior87 Closing for inactivity, feel free to reopen

closed 2016-02-21 by z3t0