Arduino-IRremote/Arduino-IRremote #247: IRrecord.ino not compiling out of the box.

mansson, 2015-12-23:
Using 1.6.6 IDE and a M0pro board I followed the installation instructions and opened Arduino-IRremote-master/examples/IRrecord/IRrecord.ino

It doesn't compile. What am I missing?

/home/tony/Arduino/libraries/Arduino-IRremote-stable/IRremote.cpp:106:5: error: expected constructor, destructor, or type conversion before '(' token
 ISR (TIMER_INTR_NAME)
     ^
exit status 1

AnalysIR, 2015-12-23:
Possibly because that board/platform is not supported by IRremote.

You would have to add timer definitions into IRremoteInt.h for that platform.

You could also ask the vendor of the board to look into adding support, as theTeensy guys did for their platform.

emeshuris, 2016-07-02:
MKR1000 same problem, this kind of sucks.
I didnt even realize this might be a problem, already wrote a bunch of code for it on my dev pro arduino board, and now back to the drawing board.

AnalysIR, 2016-07-02:
Why not add the support into the library yourself and contribute it back....I suspect the active contributors do not have this board available to test with.

FYI: I also did a quick search on google to see if there were other forks available - but couldn't see any. I did find one project where a nano($2 from china) was used for the IR part driven by this board.

As above, you could also ask the vendor of the board to look into adding support, as theTeensy guys did for their platform.

z3t0, 2016-07-02:
MKR1000 is currently being worked on myself to add support.u w

> already wrote a bunch of code for it on my dev pro 

Do you mean you wrote an Arduino sketch file or code for the IR library?

closed 2020-10-12 by ArminJo