Arduino-IRremote/Arduino-IRremote #248: irsend.sendLG() not working.

litunny, 2015-12-26:
Hello All, was working with this code:
```C++
/*
Blink
Turns on an LED on for one second, then off for one second, repeatedly.

Most Arduinos have an on-board LED you can control. On the Uno and
Leonardo, it is attached to digital pin 13. If you're unsure what
pin the on-board LED is connected to on your Arduino model, check
the documentation at http://www.arduino.cc

This example code is in the public domain.

modified 8 May 2014
by Scott Fitzgerald
*/

#include

IRsend irsend;

// the setup function runs once when you press reset or power the board
void setup() {
// initialize digital pin 13 as an output.
pinMode(13, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
digitalWrite(13, HIGH); // turn the LED on (HIGH is the voltage level)
delay(1000); // wait for a second
digitalWrite(13, LOW); // turn the LED off by making the voltage LOW
Serial.println("0x88C0051");
irsend.sendLG(0x880094D, 38);
delay(10000); // wait for a second
}
```
But nothing seems to work, I've try alot of samples too, but not working, please kindly assist

chaeplin, 2015-12-26:
https://github.com/z3t0/Arduino-IRremote/blob/master/ir_LG.cpp#L12
LG_BITS is 28.
try irsend.sendLG(0x880094D, 28).

Which Arduino/model do you use ?

litunny, 2015-12-26:
Okay, thank you so much for responding, I really really appreciate, I'm using Arduino Uno, I'll try it out. Many thanks again.

chaeplin, 2015-12-26:
I will check sendLG with https://github.com/z3t0/Arduino-IRremote/releases/tag/stable

pcoughlin, 2015-12-26:
@chaeplin ... Thanks for pitching in to help.

chaeplin, 2015-12-27:
I have tested two version with demo code https://github.com/z3t0/Arduino-IRremote/blob/master/examples/LGACSendDemo/LGACSendDemo.ino

1) https://github.com/z3t0/Arduino-IRremote/releases/tag/stable
2) https://github.com/z3t0/Arduino-IRremote/tree/master

Result : Working, no problem at all.

Also  irsend.sendLG(0x880094D, 28) is working.

@litunny Use PWM 3(in Uno D3) for IR LED(+). With in 1 m radius from A/C.

litunny, 2015-12-27:
Okay, me too will give it a try. Thank you so much, you've been an help.

z3t0, 2016-02-22:
Closed for inactivity

closed 2016-02-22 by z3t0