Arduino-IRremote/Arduino-IRremote #250: irRecv.cpp errors

photobrain1, 2015-12-31:
I consistently generate compile errors with the library. I am using Energia for Launchpad.  The code is quite simple:

#include <IRremote.h>

IRsend irsend;

void setup()
{
Serial.begin(9600);
}

void loop() {
if (Serial.read() != -1) {
for (int i = 0; i < 3; i++) {
irsend.sendNEC(0x20DFDF20, 32); //Send EZ Adjust
delay(100);
}
}
}

However the errors generated (either on TI cloud based Energia or the one on my Mac are:

/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/IRremote/irRecv.cpp: In member function 'void IRrecv::enableIRIn()':
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/IRremote/irRecv.cpp:115:6: error: 'cli' was not declared in this scope
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/IRremote/irRecv.cpp:120:2: error: 'TCCR2A' was not declared in this scope
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/IRremote/irRecv.cpp:120:2: error: 'WGM21' was not declared in this scope
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/IRremote/irRecv.cpp:120:2: error: '_BV' was not declared in this scope
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/IRremote/irRecv.cpp:120:2: error: 'TCCR2B' was not declared in this scope
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/IRremote/irRecv.cpp:120:2: error: 'CS21' was not declared in this scope
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/IRremote/irRecv.cpp:120:2: error: 'OCR2A' was not declared in this scope
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/IRremote/irRecv.cpp:120:2: error: 'TCNT2' was not declared in this scope
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/IRremote/irRecv.cpp:123:2: error: 'TIMSK2' was not declared in this scope
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/IRremote/irRecv.cpp:123:2: error: 'OCIE2A' was not declared in this scope
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/IRremote/irRecv.cpp:123:2: error: '_BV' was not declared in this scope
/Applications/Energia.app/Contents/Resources/Java/hardware/msp430/libraries/IRremote/irRecv.cpp:127:6: error: 'sei' was not declared in this scope

Please help.

AnalysIR, 2015-12-31:
AFAIK the LaunchPad and other MSP devices are not supported by this version of the library.

You might/should  find another version of IRremote that has been ported to the TI & Energia platform via Google

photobrain1, 2015-12-31:
Ironically I have tried even the version approved for the LaunchPad and it still throws the same errors.  It seems to be a problem with the header file, however, I can't seem to track down the proper declarations.

AnalysIR, 2015-12-31:
if it throws the same errors it isnt for the launchpad....as those errors are Atmel/AVR related.

You should confirm that the library version you are using, supports the platform you are using.

Sometimes the IDEs need to be restarted before a newly installed  library will be used correctly.

z3t0, 2015-12-31:
Can you provide a link to the ported library?

photobrain1, 2015-12-31:
http://energia.nu/downloads/downloadv3.php?file=energia-0101E0017-macosx-signed.dmg  It is located in the executable under "Contents/Resources/Java/hardware/msp430/libraries"  From what it looks like it is Ken's 1.0 distro of the arduino library.

z3t0, 2015-12-31:
What board are you using? and https://github.com/energia/Energia/blob/master/hardware/msp430/libraries/IRremote/IRremoteInt.h seems quite old indeed

photobrain1, 2016-01-01:
msp430g2.  I am concentrating on IRremoteInt.h right now.

z3t0, 2016-01-01:
seems that its possible (https://www.youtube.com/watch?v=xhDQqxPzLjc)

https://hackaday.com/2015/06/19/ir-remote-for-smartphone-via-bluetooth-adapter/#more-159421

z3t0, 2016-02-22:
Closed for inactivity

closed 2016-02-22 by z3t0