Arduino-IRremote/Arduino-IRremote #251: Arduino crashes on analogWrite when using this library

Marcel-Robitaille, 2016-01-02:
As stated, my arduino crashes on analogWrite. If I comment line 59, everything is fine. Otherwise, my arduino freezes.

````c
#include <IRremote.h>

int pinR = 11;
int pinG = 13;
int pinB = 12;
int pinSignal = 7;

IRrecv irrecv(pinSignal);
decode_results results;

long codes[6]= {
	0xFD28D7,	//4
	0xFD18E7,	//7
	0xFDA857,	//5
	0xFD9867,	//8
	0xFD6897,	//6
	0xFD58A7	//9
};
int valR = 0;
int valG = 0;
int valB = 0;

void setup(){
	Serial.begin(9600);
	irrecv.enableIRIn();

	pinMode(pinR, OUTPUT);
	pinMode(pinG, OUTPUT);
	pinMode(pinB, OUTPUT);

}

void loop(){
	if(irrecv.decode(&results)){
		if(results.value == codes[0] && valR < 255){
			valR++;
		}

		else if(results.value == codes[1] && valR > 0){
			valR--;
		}

		else if(results.value == codes[2] && valG < 255){
			valG++;
		}

		else if(results.value == codes[3] && valG > 0){
			valG--;
		}

		else if(results.value == codes[4] && valB < 255){
			valB++;
		}

		else if(results.value == codes[5] && valB > 0){
			valB--;
		}
	}
		display(valR, valG, valB);
		Serial.print("Red: ");
		Serial.println(valR);
		Serial.print("Green: ");
		Serial.println(valG);
		Serial.print("Blue: ");
		Serial.println(valB);
		Serial.println("");
		irrecv.resume();

	
}

void display(int red, int green, int blue){
	analogWrite(pinR, constrain(red, 0, 255));
	analogWrite(pinG, constrain(green, 0, 255));
	analogWrite(pinB, constrain(blue, 0, 255));
}
````

z3t0, 2016-01-02:
@Marcel-Robitaille Most likely that timer is being used by the library, can you please specify which board you are using?

Marcel-Robitaille, 2016-01-02:
@z3t0 It's the UNO rev3. I fixed it by using different pins. Now I use 11 for the ir receiver and 5, 6, and 9 for the rgb led (7 and 8 didn't work for some reason).

laufersteppenwolf, 2016-02-21:
Same for me on the Nano board, can't us PWM on pin 11, it outputs either HIGH or LOW but no PWM. Sometimes the board even freezes completely.

z3t0, 2016-02-21:
@laufersteppenwolf Can you share the sketch you are using?

@Marcel-Robitaille It probably has something to do with analogWrite using timers on those pins

laufersteppenwolf, 2016-02-21:
It'snot really nice, as I kinda messed it up searching for an error within my code that would break the analogWrite and haven't reverted my mess yet :D 
It is working now, but blue still seems to flicker while PWM is increasing or decreasing, just not sure yet if it also is lib related or hardware (a transistor exploded yesterday, so could possibly also be hw related)
Anyways, here you go: https://gist.github.com/laufersteppenwolf/34aa35c1a235b4a13220

z3t0, 2016-02-21:
@laufersteppenwolf Thanks I will take a look and respond although I may take a few days to a week... sorry in advance for any inconvenience

laufersteppenwolf, 2016-02-21:
no problem at all, it's not like I need it in order to survive ;)
so take your time, no need to hurry

laufersteppenwolf, 2016-02-22:
@z3t0 I've done some more testing, and pin 3 isn't able to output PWM either. Furthermore, pin5 and 6 do not output a proper PWM signal on lower values, resulting in a flicker rather than a dimming of the LEDs.
pin 9 and 10 are working just fine, though
hope this helps ;)

GeorgeTG, 2016-06-01:
@z3t0 Any update on this? This happens to me on Arduino Nano 328. Writing PWM on PIN 11 makes my arduino instantly freeze while 9 and 10 work fine. Also this has to do something with the library, because if I remove all the IR related code, PWM on pin 11 works fine again.

z3t0, 2016-06-01:
Sorry. Iv been caught up. Il take a look in the next few days and get back to you

eaghra, 2016-06-26:
I can confirm this as well.  Drove me nuts as a beginner Arduino programmer.  I'm glad I found this thread to ease my mind that my newbie code was doing me in.  Doing analogWrite to pin 11 crashes my Arduino Uno R3.  Not always instantly, but usually at some random point within a few hundred milliseconds, even right in the middle of outputting some Serial text.  Pin 3 is unusable as well, but it doesn't crash it.  With code similar to above, I just get occasional blips on an LED attached to pin 3 with analogWrite constantly sending it a value.  So 3 and 11 are no-go's with the IR library enabled, while 5,6,9,10 seem to work fine.

oculus42, 2016-11-24:
I'm experiencing a similar issue on the [DFRobot LED Shield](https://www.dfrobot.com/index.php?route=product/product&product_id=1032) with an Arduino R3 Uno. DFRobot's code example  recommends this library. Green is on Pin 11. If I try to set anything lower than 255, it will freeze or not work.

z3t0, 2017-01-30:
The issue most likely has to do with a timer being used by the library which your sketch is trying to use. Analogwrite uses PWM which uses the Timers, Refer to the code and datasheet of your microcontroller to figure out which timers are not being used by the library

technicalfiddles, 2020-07-25:
This thread was helpful for me too. IR Receive pin is 3 and PWM output is working on 10. 
PWM output did not work on 11.

closed 2020-10-12 by ArminJo