Arduino-IRremote/Arduino-IRremote #252: how i can set transmit pin

jonycoo, 2016-01-05:
Hello
I use your irRemote library to send data but I want to use the PCI21 pin of the ATmega 328, where can I set the transmit pin?
great regards jonycoo

AnalysIR, 2016-01-05:
I am not familiar with that pin # designation for Atmega328. Can you provide a reference?

In general, the transmit pin is determined by the timer used. Have a look at the  IRrremoteInt.h file for details. 

You could also try running the helper sketch in the examples (latest master version), which may provide some insight.

z3t0, 2016-02-21:
That would be PD5 and pin 11 which is OC0B and Timer 0. Currently the library does not support Timer 0, do you have any experience with Timers in AVR? We could work on adding support, otherwise you could simply use PCINT1 or PCINT19 which are on Timers 1 and 2 both of which are supported by this library

closed 2020-10-17 by ArminJo