Arduino-IRremote/Arduino-IRremote #254: Problem transmitting LG IR Code

dpswtf, 2016-01-10:
Hi, I've been having some problems making my IR Transmitter work with my LG TV. Let me preface this by saying that it worked already with a RC5 remote (the circuit is ok). Also all of the codes used were not from dumps on the internet, but rather gathered from the remote itself + an IR receiver + IRrcvDemo/Dump from this library. If it's useful, the remote id is AKB72914209.

For some reason it simply does not want to work. Here's a snippet of code I'm working with:

    const unsigned long LG_POWER = 0x20DF10EF;
    unsigned int irSignal[] = {8950, -4400, 550, -550, 600, -500, 600, -1650, 550, -550, 600, -500, 600, -550, 600, -500, 600, -550, 550, -1650, 600, -1600, 600, -550, 550, -1650, 600, -1600, 600, -1600, 600, -1650, 600, -1600, 600, -500, 600, -550, 600, -500, 600, -1600, 600, -550, 600, -500, 600, -550, 600, -500, 600, -1600, 600, -1650, 600, -1600, 600, -500, 600, -1650, 550, -1650, 600, -1600, 600, -1600, 600};
    
    IRsend irsend;

    void setup()
    {
    	Serial.begin(9600);
    }

    void loop() {
    	irsend.sendNEC(LG_POWER, 32);
    	irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), 38);
      	delay(100);
    }

Both NEC hex code and raw should be the same (power button), just included and tested both in case it's helpful.

I've also tried searching to no avail. Is there something obvious I'm missing? Is it something with LG TV's?

chaeplin, 2016-01-10:
LG_POWER is OK.
Raw is the following.
unsigned int irSignal[] = {8950, 4400, 550, 550, 600, 500, 600, 1650, 550, 550, 600, 500, 600, 550, 600, 500, 600, 550, 550, 1650, 600, 1600, 600, 550, 550, 1650, 600, 1600, 600, 1600, 600, 1650, 600, 1600, 600, 500, 600, 550, 600, 500, 600, 1600, 600, 550, 600, 500, 600, 550, 600, 500, 600, 1600, 600, 1650, 600, 1600, 600, 500, 600, 1650, 550, 1650, 600, 1600, 600, 1600, 600}; // 0x20DF10EF
irsend.sendRaw(irSignal, (int)(sizeof(irSignal) / sizeof(int)), 38);

dpswtf, 2016-01-10:
Thanks for correcting but is there a difference though? I mean in terms of the end result. I get that the raw posted could be wrong, but shouldn't the sendNEC work? I haven't tested the the raw code you posted, but it should be equivalent to the NEC one right? In which case it still won't work, unfortunately.

chaeplin, 2016-01-10:
@dpswtf I just removed - sign and changed sizeof(int). Digital pin 3/Uno is used
irsend.sendNEC(LG_POWER, 32);
delay(5000);
irsend.sendRaw(irSignal, (int)(sizeof(irSignal) / sizeof(int)), 38);
delay(5000);

dpswtf, 2016-01-10:
@chaeplin I tried the changes, and unfortunately still doesn't work. The delay makes sense though, and thanks for the help.

chaeplin, 2016-01-10:
@dpswtf 
Power on off is same with my remote(LG TV use NEC protocol). 
If you use IR LED only(not driven by tr), it will not work when TV is more than 1M way from LED.

dpswtf, 2016-01-10:
Weird, I'm doing all the things you mentioned but still nothing. It is just an IR LED Only, but I'm like testing centimeters away from the TV ... As I've already said, I've tested the circuit with another receiver (a tv box? don't know the name but basically gives me access to more channels) and it works fine with its protocol (RC5). Only the TV itself doesn't work... For example:

    irsend.sendRaw(irSignal, (int)(sizeof(irSignal) / sizeof(irSignal[0])), 38); // LG Power - Does not work
    delay(5000);
    irsend.sendNEC(LG_POWER, 32); // LG Power - Does not work
    delay(5000);
    irsend.sendRC5(0x546, 12); // Other receiver, works!
    delay(5000);

So I don't think the problem is the circuit (it is on PWM pin 3!) and also not the distance. I have no other ideas though :/

chaeplin, 2016-01-10:
Wired. Hmm. Clearly circuit is ok. Code is dumped.

AnalysIR, 2016-01-10:
Some Comments:
- The negative values in you first post should be positive.
- The RAW timings for the NEC protocol are a bit off in the example code above.
-  There could be an issue with your emitter circuit, which could work with one an not the other. Diffiecult to comment without details.
- try the code beow to see, but the sendNC should work, which suggests an issue with the emitter circuit or range or a particulary demanding receiver logic in the TV(unusual)

````
/*
Automatically Generated by AnalysIR - Batch Export Utility
Session History
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : NEC :  : 20DF10EF : 32 : 0
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit http://www.AnalysIR.com or email info@....... for further details

int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

unsigned int Signal_0_0[] = {9000,4500,560,560,560,560,560,1690,560,560,560,560,560,560,560,560,560,560,560,1690,560,1690,560,560,560,1690,560,1690,560,1690,560,1690,560,1690,560,560,560,560,560,560,560,1690,560,560,560,560,560,560,560,560,560,1690,560,1690,560,1690,560,560,560,1690,560,1690,560,1690,560,1690,560}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW

delay(5000);

irsend.sendNEC(0x20DF10EF, 32); //AnalysIR Batch Export (IRremote) // AnalysIR IR Protocol: NEC, 
 
 delay(5000);
````

dpswtf, 2016-01-10:
@AnalysIR Thanks for the help. It still doesn't work with the code provided but I figured I would post everything I have below.
So the code I'm using for testing, which is basically yours, is:

    #include <IRremote.h>

    int khz=38;
    unsigned int Signal_0_0[] = {9000,4500,560,560,560,560,560,1690,560,560,560,560,560,560,560,560,560,560,560,1690,560,1690,560,560,560,1690,560,1690,560,1690,560,1690,560,1690,560,560,560,560,560,560,560,1690,560,560,560,560,560,560,560,560,560,1690,560,1690,560,1690,560,560,560,1690,560,1690,560,1690,560,1690,560};

    IRsend irsend;

    void setup() {}

    void loop() {
      irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
      delay(5000);
  
      irsend.sendNEC(0x20DF10EF, 32); //AnalysIR Batch Export (IRremote) // AnalysIR IR Protocol: NEC, 
      delay(5000);
    }

I did test the RC5 again and it works as expected, infact it works pretty well, from a bit of distance aswell, just this one does not work at all.

In terms of circuitry, it's just an IR LED with a resistor: the anode is connected to PWM D3 with a 180 ohm resistor (dont think I have a lower one available atm), and the cathode to GND.

I realize the RAW dump timing I provided might be a little off, but IRrcvDemo still identified correctly as 0x20DF10EF everytime when I tested the remote with my receiver, so I'm guessing it really is that code.

Now something that might be somewhat useful: before I made the post, I actually made it work 2 times! It was indeed simply a sendNEC with a delay(50), and I basically had to move it around a lot until it worked two times, but after those two times it didn't work anymore even with a lot of fiddling and testing. To keep in mind: I'm doing all of this literally glued to the tv, which makes the distance not relevant at all.

Again thanks for all the help so far!

AnalysIR, 2016-01-10:
Everything seems to be in order.

I presume you are pointing the top of the IR LED towards the TV & you are confident where in the TV the IR receiver is???

FYI: Putting 2 x 180R resistors in parallel will be fine and give you double the IR output. 

...otherwise out of ideas....

dpswtf, 2016-01-11:
@AnalysIR Oh my god... I'm so embarassed right now.. First I'm embarassed because the parallel resistor thing I learned in my first year of college and never thought of it again (CS degree, generally don't fiddle with circuits) so thanks a lot for the tip.

The second thing was.. you were right! It was the position of the IR receiver.. I was so confident it was in one place, but decided to check thoroughly with my remote again anyway and it was not where I thought it was! Once I found it, it started working!

Sorry for wasting both you and @chaeplin time, but I appreciate all of the help given. Have a nice one!

closed 2016-01-11 by dpswtf