Arduino-IRremote/Arduino-IRremote #255: IR SEND (DECODE)

EscuderoKevin, 2016-01-10:
Hi good afternoon, I make a query as to convert the value that gives me this:

#include <IRremote.h>

const int RECV_PIN = 6;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
  irrecv.blink13(true);
}

void loop() {
  if (irrecv.decode(&results)) {
    if (results.decode_type == NEC) {
      Serial.print("NEC: ");
    } else if (results.decode_type == SONY) {
      Serial.print("SONY: ");
    } else if (results.decode_type == RC5) {
      Serial.print("RC5: ");
    } else if (results.decode_type == RC6) {
      Serial.print("RC6: ");
    } else if (results.decode_type == UNKNOWN) {
      Serial.print("UNKNOWN: ");
    }
    Serial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
}
The Value is

NEC: 188008 OFF
NEC: 10188008 ON
A value that I have to send in the irsend.NEC (0x68B92, 20); ??

Thank you.

EscuderoKevin, 2016-01-10:
This value "irsend.NEC (0x68B92, 20)" How to get it ?

z3t0, 2016-01-10:
As a base 10 number? or binary?
as binary: 0b1101000101110010010

AnalysIR, 2016-01-10:
Assuming the values you posted are HEX then try the following:
````
irsend.NEC (0x188008, 32);
delay(5000);
irsend.NEC (0x10188008, 32);
delay(5000);
````
or if they were decimal try
````
irsend.NEC (188008, 32);
delay(5000);
irsend.NEC (10188008, 32);
delay(5000);
````
NEC protocols are nearly always 32 bits long

z3t0, 2016-02-22:
Closed for inactivity

closed 2016-02-22 by z3t0