Arduino-IRremote/Arduino-IRremote #256: Unexpected codeType 5 and 11 on Hisense AC remote

NimitzDEV, 2016-01-20:
Hi,
I am using example IRrecord.ino
When I press the button on the remote, it shows:
[image]
Sometimes it shows:
Received unknown code, saving as raw
 m3350 s1750 m350 s450 m400 s450 m400 s450 m350 s500 m350 s500 m350 s500 m350 s1350 m300 s500 m350 s500 m350 s500 m350 s500 m350 s500 m350 s500 m350 s450 m400 s450 m400 s450 m350 s500 m350 s500 m350 s1350 m350 s450 m350 s1350 m350 s500 m350 s500 m350 s450 m400 s450 m350 s500 m350 s500 m350 s500 m350 s500 m350 s500 m350 s500 m350 s1300 m350 s1300 m400 s1300 m350 s500 m350 s500 m350 s500 m350 s450 m400 s1300 m350 s500 m350 s500 m350 s1300 m350 s500 m350 s500 m350 s500 m350 s500 m350 s1300 m350 s500 m350 s500

but most time it shows Unexpected codeType 5 and 11

Using IRrecvDumpV2
IR code too long. Edit IRremoteInt.h and increase RAWLEN
Timing[100]: 
     +3450, -1650     + 450, - 400     + 450, - 400     + 450, - 400
     + 450, - 350     + 500, - 350     + 450, - 400     + 450, -1250
     + 450, - 350     + 500, - 350     + 500, - 350     + 450, - 400
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, - 400
     + 450, - 400     + 450, - 350     + 500, - 350     + 450, -1250
     + 450, - 400     + 450, -1200     + 450, - 400     + 450, - 400
     + 450, - 400     + 450, - 400     + 450, - 350     + 500, - 350
     + 500, - 350     + 450, - 400     + 450, - 400     + 450, - 400
     + 450, -1250     + 450, -1200     + 450, -1200     + 500, - 350
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, -1250
     + 400, - 400     + 450, - 400     + 450, -1250     + 450, - 350
     + 500, - 350     + 450, - 400     + 450, - 400     + 450, -1250
     + 450, - 350     + 450, - 400
unsigned int  rawData[100] = {3450,1650, 450,400, 450,400, 450,400, 450,350, 500,350, 450,400, 450,1250, 450,350, 500,350, 500,350, 450,400, 450,400, 450,400, 450,400, 450,400, 450,400, 450,350, 500,350, 450,1250, 450,400, 450,1200, 450,400, 450,400, 450,400, 450,400, 450,350, 500,350, 500,350, 450,400, 450,400, 450,400, 450,1250, 450,1200, 450,1200, 500,350, 450,400, 450,400, 450,400, 450,1250, 400,400, 450,400, 450,1250, 450,350, 500,350, 450,400, 450,400, 450,1250, 450,350, 450,400 };  // UNKNOWN DABAE293

Encoding  : UNKNOWN
Code      : FCABFFBF (32 bits)
Timing[5]: 
     + 550, - 350     +1200, - 750     + 250
unsigned int  rawData[5] = {550,350, 1200,750, 250};  // UNKNOWN FCABFFBF

AnalysIR, 2016-01-20:
The main issue seems to be....as indicated in your owm post...

**IR code too long. Edit IRremoteInt.h and increase RAWLEN**

try setting it to 230  (max 255)

NimitzDEV, 2016-01-20:
hi, sorry I still have no idea about which line should I change in IRremoteInt.h, did not see related code in the file, only struct definition code.
but did you mean change RAWBUF? tried 230 and 255, still not working properly.
your help would be appreciated~

AnalysIR, 2016-01-20:
Make sure you are using the latest version of IRremote
.....
try this instead, if changing RAWbuf did not work
http://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/

NimitzDEV, 2016-01-20:
Thanks a lot. According to your code, I changed RAWBUF value to 280 and change the definition of rawlen to  unsigned long. Now the program can detect encoding as PANASONIC and still sometimes UNKNOWN, and when I change the settings on remote, the value still the same after some changes, and I am pretty sure the settings on remote had changed. I dont know why...could you help please?

----------------------
here are some outputs from IRrecvDumpV2

Encoding  : PANASONIC
Code      : 200:2801C242 (48 bits)
Timing[275]: 
     +3400, -1650     + 450, - 400     + 450, - 400     + 450, - 400
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, -1200
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, - 350
     + 500, - 350     + 500, - 350     + 450, - 400     + 450, - 400
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, -1200
     + 450, - 400     + 450, -1200     + 500, - 350     + 500, - 350
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, - 400
     + 450, - 400     + 450, - 350     + 500, - 350     + 500, - 350
     + 450, -1250     + 450, -1200     + 450, -1250     + 450, - 350
     + 500, - 350     + 500, - 350     + 450, - 400     + 450, -1200
     + 500, - 350     + 500, - 350     + 500, -1200     + 450, - 400
     + 450, - 350     + 500, - 350     + 500, - 350     + 450, -1250
     + 450, - 400     + 450, - 350     + 500, - 350     + 450, - 400
     + 450, - 400     + 450, -1200     + 500, -1200     + 450, - 400
     + 450, - 400     + 450, -1200     + 450, -1250     + 450, - 350
     + 500, - 350     + 500, -1200     + 450, - 400     + 450, -1200
     + 450, -1250     + 450, -1200     + 450, -1200     + 500, - 350
     + 500, - 350     + 500, - 350     + 500, -1200     + 450, - 350
     + 500, - 350     + 500, - 350     + 500, - 350     + 500, - 350
     + 500, - 350     + 500, - 350     + 500, - 350     + 450, - 350
     + 500, - 350     + 500, - 350     + 500, - 350     + 500, - 350
     + 500, -1150     + 500, - 350     + 500, -1200     + 450, -1200
     + 500, - 350     + 500, - 350     + 500, - 350     + 500, - 350
     + 450, - 400     + 450, - 400     + 450, - 350     + 500, - 350
     + 500, - 350     + 500, - 350     + 500, - 350     + 500, - 350
     + 500, - 350     + 450, - 400     + 450, - 350     + 500, - 350
     + 500, - 350     + 500, -1200     + 450, - 400     + 450, - 350
     + 500, - 350     + 500, - 350     + 500, - 350     + 500, - 350
     + 500, - 350     + 500, - 350     + 450, - 400     + 450, - 350
     + 500, - 350     + 500, - 350     + 500, - 350     + 500, - 350
     + 500, - 350     + 500, - 350     + 450, - 400     + 450, - 350
     + 500, - 350     + 500, - 350     + 500, - 350     + 500, - 350
     + 500, - 350     + 500, -1150     + 500, - 350     + 500, -1200
     + 500, -1150     + 500, -1200     + 450, -1200     + 500, - 350
     + 500, - 350     + 450
unsigned int  rawData[275] = {3400,1650, 450,400, 450,400, 450,400, 450,400, 450,400, 450,400, 450,1200, 450,400, 450,400, 450,400, 450,350, 500,350, 500,350, 450,400, 450,400, 450,400, 450,400, 450,400, 450,1200, 450,400, 450,1200, 500,350, 500,350, 450,400, 450,400, 450,400, 450,400, 450,400, 450,350, 500,350, 500,350, 450,1250, 450,1200, 450,1250, 450,350, 500,350, 500,350, 450,400, 450,1200, 500,350, 500,350, 500,1200, 450,400, 450,350, 500,350, 500,350, 450,1250, 450,400, 450,350, 500,350, 450,400, 450,400, 450,1200, 500,1200, 450,400, 450,400, 450,1200, 450,1250, 450,350, 500,350, 500,1200, 450,400, 450,1200, 450,1250, 450,1200, 450,1200, 500,350, 500,350, 500,350, 500,1200, 450,350, 500,350, 500,350, 500,350, 500,350, 500,350, 500,350, 500,350, 450,350, 500,350, 500,350, 500,350, 500,350, 500,1150, 500,350, 500,1200, 450,1200, 500,350, 500,350, 500,350, 500,350, 450,400, 450,400, 450,350, 500,350, 500,350, 500,350, 500,350, 500,350, 500,350, 450,400, 450,350, 500,350, 500,350, 500,1200, 450,400, 450,350, 500,350, 500,350, 500,350, 500,350, 500,350, 500,350, 450,400, 450,350, 500,350, 500,350, 500,350, 500,350, 500,350, 500,350, 450,400, 450,350, 500,350, 500,350, 500,350, 500,350, 500,350, 500,1150, 500,350, 500,1200, 500,1150, 500,1200, 450,1200, 500,350, 500,350, 450};  // PANASONIC 200:2801C242
unsigned int  addr = 0x200;
unsigned int  data = 0x2801C242;

Encoding  : PANASONIC
Code      : 200:2801C282 (48 bits)
Timing[275]: 
     +3400, -1650     + 500, - 350     + 450, - 400     + 450, - 400
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, -1200
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, - 400
     + 450, - 350     + 500, - 350     + 500, - 350     + 450, - 400
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, -1200
     + 450, - 400     + 450, -1200     + 500, - 350     + 500, - 350
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, - 400
     + 450, - 400     + 450, - 400     + 450, - 350     + 500, - 350
     + 450, -1250     + 450, -1200     + 450, -1250     + 450, - 400
     + 450, - 350     + 500, - 350     + 500, - 350     + 450, -1250
     + 450, - 350     + 500, -1200     + 450, - 400     + 450, - 400
     + 450, - 400     + 450, - 350     + 500, - 350     + 500, -1200
     + 450, - 400     + 450, - 400     + 450, - 350     + 500, - 350
     + 500, - 350     + 450, -1250     + 450, -1200     + 450, - 400
     + 450, - 400     + 450, -1200     + 450, -1250     + 450, - 400
     + 450, - 350     + 500, -1200     + 450, - 400     + 450, -1200
[... cut ...]  450,400, 450,1250, 450,1200, 450,400, 450,400, 450,400, 450,400, 450,350, 500,350, 500,350, 500,350, 450,400, 450,400, 450,400, 450,400, 450,350, 500,350, 500,350, 500,350, 450,400, 450,1200, 500,350, 500,350, 500,350, 450,400, 450,400, 450,400, 450,400, 450,350, 500,350, 500,350, 500,350, 450,400, 450,400, 450,400, 450,400, 450,350, 500,350, 500,350, 500,350, 450,400, 450,400, 450,400, 450,400, 450,400, 450,350, 500,1200, 450,400, 450,400, 450,400, 450,1200, 450,400, 450};  // UNKNOWN E1C3561B

AnalysIR, 2016-01-20:
I loaded your first signal into AnalysIR & it decoded to the Olympia136AC. So it looks good from here & you can use sendRAW to send these signals.

Row: 1
Hex: FDFFD7FE3DBDF3343BFFE9FFFF7FFFFF43
Binary:
11111101 11111111 11010111 11111110
00111101 10111101 11110011 00110100
00111011 11111111 11101001 11111111
11111111 01111111 11111111 11111111
01000011

cosmopaco, 2016-01-20:
Hello.
Some time ago, I decode a remote control AC Hisense specifically the model Y-H1-01, Are theoretically equivalent Y-H1-02 (E), Y-J1, Y-E407 
You can try https://github.com/ToniA/arduino-heatpumpir

z3t0, 2016-01-25:
@NimitzDEV Just a heads up, please use a gist if you need to paste a block of code or data this large/

Thanks!

Joypacharya, 2016-12-09:
same problem i am facing right now. help me sort out this problem.

AnalysIR, 2016-12-09:
@Joypacharya 

FYI: you are unlikely to get any meaningful assistance, because you have provided **no details** about your 'problem'.

Also, you should consider opening a new issue and if neccessary, link back to this one if it is relevant.

closed 2020-10-17 by ArminJo