Arduino-IRremote/Arduino-IRremote #257: If unknown IR protocol's solution

ying-chiao-liao, 2016-01-23:
I recent want to do a project is copy every device's remote IR data and then control

these device by Arduino.

And if I can not known every user's device and IR protocol.

In this Situation can I just copy the raw data and send raw to avoid some device not support?

Thanks you a lot.

I am a Taiwanese, Nice to meet you!

ying-chiao-liao, 2016-01-23:
I saw the default frequency is 38kHz, so it will work in RC5/RC6 protocol?

thanks a lot

AnalysIR, 2016-01-23:
If you have unlimited storage on your MCU it is possible to use RAW to record, store and send every signal,

In practice, most Arduinos have very limited storage.

You can easily set the carrier frequency when sending RAW signals.
rc5/6 are 36khz.

The best thing is to run some example sketches in the library - you will then understand better.

ying-chiao-liao, 2016-01-24:
I have a question is, if I set 38khz to send RC5/RC6 command it will work or not?
Because the IR receiver can receive the 36khz and 38khz, and then to judge the command is RC5/RC6 or NEC IR protocol command.
So the machine which just can receive the RC5/RC6(36khz) protocol, it can also receive the RC5/RC6 which is be send by 38khz? 
This theory is right or not?

Thanks a lot !

AnalysIR, 2016-01-24:
RC5//6 is 36kHz
NEC is 38kHZ

However, most IR receivers will receive signals at orher frequencies, but the range & accuracy will decrease. You can see this from the data sheet of the IR receiver.

The answer to your question is Yes, but with a slightly reduced range or performance.

Make sure you are using a quality IR receiver like a TSOPxxxxx from Vishay.

(Note: this approach would not work with some tricky protocols, but should be OK with RC5/6/NEC/SONY/SAMSUNG.)

ElectricRCAircraftGuy, 2016-01-24:
@AnalysIR, you said, "Make sure you are using a quality IR receiver like a TSOPxxxxx from Vishay. Avoid the cheap IR receivers on ebay/aliexpress etc." 

Why is that? I have been doing some analysis on the cheapo Chinese [1838 IR receiver from Ebay for <$1 for 10](http://www.ebay.com/sch/i.html?LH_BIN=1&_from=R40&_sacat=0&_nkw=1838+ir+10&_sop=15), but I also have the nicer TSOP4838. With this IRremote library, the 4838 works *great* and the 1838 works *horrible,* at the moment. However, looking at the two on an oscilloscope, I believe the problem is actually in the *library* *not* in the 1838 device. I think the TOLERANCE and MARK_EXCESS parameters simply need to be tweaked for the 1838 to work better. I'm going to work on it and see. Do you have any other evidence or data on why we should not use the cheap 1838 devices?

AnalysIR, 2016-01-24:
**Why is that?**...because they are rubbish & that is why they are cheap.
**Evidence:** see http://www.analysir.com/blog/2014/12/08/infrared-receiver-showdown-tsop34438-vs-vs1838b-winner-revealed/

**Other reading:** http://www.analysir.com/blog/2014/03/27/infrared-receivers-signal-lag/

In my opinion, no point changing the library.

:)

PS: TSOP34438 is better for general use and longer AC signals. 

If you are ever looking ofr the beter ones to use, we have kits avaialable on our website that you can refer to and you may be able to get them cheaper directly.

ying-chiao-liao, 2016-01-25:
Thanks a lot, AnalysIR

z3t0, 2016-01-26:
@ying-chiao-liao Hi, is the issue resolved or do you have any further questions? 
Thank you very much

ying-chiao-liao, 2016-02-06:
@z3t0 
It's resolved thank you !!!

ElectricRCAircraftGuy, 2016-02-06:
PS I got my Ebay 1838 IR  receivers to work fine now, using this library here: https://github.com/ElectricRCAircraftGuy/IRLib. See the readme notes, and the 3 working examples I mention in the readme for additional help and info. The main problem was MARK_EXCESS, as the 1838s actually have space excess not mark excess, so it is imperative that MARK_EXCESS be adjustable, and for the 1838s it needs to be *negative.* The default +100 us MARK_EXCESS that IRremote uses is so far off for the 1838s that they won't work at all.  

I have placed a paragraph of notes above the 3 working examples I mention in the readme. Also, my updates to that library include basic low pass filtering to remove any spurious, very short marks and spaces that may occur due to interference any IR receiver may see.

z3t0, 2016-02-07:
Thanks! I will look into that!

closed 2016-02-17 by z3t0