Arduino-IRremote/Arduino-IRremote #260: Make TOLERANCE and MARK_EXCESS changeable parameters

ElectricRCAircraftGuy, 2016-01-26:
I have been doing some analysis on the cheapo Chinese 1838 IR receiver (from Ebay for <$1 for 10), but I also have the nicer TSOP4838. With this IRremote library, the 4838 works great and the 1838 works horrible, initially. However, looking at the two on an oscilloscope, I believed the problem is actually in the library not in the 1838 device. 

When I change MARK_EXCESS to 0 the 1838 works fine. Since IR receivers can vary, I think MARK_EXCESS and TOLERANCE should be input parameters so you can store them in each sketch rather than having to change them in the library .h file for each different sketch and hardware device.

pcoughlin, 2016-01-26:
This is an interesting point...

What about having different define in the header file for 4838 vs 1838.  Chances are that an individual programmer will be using one set of hardware parts.  Just like the different CPU configurations in the header files.

We can make it a sketch variable with some default (4838 maybe).  The downside is it will increase the RAM footprint of the library.  There are already other people here that have threads that are wanting to shrink the size.

Another option would be to write the code so that a define in the library has 3 options:
1) defined compiler constants for 4838
2) defined compiler constants or 1838
3) define that turns on a code block to make the two values RAM variables.

Any other options?

Would you like to make the code changes and submit them?

ElectricRCAircraftGuy, 2016-01-26:
I still think they should be parameters no matter how I look at it. They could be passed in during object creation in your Arduino sketch, as *optional* input parameters. If no inputs are given, defaults are used. If inputs are given, they override the defaults. If it's important enough to me in the future yes I'm willing to program it, but at the moment I've got too many other things to do to work on it. I just wanted to put it out there is all.

By the way, I've determined the best MARK_EXCESS value for my 1838 is -30us, and the best for my 4838 is +45us. That means the solution needs to allow negative values too.

ArminJo, 2020-10-19:
see #266

closed 2020-10-19 by ArminJo