Arduino-IRremote/Arduino-IRremote #261: Save program memory space by storing strings in once

z3t0, 2016-01-27:
Kudos: @pcoughlin @ElectricRCAircraftGuy
Related: #258 

This is an issue about decreasing the footprint of this library by storing all strings into the flash memory at the beginning and only once.

bengtmartensson, 2017-08-22:
Smartass mode: You mean __constant__ strings of course...

I made a quick scan, and except for commented-out stuff and debug stuff (that is not compiled unless DEBUG is defined (or true?), I could not find any constant strings. (Except for `ir_Pronto.cpp` but that is another story...) So it appears that there is nothing to do. If I am wrong, please correct me.

z3t0, 2017-08-22:
I think this issue was before we had DEBUG so it is no longer relevant.

closed 2017-08-22 by z3t0