Arduino-IRremote/Arduino-IRremote #262: upgrade request: using pin change interrupts & double-buffering to flawlessly receive IR signals

ElectricRCAircraftGuy, 2016-01-31:
I just made some major updates to IRLib. If you'd like to implement Pin Change interrupts for receiving, and you want to see a working example, see the code I added to IRLib here: https://github.com/ElectricRCAircraftGuy/IRLib. Notice that I got rid of the need to call resume() at all, when using the double-buffering method.

Working examples to look at so far include: IRrecvDump.ino, IRanalyze.ino, IRrecvPCIDump_UseNoTimers.ino, though it wouldn't be hard to make any of the examples compatible with the new changes.

t3chguy, 2016-02-07:
Just tested your fork of IRLib, works fantastically. Had issues with this Lib and the Apple Remote sometimes getting recognised as random or unknown encodings, but with your fork and Double Buffering it works 99% of the time. Thanks @ElectricRCAircraftGuy

5shekel, 2016-05-08:
same here, works great.
had a problem with irRemote and pulse sensor sharing timers.  @ElectricRCAircraftGuy fork  solved it.

closed 2020-10-19 by ArminJo