Arduino-IRremote/Arduino-IRremote #263: How do the codes read here relate to HEX codes found on the internet?

kallegit1950, 2016-02-02:
Hi, I want to use your library to read codes I need to translate later to an on/off pattern. The HEX codes given on the internet, e.g. https://www.remotecentral.com/cgi-bin/codes/ have that information. I wonder how the codes you produce relate either to the HEX codes, or to on/off cycles. Can you shed some light onto it? Thanks, K.

ElectricRCAircraftGuy, 2016-02-02:
Hi, there are already two good general-use libraries for this purpose: IRLib (https://github.com/cyborg5/IRLib) and IRremote (https://github.com/z3t0/Arduino-IRremote)

ElectricRCAircraftGuy, 2016-02-02:
One more IRLib link: https://github.com/ElectricRCAircraftGuy/IRLib

pcoughlin, 2016-02-02:
To answer your question, HEX is a different base value for the number system.
Hex is base 16, Binary is base 2, Decimal or Arabic numerals (our number system) is base 10.

Therefore you convert the Hex codes to decimal.  Or in the programming language you can put the numbers in as hex.

Here is the code from your referenced site for [Sony TV Generic, the TV On command](https://www.remotecentral.com/cgi-bin/codes/sony/tv_generic_driver/page-3/):
0000 0067 0000 000d 0060 0018 0018 0018 0018 0018 0018 0018 0030 0018 0030 0018 0030 0018 0018 0018 0030 0018 0018 0018 0018 0018 0018 0018 0018 0407

Those are Hex numbers.  To use them directly in your program you can put them into the array like this:
```
int SonyTVon[] = {0x0000, 0x0067, 0x0000, 0x000d, 0x0060, 0x0018, 0x0018, 0x0018, 0x0018, 
0x0018, 0x0018, 0x0018, 0x0030, 0x0018, 0x0030, 0x0018, 0x0030, 0x0018, 0x0018, 0x0018, 
0x0030, 0x0018, 0x0018, 0x0018, 0x0018, 0x0018, 0x0018, 0x0018, 0x0018, 0x0407};
```

Just for reference, as examples:
0x0067 Hex = 103 decimal
0x0018 Hex = 24 decimal
0x000d Hex = 13 decimal

Here is a [Hex to Decimal converter](http://www.binaryhexconverter.com/hex-to-decimal-converter)

There is no need to convert them.  Just use the hex values as above.

The IRremote library uses decimal numbers for its output and therefore you will see people using decimal numbers primarily in the posts on this site, although it is not required.  The Arduino compiler can use several different base numbers, it does not care.

bengtmartensson, 2016-02-02:
@ElectricRCAircraftGuy:
I would like to add my library [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino) to the list. See also [AGirs](https://github.com/bengtmartensson/AGirs).

@pcoughlin: "Every question has an answer that is neat, short, easy to understand -- and wrong". The "HEX signals" are not microsecond durations in hexadecimal, but the "Pronto CCF/HEX" notation, which I have tried to explain [here](http://www.harctoolbox.org/Glossary.html#ProntoSemantics). To transform from "HEX format" to durations of micro seconds, there are some alternative, like [IrScrutinizer](http://www.harctoolbox.org/IrScrutinizer.html) that can be used. It can also generate C code directly (for Arduino, you presently have to remove the minus signs though).

pcoughlin, 2016-02-02:
Upon more research... here is a cut and paste converter:
**http://irdb.tk/convert/**

You can use the above website to paste the Pronto CCF/HEX into the field and get the RAW data out.

There is a delay when you click the "Get Protocol Information" button (so be patient). Then you will need to click the "RAW" tab to see the data that is used in IRremote.

The example used in the previous post is converted to +/- format to use on the irdb.tk site:
+2385 -596 +596 -596 +596 -596 +596 -596 +1193 -596 +1193 -596 +1193 -596 +596 -596 +1193 -596 +596 -596 +596 -596 +596 -596 +596 -25619

Paste the above +/- number string into the corresponding input and see it convert to Sony12 1 56.

Still "neat, short, and easy" ... hopefully "Right" this time.

>  "Every question has an answer that is neat, short, easy to understand -- and wrong". The "HEX signals" are not microsecond durations in hexadecimal, but the "Pronto CCF/HEX" notation, which I have tried to explain [here](http://www.harctoolbox.org/Glossary.html#ProntoSemantics). To transform from "HEX format" to durations of micro seconds, there are some alternative, like [IrScrutinizer ](http://www.harctoolbox.org/IrScrutinizer.html)that can be used. It can also generate C code directly (for Arduino, you presently have to remove the minus signs though).
>
>The give Sony signal (protocol Sony12, device number 1, function number 56) corresponds to
>
>const int raw_Sony12_1__56[] = { 2385,596,596,596,596,596,596,596,1193,596,1193,596,1193,596,596,596,1193,596,596,596,596,596,596,596,596,25619 }

kallegit1950, 2016-02-02:
Hi, all very good and interesting links. But I cannot find the answer to my question. Let me try to explain again. I have a remote control, I don't know anything about this remote control, no Pronto HEX, nothing. If I use the IR Library to catpure the signal I get as code 4 HEX values. These are obviously not the Pronto HEX values. I need to understand what kind of a code I get, because at the end I need the timings in microseconds of an on/off pattern. If can get this directly out of the output of the library i am fine, if I have to go via these Pronto HEX codes I am good as well.

@pcoughlin How do I have to interpret these raw formats?

bengtmartensson, 2016-02-02:
> I have a remote control, I don't know anything about this remote control, no Pronto HEX, nothing.
> If I use the IR Library to catpure the signal I get as code 4 HEX values.

If you are "lucky", namely if the remote uses a protocol known to the library. At the risk of people throwing things at me (this is the IrRemote issues...) I dare to say that this is not a very good idea. Better is to use an IR sensor, like the Arduino, and analyze the signal at a computer. For this, you may use e.g. IrScutinizer, AnalyseIR, or IrScope. You may want to flash an appropriate sketch on your Arduino for usage with IrScrutinizer or AnalyzeIR. All these programs can analyze the signal and present it as Pronto, micro second sequencies, or as protocol/parameters. 

After you have analyzed and understood the signal, you may like to use the Arduino for a deployment solution receiving and/or sending the signal of your choice.

AnalysIR, 2016-02-02:
@kallegit1950  
I suggest you install one of the libraries and work thru all of the examples provided to assist new users.
You should then understand some things a lot better.

pandacriss, 2016-03-01:
hello kallegit1950. seem we have a similar problem. i find remotecentral.com too and still dont understand how to translate this:
0000 0067 0000 0015 0060 0018 0030 0018 0018 0018 0030 0018 0018 0018 0030 0018 0018 0018 0018 0018 0018 0018 0030 0018 0018 0018 0030 0018 0030 0018 0030 0018 0018 0018 0018 0018 0030 0018 0018 0018 0018 0018 0030 0018 0018 0216

into soemthing like:    irsend.sendSAMSUNG(0x787828D7,32);
for example.
did you find how?

kallegit1950, 2016-03-03:
Hi, I found somewhere how to translate that into micro second patterns. Cannot find it right now. What I used is https://learn.adafruit.com/using-an-infrared-library/hardware-needed to get the mark and space times. The library used rounds sometimes a bit much. So you read the signal, then you see what code you have, e.g. RC5 and then you can adjust the timings, e.g. RC5 is always a multiple of 889 microseconds. So regardless if you measure 850 or 900 you use 889. I used on my Samsung S5 then http://developer.android.com/reference/android/hardware/ConsumerIrManager.html#transmit%28int,%20int[]%29 working fine for me. Hope it helps.

bengtmartensson, 2016-03-03:
You can use IrScrutinizer at least to translate the hex codes to microseconds timings, and to identify most IR protocols. 

I am working on a more elaborate export to C code for IRremote/IRlib/Infrared4Arduino; will probably have something to show this weekend. Stay tuned!

JPaulMora, 2016-03-11:
@bengtmartensson Could you please elaborate on how to do this? I feel a total noob since I couldn't even import the hex (Pronto) formats to IrScrutinizer thanks.

bengtmartensson, 2016-03-11:
Just paste the Pronto Hex to the data window of "Scrutinize signal", and press "Scrutinize".  If you want the raw timings, select Options -> Output text format -> Raw.

There is a [tutorial here](http://www.hifi-remote.com/wiki/index.php?title=IrScrutinizer_Guide). The manual is available [online](http://www.harctoolbox.org/IrScrutinizer.html) and is also installed along with the program. Help -> Program documentation, or just F1.

Stonator, 2017-01-24:
If you want to generate 4 byte hex codes to be used with SendSAMSUNG(...) you can easily generate them by yourself. First of all, Samsung uses the NECx protocol which is described here:
http://www.sbprojects.com/knowledge/ir/nec.php

This means you need to send 32 bits which are created using DDSSFFII where
DD is the bit order inverted device id
SS is the bit order inverted sub-device id
FF is the bit order inverted function id
II is inverted FF

If you now use IRRemote, you can use the following example code to generate your own hex codes:

```
// Reverse the order of bits in a byte
// Example: 01000000 -> 00000010
unsigned char ReverseByte(unsigned char b)
{
  b = (b & 0xF0) >> 4 | (b & 0x0F) << 4;
  b = (b & 0xCC) >> 2 | (b & 0x33) << 2;
  b = (b & 0xAA) >> 1 | (b & 0x55) << 1;
  return(b);
}

// Calculate 32 bit NECx code
unsigned long GenNECXCode(unsigned char p_Device, unsigned char p_SubDevice, unsigned char p_Function)
{
  unsigned long ReverseDevice = (unsigned long)ReverseByte(p_Device);
  unsigned long ReverseSubDevice = (unsigned long)ReverseByte(p_SubDevice);
  unsigned long ReverseFunction = (unsigned long)ReverseByte(p_Function);
  return((ReverseDevice << 24) | (ReverseSubDevice << 16) | (ReverseFunction << 8) | ((~ReverseFunction) & 0xFF));
}

int SAMSUNG_BITS = 32;
#define SAMSUNG_POWER_ON  GenNECXCode(7,7,153)

irsend.sendSAMSUNG(SAMSUNG_POWER_ON, SAMSUNG_BITS);

```

Samsung always uses device 7 and sub-device 7. You can find a function reference here:
http://www.remotecentral.com/cgi-bin/mboard/rc-discrete/thread.cgi?6024

Using the hex codes is much more efficient than using the raw data. If you do not want to have the overhead of generating your hex codes all the time, you could pre-calculate and use them directly.

deathelk, 2019-02-27:
> If you want to generate 4 byte hex codes to be used with SendSAMSUNG(...) you can easily generate them by yourself. First of all, Samsung uses the NECx protocol which is described here: [...]

Perfect. Works great, especially useful for those obscure Samsung discrete code references found on the Internet.

bengtmartensson, 2020-07-11:
This question has been thoroughly answered. Please close  the issue.

closed 2020-08-03 by ArminJo