Arduino-IRremote/Arduino-IRremote #264: panasonic decodes show different codes with each button press.

pandacriss, 2016-02-07:
i have two generic panasonic remotes, them work well on TV and the blueray, however they show different codes with every button press. the dumbV2 show them as UNKNOWN .
also dump print twice with each button press.
this are the remotes: https://imgur.com/5RamVG7

pandacriss, 2016-02-07:
Encoding  : UNKNOWN
Code      : 2BE65D85 (32 bits)
Timing[99]: 
     +3400, -1650     + 450, - 500     + 350, -1300     + 450, - 450
     + 400, - 450     + 400, - 500     + 400, - 450     + 400, - 450
     + 400, - 500     + 400, - 450     + 400, - 450     + 400, - 500
     + 350, - 500     + 400, - 450     + 400, -1250     + 450, - 500
     + 400, - 450     + 400, - 500     + 350, - 500     + 400, - 450
     + 400, - 500     + 350, -1250     + 450, -1300     + 450, - 450
     + 400, -1250     + 450, - 500     + 350, - 500     + 400, - 450
     + 400, - 500     + 350, - 500     + 400, - 450     + 400, - 500
     + 350, - 500     + 400, - 450     + 400, -1250     + 450, - 500
     + 350, - 500     + 400, - 450     + 400, - 500     + 350, - 500
     + 400, -1250     + 450, - 450     + 400, -1250     + 450, - 500
     + 400, - 450     + 400, -1250     + 450, -1250     + 450, - 500
     + 400, - 450     + 400
unsigned int  rawData[99] = {3400,1650, 450,500, 350,1300, 450,450, 400,450, 400,500, 400,450, 400,450, 400,500, 400,450, 400,450, 400,500, 350,500, 400,450, 400,1250, 450,500, 400,450, 400,500, 350,500, 400,450, 400,500, 350,1250, 450,1300, 450,450, 400,1250, 450,500, 350,500, 400,450, 400,500, 350,500, 400,450, 400,500, 350,500, 400,450, 400,1250, 450,500, 350,500, 400,450, 400,500, 350,500, 400,1250, 450,450, 400,1250, 450,500, 400,450, 400,1250, 450,1250, 450,500, 400,450, 400};  // UNKNOWN 2BE65D85

Encoding  : UNKNOWN
Code      : 749385D3 (32 bits)
Timing[99]: 
     +3350, -1700     + 450, - 400     + 450, -1250     + 450, - 500
     + 350, - 500     + 400, - 450     + 400, - 500     + 350, - 500
     + 400, - 450     + 400, - 500     + 350, - 500     + 400, - 450
     + 400, - 500     + 350, - 500     + 400, -1250     + 450, - 450
     + 400, - 500     + 350, - 500     + 400, - 450     + 400, - 500
     + 350, - 500     + 400, -1250     + 450, -1250     + 450, - 500
     + 350, -1300     + 450, - 450     + 400, - 450     + 400, - 500
     + 400, - 450     + 400, - 450     + 400, - 500     + 350, - 500
     + 400, - 450     + 400, - 500     + 350, -1300     + 450, - 450
     + 400, - 500     + 350, - 500     + 400, - 450     + 400, - 450
     + 400, -1250     + 450, - 500     + 400, -1250     + 450, - 450
     + 400, - 500     + 350, -1300     + 450, -1250     + 450, - 450
     + 400, - 500     + 350
unsigned int  rawData[99] = {3350,1700, 450,400, 450,1250, 450,500, 350,500, 400,450, 400,500, 350,500, 400,450, 400,500, 350,500, 400,450, 400,500, 350,500, 400,1250, 450,450, 400,500, 350,500, 400,450, 400,500, 350,500, 400,1250, 450,1250, 450,500, 350,1300, 450,450, 400,450, 400,500, 400,450, 400,450, 400,500, 350,500, 400,450, 400,500, 350,1300, 450,450, 400,500, 350,500, 400,450, 400,450, 400,1250, 450,500, 400,1250, 450,450, 400,500, 350,1300, 450,1250, 450,450, 400,500, 350};  // UNKNOWN 749385D3

Encoding  : UNKNOWN
Code      : C3293526 (32 bits)
Timing[99]: 
     +3400, -1650     + 450, - 450     + 400, -1300     + 450, - 400
     + 450, - 500     + 350, - 450     + 450, - 450     + 400, - 500
     + 350, - 500     + 400, - 450     + 400, - 400     + 450, - 500
     + 400, - 400     + 450, - 450     + 400, -1250     + 450, - 450
     + 450, - 450     + 400, - 500     + 350, - 500     + 400, - 450
     + 400, - 450     + 400, -1300     + 400, -1300     + 450, - 400
     + 450, -1250     + 450, - 500     + 350, - 500     + 400, - 450
     + 400, - 500     + 350, - 500     + 400, - 450     + 400, - 500
     + 350, - 500     + 400, - 450     + 400, -1250     + 450, - 450
     + 450, - 400     + 450, - 450     + 400, - 500     + 350, - 500
     + 400, -1250     + 450, - 500     + 350, -1300     + 400, - 450
     + 450, - 450     + 400, -1250     + 450, -1300     + 400, - 500
     + 400, - 450     + 400
unsigned int  rawData[99] = {3400,1650, 450,450, 400,1300, 450,400, 450,500, 350,450, 450,450, 400,500, 350,500, 400,450, 400,400, 450,500, 400,400, 450,450, 400,1250, 450,450, 450,450, 400,500, 350,500, 400,450, 400,450, 400,1300, 400,1300, 450,400, 450,1250, 450,500, 350,500, 400,450, 400,500, 350,500, 400,450, 400,500, 350,500, 400,450, 400,1250, 450,450, 450,400, 450,450, 400,500, 350,500, 400,1250, 450,500, 350,1300, 400,450, 450,450, 400,1250, 450,1300, 400,500, 400,450, 400};  // UNKNOWN C3293526

Encoding  : UNKNOWN
Code      : A5A9F5F0 (32 bits)
Timing[99]: 
     +3400, -1650     + 450, - 400     + 450, -1250     + 450, - 450
     + 450, - 450     + 400, - 500     + 350, - 500     + 400, - 450
     + 400, - 500     + 350, - 500     + 400, - 400     + 450, - 400
     + 450, - 500     + 400, - 450     + 400, -1250     + 450, - 450
     + 400, - 450     + 450, - 450     + 400, - 500     + 350, - 500
     + 400, - 400     + 450, -1250     + 450, -1250     + 450, - 500
     + 400, -1250     + 450, - 450     + 400, - 500     + 350, - 500
     + 400, - 400     + 450, - 500     + 350, - 500     + 400, - 450
     + 400, - 500     + 350, - 450     + 450, -1250     + 450, - 450
     + 400, - 450     + 400, - 450     + 450, - 450     + 400, - 500
     + 350, -1300     + 450, - 400     + 450, -1250     + 450, - 500
     + 350, - 500     + 400, -1250     + 450, -1250     + 450, - 450
     + 400, - 500     + 400
[... 390 lines cut ...]
     + 350, - 500     + 400, - 450     + 400, - 500     + 350, - 500
     + 400, - 450     + 400, - 500     + 350, -1300     + 450, - 450
     + 400, - 450     + 400, - 500     + 350, - 500     + 400, - 450
[... cut ...]
unsigned int  rawData[99] = {3400,1650, 450,500, 350,1300, 450,450, 400,450, 400,500, 350,500, 400,400, 450,500, 350,500, 400,450, 400,500, 350,450, 450,450, 400,1250, 450,500, 400,450, 400,450, 400,450, 400,500, 400,450, 400,1250, 450,1250, 450,500, 400,1250, 450,500, 350,500, 400,450, 400,450, 400,500, 400,450, 400,450, 400,450, 450,400, 450,1250, 450,500, 350,500, 400,450, 400,500, 350,500, 400,1250, 450,450, 400,1250, 450,500, 400,400, 450,1250, 450,1250, 450,500, 350,500, 350};  // UNKNOWN 4D2AA05

ElectricRCAircraftGuy, 2016-02-07:
try reducing MARK_EXCESS in the header file from 100 to 50 or 0, then test again.

pandacriss, 2016-02-07:
it works perfect now. thank you very much for your time.
im not sure what does that part of the code. i guess my sensor is not suffering of any lag. maybe because i put a low pass filter in it as recommend in the data-sheet
muchas gracias.

ElectricRCAircraftGuy, 2016-02-07:
Now that you see how MARK_EXCESS needs to be customized to individual IR receivers, and this library does not yet support that, I encourage you to please open up an issue to request that MARK_EXCESS be made a user-adjustable parameter that the user sets in the Arduino sketch, either during creation of the IRrecv object, or as a function call in setup().

z3t0, 2016-02-17:
Closing as it is solved, will follow up on the MARK_EXCESS problem in #266

closed 2016-02-17 by z3t0