Arduino-IRremote/Arduino-IRremote #266: 1838 enhancement

z3t0, 2016-02-07:
See #257 

By @ElectricRCAircraftGuy
```
P.S. Just to finish up my little side-conversation above: I got my Ebay 1838 IR receivers to work fine now, using this library here: https://github.com/ElectricRCAircraftGuy/IRLib. See the readme notes, and the 3 working examples I mention in the readme (currently IRrecvDump.ino, IRanalyze.ino, and IRrecvPCIDump_UseNoTimers.ino) for additional help and info. The main problem was MARK_EXCESS, as the 1838s actually have space excess not mark excess, so it is imperative that MARK_EXCESS be adjustable, and for the 1838s it needs to be negative. The default +100 us MARK_EXCESS that IRremote uses is so far off for the 1838s that they won't work at all.

I have placed a paragraph of notes above the 3 working examples I mention in the readme. Also, my updates to that library include basic low pass filtering to remove any spurious, very short marks and spaces that may occur due to interference any IR receiver may see.

My final test would be to see how the 1838s compare outside, with some sunlight, to the TSOP4838s, using my newly updated library (linked to above).
```

AnalysIR, 2016-02-07:
@z3t0 
On this topic, I suggest you consider the following, if planning to add to the library:

- adding additional code to the library to cater for deficiencies in this really poor device
- the additional overhead (MCU cycles) in sketches to process noise from this device
- Extra Flash required for this code (not sure how much that is)
- Extra SRAM used  (if any)
- The impact on users who will not know why their sketch is 'sluggish' or not performing as expected/normal because (unkown to them) there just happens to be a lot of spurious noise coming from this device - using up critical resources. I expect this may be of more relevance for users who are using the library along with other time critical tasks which are usually more difficult to debug.
- The likely support requests from users of the library.

z3t0, 2016-02-13:
@AnalysIR In that case, maybe it would be a better idea to simply reroute those with this receiver to @ElectricRCAircraftGuy 's code?

AnalysIR, 2016-02-13:
@z3t0 
Having a customisable parameter is always fine. My main concern is that this device is really poor (polite understatement) & (for me at least) its usage should be discouraged. Search thru all of the wasted time spent on the Arduino forum supporting issues arising from this device.

So, if the intent is to provide a workaround to make this device usable with the library - I think the issues I listed above should be considered and a 're-route' might be appropriate. 

On the other hand, if the intent is to provide a customisable parameter for all receivers, ignore my comments. (but please remove references to this particular device)

 In all scenarios, I am strongly of the opinion, that any usage of or (non-negative) comment about this particular device should be discouraged.

As a funny aside - I did consider a potential use for this device as an IR noise/interference detector and for a time gave it some serious consideration. :)

ElectricRCAircraftGuy, 2016-02-13:
We just need this parameter (MARK_EXCESS) to be adjustable, and able to be positive or negative, period, regardless of receiver used.  

I took a quick look at the outputs of an 1838 simultaneously with a TSOP4838, one on each channel of my oscilloscope, while sending signals from a remote, and they were identical. The only difference was that the Marks on the 4838 were on avg. ~80us longer than the 1838, and the spaces on the 4838 were on avg ~80us shorter than the 1838. That's it. Both had error and some fluctuations in interpreting the IR pulses, and both fluctuated about the same--ie: the precision, or resolution and repeatability, of the two was approximately equal. The 1838 worked fine. No glitches, no problems.   

Set MARK_EXCESS to approx. 50us and the TSOP4838 works best. Set MARK_EXCESS to 50us and the 1838 doesn't work at all. Set it to -30us and it works best.  

This is not a receiver issue, it's a library issue, and it's frankly a dead-simple one for which there's no logical reason *not* to implement it.  

I read @AnalysIR's articles and criticism of the 1838, and I think it's a marketing thing, and I don't *yet* trust the "analysis" that it's rubbish.  I haven't tested mine in the sun. I still need to do that. But so far I distrust this open-ended, repeated statement that the 1838 is "rubbish." The "evidence" given by @AnalysIR is incomplete and fails to prove anything at this point. If anything, it supports my evidence that MARK_EXCESS needs to be adjustable to support *all* receivers, including the 1838 $0.09 receiver, which so far for me, works just fine.  

**NONE** of my IRLib mods I did here (https://github.com/ElectricRCAircraftGuy/IRLib) were targeted at the 1838. They were to improve reception of all receivers.  

Making MARK_EXCESS adjustable isn't to "support a rubbish receiver," it's to support the fact that different manufacturers implemented the band-pass filter in their receivers differently, which means the marks and spaces are cut off differently.  

If anything, as far as MARK_EXCESS is concerned, all TSOP receivers are *less* accurate than the 1838. The 1838 only skews the marks and spaces by ~30us, whereas the Vishay TSOP ones skew them by 50+. Just because Ken Shirriff originally made his library for TSOPs doesn't mean that 1838s are garbage, it means he didn't have any to test.

So, at this point, until one of us does additional, more thorough testing, with oscilloscope screenshots, and real, hard data, including error analysis, frequency of spurious pulses, and Oscilloscope screenshots of data received both in and out of the sun and other sources of interference, to prove one way or the other, I refute all statements that 1838s are "rubbish," I fully support their use. 

This is a basic library change to support the fact that not all IR receivers have as much error as the original ones Ken Shirriff used. Stated with part #'s, this is a basic library change to support the fact that 1838's have *less* band-gap bias error than TSOP4838s and other "high quality" Vishay receivers. 

This isn't a business for me. I have nothing to gain by "promoting" 1838s. I promote smart library making is all, and saving money where able.

z3t0, 2016-02-14:
@ElectricRCAircraftGuy 

I am going to have to ask you to carefully moderate your post above as you have made some very strong and bold personal attacks against @AnalysIR who is a very important contributor to this project, this is me giving you a chance; Otherwise I will take it upon myself to edit out those parts which I find inappropriate

ElectricRCAircraftGuy, 2016-02-15:
done, please see above.

ElectricRCAircraftGuy, 2016-02-15:
@z3t0,  PS. As a side note, I did see your offer to be a collaborator on the project, on your post elsewhere, and  I'd like to help out, but I have a few reservations: 1) I am unsure whether or not you will like all my changes, 2) I see a lot of empty space in this library's coding style (ex: see snippet below), and I kind of want to remove all of that, which has no value added for me to do that, and it just might make you upset, 3) I'm overextended in projects right now and will only be able to contribute in spurts here and there, and 4) my greatest contributions and most significant changes, improvements, and additions, will be in the most fundamental hardware aspects of the libraries, including improving/rewriting the existing 50us-based ISR, as well as adding and writing the pin-change interrupt-based method ISR. I assume that once I begin rewriting ISRs, according to what I think is best, I'll get a lot of push-back from the existing contributors who say, "why are you fixing what isn't broken." I'm afraid of disagreement. I don't want to fight with people over it. I want to make changes, test them, let you see for yourself they are better, and move on. What I foresee happening, however, is me spending dozens of hrs to make a better product, only to have it use 20 bytes more SRAM and 300 bytes more Flash, which will be utterly unacceptable to some, and then my changes will get rejected. 

What do you think? perhaps we can talk about this elsewhere if this is not the appropriate place for this discussion.

[image]

z3t0, 2016-02-15:
> Done, please see above. Also, thank you for changing the title from "1838 enhancement" to "Mark_Excess Enhancement". That is a much more correct description of what this will be doing here.

Thank you for understanding and modifying your post, also I am sorry for the confusion about the Issue's name, I had made the issue quickly and did not spend time reading the details carefully.

On the other things, I will get back to you asap.

AnalysIR, 2016-02-15:
Now that the topic has been clarified, here are some additional comments:
What is referred to as Mark_Excess is usually covered in the sensors data sheet and will often be specified as a multiple of the carrier period (negative and/or positve). e.g +/- 6 times the carrier period.
_(Ironically, the equivalent spec in the VS1838B  data sheet is quite hilarious)_

We published a blog post 2 years ago on a related topic:
https://www.analysir.com/blog/2014/03/27/infrared-receivers-signal-lag/

FYI: in our own IR decoding technology we do not depend directly on this parameter when decoding, because the variance is due to a multitude of environmental factors. (e.g. ambient lighting levels at different times/locations.). It is not a constant for each model or each device, it is a range which can vary for every signal received and in fact will vary for each mark/space pair received, even within the same signal.
If the parameter is being used, it should take into account a +/- window, which for a 30kHz carrier would be exactly 200uS (for 6 times). 
Caution: I believe within the library, that this +/- 200 uSecs has been handled by both the mark_excess **and** the tolerance parameter - so any configurable adjustments should really consider both parameters, in the context of what specs the library is aiming to match and the ability of users to configure these parameters.

Conclusion: As I mentioned earlier, having this parameter configurable sounds like an improvement. I also suggest that the default behaviour should remain as is to avoid the potential for breaking any existing user's code.

ElectricRCAircraftGuy, 2016-02-15:
@AnalysIR, thanks for the additional information. I agree, we should make both tolerance and Mark_Excess easily adjustable as input parameters. 

Can you please help me understand how to read the datasheets better?

I see your statement in your article you linked to, "Looking at the data sheet again, tells us that this signal distortion can in fact range from -125 microseconds to +150 microseconds (@40kHz modulation) for marks", and I still don't see where to find that in the datasheet. I may be scanning too quickly and overlooking it. I see the datasheet you are using is here: http://www.vishay.com/docs/82479/tssp58038.pdf. I would appreciate your help to teach me where to find and/or calcualte the -125us and +150us from the datasheet. 

I'd like to do the same for the TSOP4838 and VS1838 datasheets too. Here are the datasheets I'm looking at for those parts:

* TSOP4838: http://www.vishay.com/docs/82459/tsop48.pdf
* VS1838: http://electronilab.co/wp-content/uploads/2015/05/vs1838b.pdf

AnalysIR, 2016-02-15:
As the TSOP34438 would likely be the best general pupose IR receiver for most typical users of IRremote, I will point you to Figure 1 of that data sheet, where it is covered. (I think it is Figure 1 in most Vishay datasheets ??)

````
-125 microseconds to +150 microseconds (@40kHz modulation)
````
**_1000000/40000=25uSecs  (the carrier period)
25 x 5=125 uSecs
25 x 6=150 uSecs_**

Regarding "that other device", I decided a long time ago to minimise the amount of time being wasted  with it, so I will leave you to figure that one out for yourself. But it did give me a few laughs reading the data sheet, which compensated for the $ I wasted on them

ElectricRCAircraftGuy, 2016-02-15:
I see it now. Thank you. It's hard to notice all the small details of datasheets, as one learns about new devices. 

To help others: 

TSOP34438 datasheet: http://www.vishay.com/docs/82489/tsop322.pdf
Figure 1: 
[image]

* 1) 7/f0 < td < 15/f0
** Since pd = 1/f, this is the same as saying: 7*pd < td < 15*pd
** This gives us the *delay time* range, which isn't really that important for decoding, but is interesting anyway.
** For a 38Khz receiver, we get a carrier pd of 1e6/38000 = 26.3us
** That means the delay period is between 7*26.3 = 184.1us and 15*26.3 = 394.5us. 
* 2) tpi - 5/f0 < tpo < tpi + 6/f0
** again, substituting 1/pd for f, we get: tpi - 5*pd < tpo < tpi + 6*pd
** carrier pd = 26.3us
** So, the output time period will range from -5*26.3 = **-131.5us** to +6*26.3 = **+157.8us**
*** The midpoint of this is 157.8 + -131.5 = 26.3us, which makes sense now why you chose 30us as your default Mark_Excess in your article above. 

I see it now. Thanks.

ElectricRCAircraftGuy, 2016-02-15:
Now, to answer my own question about the 1838, it looks like the datasheet doesn't have that kind of detail. They simply say here (see image below), that for a nominal high or low period of 600us sent by the sender IR LED, the VS1838 receiver will read in anywhere from 500 to 700us. That means they state a simple value, that **the receiver's reading error is a simple +/-100us for reading both Marks (output LOW) and Spaces (output HIGH)**. This information is good enough for most applications, and they've done a pretty good job at at least centering their bias error (corresponding to the Mark_Excess parameter) close to 0us. Good enough for me. I'm used to working on less information when you use less expensive Chinese products. Pay less, get less documentation. I'm satisfied with these answers. Again, thanks for your help with the Vishay TSOP datasheets.  

[image]

AnalysIR, 2016-02-15:
**Regarding the 30uSecs** - that is primarily related to hardware and software decoding algorithms and not directly based on the referenced data-sheet specs. Since that figure was published we have improved on that again. So not directly relevant to IRremote.

For libraries like  IRremote, a variation of up to +/- 200 uSecs needs to be catered for on each timing value to cover the standard carrier frequencies from 30->56/58kHz, whether it is via .mark-excess and/or tolerance parameters. I suspect just one parameter would suffice.
_____________________________________________________________________
Alas, time to bring out the vampire repellent cross and silver stake wrt to the 1838...if only they existed. (Apparently, sunlight does the job on Vampires too)

**Regarding, your earlier post after your scope tests**
First of all thanks for taking the time to do all of that testing and posting your results.

I have studied your photos, annotations & posts above arising from your tests.

All I can say is that we will have to agree to disagree, because I have not reached the same conclusions as you have reached, even with your updated information. In fact based on the annotated oscilloscope photos you posted, I believe the position I stated on this VS1838 device is strengthened.

My advice to others is to avoid this device at all costs, even at the newer low cost of ~US$0.05 cents each (55 cents for 10 including worldwide shipping!). **_"Penny wise, pound foolish"_** springs to mind.

In particular, do not use the VS1838 in any situation where damage to property or personal injury could occur

**Regarding your last post on the 1838 data sheet**,. There is a world of difference between less information and incorrect information. I suggest you try to figure out the "temporal discontinuity" shown (sorry for being cryptic) and then figure out why the unknown manufacturer's name has been removed from all copies of the data-sheet I have seen online and why VS is placed in the name to let-on that it could be a Vishay product and then why it costs the price of a few  resistors. **_Caveat Emptor!_** ( aka Buyer beware)

Although it has been some fun and helpful to see all of the Scope photos, i won't be posting on this thread again. **_Tempus fugit_**

ElectricRCAircraftGuy, 2016-02-15:
Sounds good. Thanks. 
I hope we can still be cordial despite any disagreements. Your kits do look useful. I hope your business is going well.

commit 555acf34 (2020-10-19): Closes #266
closed 2020-10-19 by ArminJo