Arduino-IRremote/Arduino-IRremote #269: Connecting in series

lior22lior22, 2016-02-13:
Hello i working on universal remote project, and i need to send ir signal to any direction of the room, so i need multiple ir emitter (i have IRL80A emitter), i read that you can only connect 1 to arduino uno on pin 3, so i thought about connecting multiple emitters in series, what do you think?

by the way i will be very happy if someone can explain my why there is only 1 pin support to IRSend?

t3chguy, 2016-02-13:
put them in parallel

lior22lior22, 2016-02-13:
if i will put them in parallel and send signal it should sended from all of them? @t3chguy

t3chguy, 2016-02-13:
as long as your arduino pin can provide current for all of them, yeah
if not, you may need to use a transistor with a base resistor for running multiple IR emitters

lior22lior22, 2016-02-13:
thank you very much :) @t3chguy

t3chguy, 2016-02-13:
no problem

AnalysIR, 2016-02-13:
The reason for one pin - it is tied to the PWM feature in and AVR/Arduino. There are several such pins, so you could use all of them.
You could drive multiple IR LEds from one pin with some circuitry. 
See some posts on our blog:
https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/
https://www.analysir.com/blog/2014/10/03/driving-infrared-led-directly-arduino-pin/
https://www.analysir.com/blog/2015/05/12/simple-infrared-pwm-on-arduino/  (3-part series)

You could change the last one to generate IR output on as many pins as you want. Using direct port mapping on the AVR, should not introduce any overhead, which would give you up to 8 pins.

Another way to get full coverage is to use IR LEds wider angles of coverage. eg TSAL6400.
Another, way is to just pump more current thru the IR led - it is light and will just reflect off walls, ceilings etc.

Your IRL80A seems to have lower max current than typical IR LEds, but should still work.

I would suggest 2 in series (if using 5v) and if in parallel each Led should have its own current limiting resistor.

A lot depends on the specifics of your application.

lior22lior22, 2016-02-14:
thanks alot :) @AnalysIR

AnalysIR, 2016-02-14:
from my quick read of the data sheet, it seems adequate.

If possible use a transistor circuit to drive the LED, for best results - otherwise if driving directly from the Arduino use a 100 ohm resistor or maybe as low as 74 ohms.

It wasnt clear to me if the Led emitts out the from the from or top of the plastic. so you may have to try a few different orientations.

z3t0, 2016-02-17:
Seems to be solved, I will keep this open as a reminder to add a note regarding multiple transmitters in the wiki

closed 2020-10-17 by ArminJo