Arduino-IRremote/Arduino-IRremote #270: Panasonic a/c ir remote issue

psylo24, 2016-02-15:
Dear all,

I tried to decode ircode from my ir remote for my PANASONIC A/C, but it seems the raw code are different than  Arduino-IRremote/ir_Panasonic.cpp.

In the file ir_Panasonic.cpp, i can see : 
#define PANASONIC_BITS          48
#define PANASONIC_HDR_MARK    3502
#define PANASONIC_HDR_SPACE   1750
#define PANASONIC_BIT_MARK     502
#define PANASONIC_ONE_SPACE   1244
#define PANASONIC_ZERO_SPACE   400

But in my raw file (extract of mode2), codes are differents :  (there is 6 columns)
**Test1**
`     3500     1650      500      400      500     1200
      500      350      500      400      500      350
      500      350      500      400      450      400
      500      350      500      350      500      400
      450      400      500      350      500     1250
      450      400      500      350      500      400
      450      400      500      350      500      400
      450      400      450     1300      450     1250
      450     1300      450      400      450      400
      500     1250      450      400      450      400
      500      400      450      400      450      400
      500      400      450      400      450      400
      450      400      500      400      450      400
      450      400      500      400      450      400
      450      400      450      400      500      400
      450      400      450      400      450      450
      450      400      450      400      450      400
      450      450      450      400      450      400
      450      450      450      400      450      400
      450      400      450     1300      450     1300
      450      400      450      400      450      400
      450      450      450      400      450     9950
     3450     1700      450      450      450     1250
      450      400      450      450      450      400
      450      400      450      450      450      400
      450      400      450      450      400      450
      450      400      450      400      450     1300
      450      400      450      450      400      450
      450      400      450      400      450      450
      450      400      450     1300      400     1300
      450     1300      450      400      450      400
      450     1300      450      400      450      400
      450      450      450      400      450      400
      450      450      400      450      450      400
      450      400      450      450      450      400
      450      400      450      450      400     1300
      450      450      400      450      450     1250
      450      450      400      450      450      400
      450      450      400      450      450     1250
      450      450      450     1250      450      450
      400     1300      450      400      450      450
      450      400      450      400      450      450
      400      450      450      400      450      450
      400      450      450     1250      450      450
      400     1300      450      450      400      450
      450      400      450     1300      400      450
      450     1300      400      450      450      400
      450     1300      400     1300      450      450
      400      450      450      400      450      400
      450      450      400      450      450      400
      450      450      400      450      450      400
      450      450      400      450      400      450
      450     1300      400     1300      450     1300
      450      400      450      400      450      450
      400      450      450      400      450      450
      400      450      450      400      450      400
[... 153 lines cut ...]
      450      450      400      450      450      400
      450     1300      450     1250      450     1300
      450      400      450      450      400      450
      450      400      450      400      450      450
      400      450      450      400      450      450
      400      450      450      400      450      450
      400      450      400      450      450      400
      450      450      400     1300      450      450
      400      450      400      450      450      400
      450      450      400      450      450     1300
      400      450      400      450      450      400
      450      450      400      450      450      400
      450      450      400      450      400      450
      450      400      450      450      400      450
      450      400      450      450      400      450
      400      450      450      450      400     1300
      450     1300      400      450      450     1300
      400      450      400     1350      400     1300
      450
`

And i don't undrestand why the results are differents because** i make the same test : ** 
The test is : configure the remote control to 20 degrees and start the recording (with mode2 in raw mode) of infrared codes and press the button to set temperature above the temperature at 21 degrees.

Thanks in advance for your help.

ElectricRCAircraftGuy, 2016-02-15:
I don't understand why you're seeing what you're seeing either, but I wanted to point out a couple things that may help others solve your problem. 

1. in IRremoteInt.h, RAWBUF is set to only 101. That means you can only read in 101 values. In your tests above, however, I count 439 values per test. I'm not sure how this is possible. Even if you set RAWBUF to 439 or more, it still wouldn't work, as it's a uint8_t data type. I recommend it be changed in the source code to a uint16_t data type to support buffer lengths > 255. That requires changing the data type in dozens of places throughout the source code, however.

2. What IR receiver are you using? It's possible that if you change MARK_EXCESS from 100 to 50 in IRremoteInt.h, it will work better.

AnalysIR, 2016-02-15:
@psylo24 
Your signals are fine. I was able to decode them and cleaned one up for you.

here is the Hex decode and C code to send the cleaned up signal with IRremote.
Your signals have 216 data bits :)
````
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : PANASONIC128_216_280AC :  : 40040720000000604004072000905401453000700700008100006B : 216 : 0
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit http://www.AnalysIR.com or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

unsigned int Signal_0_0[] = {3500,1750,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,9950,3500,1750,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,1300,435,435,435,435,435,435,435,1300,435,435,435,1300,435,435,435,435,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,1300,435,435,435,1300,435,1300,435}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: PANASONIC128_216_280AC, Key:  
````
All you need to do is to change the values as in the example I provided above.

You will find it difficult to have all the signals in RAM at the same time. If you search the issues or PRs here you should be able to find an example of storing the signals in flash and sending them from flash without using up RAM. (PROGMEM would be a good word to search for)

Alternatively, you will find several examples of how to send AC signals from the Hex representation on our blog - saving both flash & SRAM.

ElectricRCAircraftGuy, 2016-02-15:
Here's some progmem help to study: https://www.arduino.cc/en/Reference/PROGMEM

rickynonato, 2018-07-10:
I tried the above example to my sketch    and it turn on my panasonic but  what is the raw code for turning it off?

AnalysIR, 2018-07-10:
Use your remote and record the signal!

(you may have to do a bit of searching about recording long AC signals)

closed 2020-10-17 by ArminJo