Arduino-IRremote/Arduino-IRremote #272: Panasonic air condition IR protocol problem

ying-chiao-liao, 2016-02-16:
Hi,

I want to decode Panasonic air condition IR protocol, and then always show unknown type

so I print the raw data out, I saw that raw length is 132.

then I use "https://www.remotecentral.com/cgi-bin/mboard/rc-pronto/thread.cgi?26152" this method to

decode it , and modify the PANASONIC_BITS from 48 to 64 , it will decode success.

But the temperature button and on/off button be decoded to the same value, do you have any ideas?

the raw data is：

temperature up:
[0]:2283[1]:69[2]:27[3]:8[4]:7[5]:8[6]:21[7]:6[8]:7[9]:8[10]:6[11]:7[12]:7[13]:7[14]:8[15]:6[16]:8[17]:7[18]:7[19]:7[20]:7[21]:7[22]:8[23]:6[24]:8[25]:7[26]:7[27]:7[28]:7[29]:8[30]:21[31]:6[32]:7[33]:7[34]:8[35]:6[36]:8[37]:7[38]:7[39]:7[40]:7[41]:7[42]:8[43]:6[44]:8[45]:7[46]:21[47]:6[48]:21[49]:7[50]:21[51]:6[52]:8[53]:6[54]:8[55]:7[56]:21[57]:6[58]:8[59]:6[60]:8[61]:6[62]:8[63]:7[64]:7[65]:7[66]:7[67]:7[68]:8[69]:6[70]:8[71]:7[72]:7[73]:7[74]:7[75]:7[76]:8[77]:6[78]:8[79]:7[80]:7[81]:7[82]:7[83]:7[84]:8[85]:6[86]:8[87]:7[88]:7[89]:7[90]:7[91]:7[92]:8[93]:6[94]:8[95]:6[96]:8[97]:7[98]:7[99]:7[100]:8[101]:6[102]:8[103]:6[104]:8[105]:7[106]:7[107]:7[108]:7[109]:7[110]:8[111]:6[112]:8[113]:7[114]:7[115]:7[116]:7[117]:7[118]:21[119]:7[120]:21[121]:6[122]:8[123]:6[124]:8[125]:7[126]:7[127]:7[128]:7[129]:7[130]:8[131]:6
on/off button:
[0]:-22109[1]:69[2]:28[3]:8[4]:7[5]:7[6]:21[7]:6[8]:7[9]:8[10]:7[11]:6[12]:8[13]:7[14]:7[15]:7[16]:7[17]:7[18]:7[19]:7[20]:8[21]:7[22]:7[23]:7[24]:7[25]:7[26]:7[27]:8[28]:7[29]:7[30]:21[31]:6[32]:7[33]:8[34]:7[35]:7[36]:7[37]:7[38]:8[39]:6[40]:8[41]:6[42]:8[43]:7[44]:7[45]:7[46]:21[47]:6[48]:21[49]:7[50]:21[51]:6[52]:8[53]:7[54]:7[55]:7[56]:21[57]:6[58]:8[59]:7[60]:7[61]:7[62]:7[63]:7[64]:8[65]:6[66]:8[67]:6[68]:8[69]:7[70]:7[71]:7[72]:7[73]:7[74]:8[75]:6[76]:8[77]:7[78]:7[79]:7[80]:7[81]:7[82]:8[83]:6[84]:8[85]:7[86]:7[87]:7[88]:7[89]:7[90]:8[91]:6[92]:8[93]:7[94]:7[95]:7[96]:7[97]:7[98]:8[99]:6[100]:8[101]:6[102]:8[103]:7[104]:7[105]:7[106]:7[107]:7[108]:8[109]:6[110]:8[111]:7[112]:7[113]:7[114]:7[115]:7[116]:8[117]:6[118]:21[119]:7[120]:21[121]:6[122]:8[123]:6[124]:8[125]:7[126]:7[127]:7[128]:8[129]:6[130]:8[131]:6

Thanks you a lot !

ying-chiao-liao, 2016-02-16:
temperature up button :
[0]:2283[1]:69[2]:27[3]:8[4]:7[5]:8[6]:21[7]:6[8]:7[9]:8[10]:6[11]:7[12]:7[13]:7[14]:8[15]:6[16]:8[17]:7[18]:7[19]:7[20]:7[21]:7[22]:8[23]:6[24]:8[25]:7[26]:7[27]:7[28]:7[29]:8[30]:21[31]:6[32]:7[33]:7[34]:8[35]:6[36]:8[37]:7[38]:7[39]:7[40]:7[41]:7[42]:8[43]:6[44]:8[45]:7[46]:21[47]:6[48]:21[49]:7[50]:21[51]:6[52]:8[53]:6[54]:8[55]:7[56]:21[57]:6[58]:8[59]:6[60]:8[61]:6[62]:8[63]:7[64]:7[65]:7[66]:7[67]:7[68]:8[69]:6[70]:8[71]:7[72]:7[73]:7[74]:7[75]:7[76]:8[77]:6[78]:8[79]:7[80]:7[81]:7[82]:7[83]:7[84]:8[85]:6[86]:8[87]:7[88]:7[89]:7[90]:7[91]:7[92]:8[93]:6[94]:8[95]:6[96]:8[97]:7[98]:7[99]:7[100]:8[101]:6[102]:8[103]:6[104]:8[105]:7[106]:7[107]:7[108]:7[109]:7[110]:8[111]:6[112]:8[113]:7[114]:7[115]:7[116]:7[117]:7[118]:21[119]:7[120]:21[121]:6[122]:8[123]:6[124]:8[125]:7[126]:7[127]:7[128]:7[129]:7[130]:8[131]:6
on/off button:
[0]:-22109[1]:69[2]:28[3]:8[4]:7[5]:7[6]:21[7]:6[8]:7[9]:8[10]:7[11]:6[12]:8[13]:7[14]:7[15]:7[16]:7[17]:7[18]:7[19]:7[20]:8[21]:7[22]:7[23]:7[24]:7[25]:7[26]:7[27]:8[28]:7[29]:7[30]:21[31]:6[32]:7[33]:8[34]:7[35]:7[36]:7[37]:7[38]:8[39]:6[40]:8[41]:6[42]:8[43]:7[44]:7[45]:7[46]:21[47]:6[48]:21[49]:7[50]:21[51]:6[52]:8[53]:7[54]:7[55]:7[56]:21[57]:6[58]:8[59]:7[60]:7[61]:7[62]:7[63]:7[64]:8[65]:6[66]:8[67]:6[68]:8[69]:7[70]:7[71]:7[72]:7[73]:7[74]:8[75]:6[76]:8[77]:7[78]:7[79]:7[80]:7[81]:7[82]:8[83]:6[84]:8[85]:7[86]:7[87]:7[88]:7[89]:7[90]:8[91]:6[92]:8[93]:7[94]:7[95]:7[96]:7[97]:7[98]:8[99]:6[100]:8[101]:6[102]:8[103]:7[104]:7[105]:7[106]:7[107]:7[108]:8[109]:6[110]:8[111]:7[112]:7[113]:7[114]:7[115]:7[116]:8[117]:6[118]:21[119]:7[120]:21[121]:6[122]:8[123]:6[124]:8[125]:7[126]:7[127]:7[128]:8[129]:6[130]:8[131]:6

z3t0, 2016-02-17:
@ying-chiao-liao 
Please use 
"      ``` and ``` "

z3t0, 2016-02-20:
Is it possible that the length is longer? And you are only seeing part of the signal

ying-chiao-liao, 2016-02-22:
Yes,

I use oscilloscope to saw that my Panasonic air condition remote is sent ir twice, thank you a lot!!

closed 2016-02-22 by ying-chiao-liao