Arduino-IRremote/Arduino-IRremote #273: Virgin Tivo IR Send not working

TheGrovesy, 2016-02-17:
Hi there, I am successfully using your library to send IR codes to my Sony TV so I am happy my setup is correct. However I am now trying to send IR messages to my Virgin TV box with no success.

I have used the IRRecvDump example to see what Protocol its using and the program says its using Samsung so I have tried the following code but the Tivo does nothing: 
irsend.sendSAMSUNG(0x699D732A, 32);

I noticed another user suggested that sendNEC works with their Tivo so I also tried that but also no success.

Any ideas? I have tried googling the for Tivo library but no luck.

Any help would be greatly appreciated.

AnalysIR, 2016-02-17:
Difficult to comment without seeing an actual signal.

Why not post the output of IRRecvDump!

I also checked out a random Tivo signal posted online and it was bog standard NEC.

TheGrovesy, 2016-02-20:
Yes I tried NEC but no joy. Dump below from a few buttons.... interestingly it looks like this controller uses several different protocols, the TV guide button says unknown, but the play buttons says RC5!

Any help would be great.

**From the TV Guide Button:**
Unknown encoding: 699D732A (32 bits)
Raw (16): 1850 -1650 1850 -1700 1800 -1700 1850 -800 950 -1700 1800 -850 900 -1700 1850 

**Play button:**
AB0
Decoded RC5: AB0 (12 bits)
Raw (20): 950 -800 950 -800 1850 -1700 1800 -1700 1850 -1700 900 -850 1800 -850 900 -850 950 -800 950 

**Blue Button:**
5CE56EE
Unknown encoding: 5CE56EE (32 bits)
Raw (18): 1850 -850 900 -850 900 -1700 1850 -1700 1800 -1700 1800 -1700 950 -800 950 -850 1800

AnalysIR, 2016-02-21:
@TheGrovesy 
All 3 of your signals above decode as RC5 when imported into AnalysIR.

What IR receiver are you using (model # or link)?

Probably just a timing issue with your receiver (less likey dev board)

AnalysIR, 2016-02-21:
2AE
292
are the 2 other codes

TheGrovesy, 2016-02-21:
Hi thanks... i am pretty sure I ended up trying all manufactures functions but I will try NEC again and let you know.
I am using the following receiver: https://www.coolcomponents.co.uk/ir-receiver-38khz.html there are no other electronics involved, just directly connected to my Arduino Uno. Should I have anything else?
What do you mean 2AE & 292 at the other codes??

TheGrovesy, 2016-02-21:
Nope still nothing happening. This is my code just to try to get the Tivo to react to somerthing:-

`  //play button
  irsend.sendNEC(0x699D732A, 32);
  delay(2000);
  
  irsend.sendNEC(0x8F199BAB, 32);
  delay(2000);

  // blue button
  irsend.sendNEC(0x5CE56EE, 32);
  delay(2000);

  irsend.sendNEC(0xB237BBA3, 32);
  delay(2000);

  // pause button
  irsend.sendNEC(0x293, 32);
  delay(2000);

  irsend.sendNEC(0xA93, 32);
  delay(2000);`

AnalysIR, 2016-02-21:
try

```
irsend.sendRC5(2AE);
delay(5000);
irsend.sendRC5(292);
delay(5000);
irsend.sendRC5(AB0);
delay(5000);
```
They are **not** NEC signals.

TheGrovesy, 2016-02-21:
still nothing :(  I tried this:- 

  irsend.sendRC5(0x2AE, 32);
  delay(5000);
  
  irsend.sendRC5(0x292, 32);
  delay(5000);

  irsend.sendRC5(0xAB0, 32);
  delay(5000);

AnalysIR, 2016-02-21:
Sorry - for the "bits" paramter which is 12  (not 32)

```
irsend.sendRC5(0x2AE,12);
delay(5000);
irsend.sendRC5(0x292,12);
delay(5000);
irsend.sendRC5(0xAB0,12);
delay(5000);
```
PS: when inserting code use the code tags provided, at the top of the editing panel.

TheGrovesy, 2016-02-21:
OK now something is happening. one of the codes (AB0) is working (the play button). The other two codes doesnt seem to do anything... where have you got these from? However I have now got other codes to work... but not all of them. The following dump is of the Info button which appears to be a longer code (32), just doing a `irsend.sendRC5(0xD03780EA, 32)` does nothing again. Ultimately i plan to use the receiver to get the codes, transmit them via 2.4Ghz wireless to another room and then send them with this transmit IR setup I am currently testing with. 

Unknown encoding: 938F2CCF (32 bits)
Raw (20): 1850 -800 950 -800 950 -1700 1800 -1700 1850 -800 950 -1700 1800 -1700 950 -800 950 -800 950 
D03780EA
Unknown encoding: D03780EA (32 bits)
Raw (18): 1850 -1700 1800 -1700 1800 -1750 1800 -850 900 -1700 1800 -1750 900 -850 900 -850 900

AnalysIR, 2016-02-21:
The last 2 are both the same - only difference is one has a toggle bit set and the other doesnt.
```
irsend.sendRC5(0x297,12);
delay(5000);
```
Baseed on another discussion, this could be related  to an issue with the library.

If you try to change the definition of the `MARK_EXCESS` parameter in the library to 150,175 or even 200, you might get better results. I am not 100% certain - but worth a try. You will also have to restart the Arduino IDE each time you change any setting in the library.

`#define MARK_EXCESS 100`
change to one of the following

`#define MARK_EXCESS 150`
or
`#define MARK_EXCESS 175`
or
`#define MARK_EXCESS 200`

you can find this in the IRremoteInt.h file around line 124 of the latest version.

TheGrovesy, 2016-02-21:
So do you think this is an issue with the receive function and that all signals should appear as a 12 bit value?
Thanks again for the help very much appreciated, I'll let you know how I get on.

AnalysIR, 2016-02-21:
...it **could potentially** be an issue with the default settings of the library, when receiving/decoding?

Will know more when you try out the suggestions above

TheGrovesy, 2016-02-24:
Hi, ok I have tried all those values and more (0, 50, 500), all made no difference unfortunately (except the 500 value when all the working buttons were giving bad values also).

I also tried different values for the TOLERANCE and _GAP just out of curiosity but again no difference.

Any other ideas?

AnalysIR, 2016-02-24:
try using **sendRAW** usng the raw timings above and a frequency of 36 (kHz).

If that doesnt work, then change the RAW values to the closest of 888 or 1776

**e.g.**
_Raw (20): 950 -850 1800 -850 900 -1700 1800 -1750 1800 -1700 900 -850 1800 -850 900 -900 850 -900 900_

change to 

**_Raw (20): 888 -888 1776 -888 888 -1776 1776 -1776 1776 -1776 888 -888 1766 -888 888 -888 888 -888 888_**

closed 2020-10-17 by ArminJo