Arduino-IRremote/Arduino-IRremote #274: cant compile on an AdaFruit Feather M0

leszek77, 2016-02-19:
Hello i would like to run an ir remote on my Feather Logger M0
but there is an other processor structure

__SAMD21G18A__ 

i dont know how to fix the interrupts

thank you

z3t0, 2016-02-20:
I do not think this board is supported by the library?

ElectricRCAircraftGuy, 2016-02-20:
32 bit boards are not yet supported. This would be a great addition, but would be a lot of work. The low level hardware needs to be well understood.

z3t0, 2016-02-20:
Certain Teensy boards are supported which are 32Bit.

@leszek77 Do you know of any software simulators for that board/cpu that way I can try and work on supporting it

ElectricRCAircraftGuy, 2016-02-20:
Ah, I need to look closer next time. Thanks for the correction.

AnalysIR, 2016-02-20:
Teensy is supported because the designer/manufacturer, @PaulStoffregen , invested the time to add support as a contributor to the liberary, which is to be commended.

I guess @leszek77 could ask the supplier, Adafruit, to add the support for their board to the library. 

Otherwise it would be a big task to cover every Arm-esque board that emerges. Must be 100s out there and many more to come. How do you decide which ones to add?

@z3t0 before jumping in, you should 'maybe' devise and publish a policy for issues like this. Difficult to be "all things to all men/women"....

Of course anyone who wants to and has the time and expertise, should be encouraged to add support for any board as long as it doesn't break the core structure.

Another approach could be to publish a short guide with tips on how to add a new board/platform to the library.

z3t0, 2016-02-20:
> @z3t0 before jumping in, you should 'maybe' devise and publish a policy for issues like this. Difficult to be "all things to all men/women"....

Yes exactly, I am currently working on a set of "guidelines" and I will try and think of some thing to deal with this, for the most part i think we will recommend the supplier invest the time however if adding a board is trivial and there is a lot of support from the community we can try and take it on.

>Another approach could be to publish a short guide with tips on how to add a new board/platform to the library.

This is also possible but for the most part people using this library would not know how to do that..

AnalysIR, 2016-02-20:
Agreed. Just a list of pointers for board suppliers or experienced devs who want to take it on.
For anyone with good tech knowledge of a particular platform, it shouldn't be too difficult a task. 

One thing to remember, is that the library was originally designed to use minimal resources on an 8-bit AVR - leaving enough resources for other sketches. Anyone porting to a 32-bit platforms with lot more flash, RAM & peripherals etc - might not be happy with the historic "AVR" limitations. In that scenario, there may be a case for an 8-bit version(compact) and a **different** 32-bit version of the library(comprehensive).

AnalysIR, 2016-02-21:
re: https://github.com/z3t0/Arduino-IRremote/issues/273

It seems that somewhere along the way MARK_EXCESS is being used instead 
of tolerance for the variation in signal timings.

Tolerance used to be 25%, which is reasonable - although a fixed amount 
up to 200 would be better.
Actually, it still is defined at 25% - but doesn't seem to be used in 
the code(I couldn't find a reference to it being used)

Mark_Excess was originally meant only to deal with IR receivers.

It may be that a lot of the library is fundamentally flawed because of 
this change that happened - (when?).
I looked at some old copies of the library I have and the change was 
there (circa 2014) so it must have happened earlier than that.

The reason I didn't post this is because I haven't gone into it in great 
detail, but thought you should be aware of the possibility, for reference.

At a minimum the current terminology seems to be confusing/misleading.

Sorry, I cant be of more help at this time.

rgds
PS: Thanks for sorting out that other stuff so promptly.

    /// Pulse parms are ((X*50)-100) for the Mark and ((X*50)+100) for
    the Space.//
    //// First MARK is the one after the long gap//
    //// Pulse parameters in uSec//
    //////
    //
    //// Due to sensor lag, when received, Marks  tend to be 100us too
    long and//
    //// Spaces tend to be 100us too short//
    //#define *MARK_EXCESS*    100//
    //
    //// microseconds per clock interrupt tick//
    //#define USECPERTICK    50//
    //
    //// Upper and Lower percentage tolerances in measurements//
    //#define *TOLERANCE *25//
    //#define *LTOL *(1.0 - (*TOLERANCE*/100.))//
    //#define *UTOL *(1.0 + (*TOLERANCE*/100.))/

z3t0, 2016-02-21:
You're right... I'll see if I can track down when the change happened

z3t0, 2016-02-21:
Just as a thought, do we know if the tolerance was ever used?

z3t0, 2016-02-21:
@leszek77 Right now your best bet would be to contact Adafruit

bengtmartensson, 2020-10-18:
I _think_ that the current support for M0 should cover the Feather M0 too. Can the OP test? Otherwise I have an Feather M0 (without "logger") here...

closed 2021-01-15 by ArminJo