Arduino-IRremote/Arduino-IRremote #280: Freezing device receiver while sending NEC code

Mazdari, 2016-02-21:
Hi all.

I successfully controlled my tube led with IRremote library. Before doing so, I got all IR codes from the orignal remote control, and I got no problem. Using the damper receiver example, the program ouputs for each code a 32bits NEC code. For instance, this one:

`#define RED               0xF720DF`

I got all my codes in this format, and it works pretty well for most of them (24 codes in total). But for 8 of them, sending the code freezes the tube led, and no other sended code can make its color changing. When it freezes, I use the original remote control, and it works.

I'm using the code example of the library to send my code. I found someone using the same IR remote here (the first remote): [http://woodsgood.ca/projects/2015/02/13/rgb-led-strip-controllers-ir-codes/]
He doesn't seem to have problem...

It's like the code I am sending is not complete... When I use this code:
`#define ORANGE            0xF710EF`

The color changes and then the tube led freezes.

Any idea ? Should the code be longer than 32 bits finally, meaning I'm sending a partial code ? But as it works for others colors, I'm not sure...
If you have a clue...

AnalysIR, 2016-02-21:
I don't see any obvious issue with what you poasted. Are you using the latest version of the library?
The HEX for the code below seeoms correct.(last 2 bytes are the inverse of each other)

> It's like the code I am sending is not complete... When I use this code:
#define ORANGE 0xF710EF

>The color changes and then the tube led freezes.

Maybe post the raw output for the captured signals that are having a problem.
use irrecvDump example sketch and post the output.

Have you considered an issue with your circuit/wiring?

Mazdari, 2016-02-21:
I'm using the last version, yes (2.1.0). 

For the orange code, here are some raw outputs I have, using the IRrecvDumpV2 example:

```
Encoding  : NEC
Code      : F710EF (32 bits)
Timing[67]: 
     +8850, -4400     + 550, - 550     + 600, - 500     + 600, - 500
     + 600, - 500     + 600, - 500     + 600, - 500     + 600, - 550
     + 550, - 550     + 550, -1650     + 600, -1600     + 600, -1600
     + 600, -1600     + 600, - 500     + 600, -1600     + 600, -1650
     + 600, -1600     + 600, - 500     + 600, - 500     + 600, - 500
     + 600, -1600     + 600, - 550     + 600, - 500     + 550, - 550
     + 600, - 500     + 600, -1600     + 600, -1650     + 550, -1650
     + 600, - 500     + 550, -1650     + 600, -1600     + 600, -1650
     + 550, -1650     + 600
unsigned int  rawData[67] = {8850,4400, 550,550, 600,500, 600,500, 600,500, 600,500, 600,500, 600,550, 550,550, 550,1650, 600,1600, 600,1600, 600,1600, 600,500, 600,1600, 600,1650, 600,1600, 600,500, 600,500, 600,500, 600,1600, 600,550, 600,500, 550,550, 600,500, 600,1600, 600,1650, 550,1650, 600,500, 550,1650, 600,1600, 600,1650, 550,1650, 600};  // NEC F710EF
unsigned int  data = 0xF710EF;

Encoding  : NEC
Code      : F710EF (32 bits)
Timing[67]: 
     +8900, -4400     + 550, - 550     + 550, - 550     + 600, - 500
     + 550, - 550     + 600, - 500     + 600, - 500     + 600, - 550
     + 550, - 500     + 600, -1600     + 600, -1650     + 550, -1650
     + 600, -1600     + 600, - 500     + 600, -1650     + 550, -1650
     + 550, -1650     + 600, - 500     + 600, - 500     + 600, - 500
     + 600, -1650     + 550, - 550     + 550, - 550     + 550, - 550
     + 600, - 500     + 600, -1600     + 600, -1600     + 600, -1600
     + 600, - 550     + 550, -1650     + 600, -1600     + 600, -1600
     + 600, -1600     + 600
unsigned int  rawData[67] = {8900,4400, 550,550, 550,550, 600,500, 550,550, 600,500, 600,500, 600,550, 550,500, 600,1600, 600,1650, 550,1650, 600,1600, 600,500, 600,1650, 550,1650, 550,1650, 600,500, 600,500, 600,500, 600,1650, 550,550, 550,550, 550,550, 600,500, 600,1600, 600,1600, 600,1600, 600,550, 550,1650, 600,1600, 600,1600, 600,1600, 600};  // NEC F710EF
unsigned int  data = 0xF710EF;

Encoding  : NEC
Code      : F710EF (32 bits)
Timing[67]: 
     +8900, -4400     + 600, - 450     + 650, - 500     + 600, - 500
     + 600, - 500     + 600, - 500     + 600, - 500     + 650, - 450
     + 650, - 450     + 650, -1550     + 650, -1600     + 600, -1600
     + 600, -1600     + 600, - 500     + 650, -1550     + 650, -1600
     + 600, -1600     + 600, - 500     + 600, - 500     + 600, - 500
     + 650, -1550     + 650, - 450     + 650, - 500     + 600, - 450
     + 650, - 500     + 600, -1600     + 600, -1600     + 650, -1550
     + 650, - 500     + 600, -1600     + 600, -1600     + 600, -1600
     + 600, -1600     + 650
unsigned int  rawData[67] = {8900,4400, 600,450, 650,500, 600,500, 600,500, 600,500, 600,500, 650,450, 650,450, 650,1550, 650,1600, 600,1600, 600,1600, 600,500, 650,1550, 650,1600, 600,1600, 600,500, 600,500, 600,500, 650,1550, 650,450, 650,500, 600,450, 650,500, 600,1600, 600,1600, 650,1550, 650,500, 600,1600, 600,1600, 600,1600, 600,1600, 650};  // NEC F710EF
unsigned int  data = 0xF710EF;
```

For the circuit, it's very simple like here: [https://learn.sparkfun.com/tutorials/ir-communication] using only the led part to send of course.

I also don't understand why the code is given as an unsigned int, as I got an Uno, which unsigned int are encoded in 16 bits. I use long type in my program (32bits), even if the code has only a 24bits length (I suppose the first bytes in the code are 00F7 for the manufacturer).

Could it be an alternative NEC protocol ? Something here is missing...
Or is it possible that the irsend function doesn't properly send the final information to close the message ?
The NEC protocol documentation says:
> a final 562.5µs pulse burst to signify the end of message transmission.

What do you think ?

Mazdari, 2016-02-21:
To be more precise, here is the list of all my codes and where it bugs:

```
#define ON                0XF7C03F
#define OFF               0xF740BF
#define BRIGHTNESS_UP     0xF700FF
#define BRIGHTNESS_DOWN   0xF7807F //bug
#define FLASH             0xF7D02F
#define STROBE            0xF7F00F
#define FADE              0xF7C837 //bug
#define SMOOTH            0xF7E817
 
#define RED               0xF720DF
#define GREEN             0XF7A05F
#define BLUE              0xF7609F
#define WHITE             0xF7E01F // bug

#define ORANGE            0xF710EF //bug
#define YELLOW_DARK       0xF730CF //bug
#define YELLOW_MEDIUM     0xF708F7 //bug
#define YELLOW_LIGHT      0xF728D7
 
#define GREEN_LIGHT       0XF7906F
#define GREEN_BLUE1       0XF7B04F
#define GREEN_BLUE2       0XF78877
#define GREEN_BLUE3       0XF7A857
 
#define BLUE_RED          0XF750AF
#define PURPLE_DARK       0XF7708F //bug
#define PURPLE_LIGHT      0XF748B7 //bug
#define PINK              0XF76897
```

AnalysIR, 2016-02-21:
1. FYI: I asked for output from IRrecvDump & had to spend extra time working with IRrecvDumpV2 :(
2. The signals are fine. 100% NEC 32 bits
3. Unsigned int refers to the duration values which are 16 bit. The 32 bit refers to 0XF7C03F which would be an unsigned long.
4. xxx.irsendNEC(0xF710EF,32); should work fine. If not the issues is likely with your circuit.
5. You have provided vague/unclear information about your transmitter circuit, which could be the issue. Can't help without good info.
6. If you want to use sendRAW, then try the following. Note I have cleaned up the timings

```
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History:
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : NEC :  : 00F710EF : 32 : 0
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit http://www.AnalysIR.com or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency
irsend.sendNEC(0x00F710EF, 32); //AnalysIR Batch Export (IRremote) // AnalysIR IR Protocol: NEC, Key:
```

Mazdari, 2016-02-22:
Sorry, I thought it would be easier for you with the V2 program...
You didn't give me the raw code... I re-tried the line `irsend.sendNEC(0x00F710EF, 32);` without success.

For the circuit, I plugged the I/O pin 3 to a resistor, which is connected to the IR Led and then to the ground.
Here are details about the components:
- Resistor: 200 ohms (because the Led works at 1.6 volts)
- IR Led: L-7113F3BT model
[Datasheet here](http://www.kingbright.com/attachments/file/psearch/000/00/00/L-7113F3BT(Ver.15A).pdf)

My program is simply:

```
#include <IRremote.h>

IRsend irsend;

void setup()
{
}

void loop() {
                
     irsend.sendNEC(0x00F720DF, 32); //red color working
     delay(1000);
  
    irsend.sendNEC(0x00F710EF, 32); //orange color freezing the tube led
    delay(1000);
}
```

AnalysIR, 2016-02-22:
> You didn't give me the raw code...

...because **you** have already posted it above !

Try a 5 second delay (long shot)

Your IR Led is fine & your Vf will be more like 1.2V (1.6V is the max)

Try a 100R resistor (or 2 x 200r in parallel) or even 75R if that doesn't work.
Make sure you are pointing the top of the IR LEd towards the reciver on the LED strip & bring it close.

Mazdari, 2016-02-23:
It was not 1.6V but 1.2V absolutely. It's why I'm using a 200R resistor. 
When I talked about the raw code, it's because you told me that "you cleaned up the timings". In fact, you were talking about the HEX code.

The power of the LED has nothing to do with that. In fact, I found by myself, it was a problem of protocol, but not the information itself. It was a problem of frequency. As the tube led I use is a cheap one, I suspect the receiver to be crap... So I tried the raw code changing the frequency to 38.1 KHz. And it works.

Anyway, thanks for your time.

Mazdari, 2016-02-23:
In fact, I tried raw codes and try to change the frequency, also the resistor.

It seems that the best solution is having more power given to the led (100R resistor). But I succeed also by changing the frequency without changing the resistor (200R) which is weird... By doing so, the tube freeze after few loops. With the 100R resistor, it works perfectly. 

Thanks for your time !

AnalysIR, 2016-02-24:
great - thanks for the update.

If you used a transistor circuit to drive the IR LED the performance/range would be even better. (There are many example circuits available via google and also on our own blog)_

z3t0, 2017-08-22:
Great, closing the issue for now. Feel free to reopen if you still need help with this.

closed 2017-08-22 by z3t0