Arduino-IRremote/Arduino-IRremote #282: Mitsubishi, suspicions of faulty code

adamlhumphreys, 2016-02-22:
**Board:** N/A
**Library Version:** 2.1.0
**Protocol:** Mitsubishi

**Code Block:**
```c
12 // Looks like Sony except for timings, 48 chars of data and time/space different 
...
19 #define MITSUBISHI_HDR_SPACE	350 //  7*50+100 
20 #define MITSUBISHI_ONE_MARK	1950 // 41*50-100 
21 #define MITSUBISHI_ZERO_MARK  750 // 17*50-100 
...
60   if (!MATCH_MARK(results->rawbuf[offset], MITSUBISHI_HDR_SPACE))  return false ; 
61   offset++; 
62 
 
63   while (offset + 1 < irparams.rawlen) { 
64     if      (MATCH_MARK(results->rawbuf[offset], MITSUBISHI_ONE_MARK))   data = (data << 1) | 1 ; 
65     else if (MATCH_MARK(results->rawbuf[offset], MITSUBISHI_ZERO_MARK))  data <<= 1 ; 
66     else                                                                 return false ; 
67     offset++; 
68 
 
69     if (!MATCH_SPACE(results->rawbuf[offset], MITSUBISHI_HDR_SPACE))  break ; 
70     offset++; 
71   } 

```
Unless I'm missing something, this doesn't look like Sony. :/
It does look like a header-less decode that's been written backwards though. Swapping "MATCH" and "SPACE" with each other in the macro variables and MATCH calls in the while() loop would make it look as expected.
Unless I'm missing something, this is highly suspect. ^^; There's also the same misleading debug code (lines 35-41) I corrected in the Sanyo protocol in #268. There's also a lot of other commented code where offset could be initially set to 1 (without offset++;) as most decodes.

z3t0, 2016-02-22:
Interesting, we should look for some official documentation for the Mitsubishi protocol

z3t0, 2016-02-22:
This could possibly be mitsubishi http://www.sbprojects.com/knowledge/ir/xsat.php

adamlhumphreys, 2016-02-22:
I have to say, that definitely looks a lot different. :O Official documentation would be awesome, though I doubt that information is available to the public. (Could be wrong.) Nevertheless, that link may well be accurate, not to mention recent. :)

z3t0, 2016-07-11:
#333 
trying to keep the effort on this collectivized

closed 2016-07-11 by z3t0